espressif/esp-matter #1054: Only a few pieces of information can be written using 'generate_esp32uchip_factory-bin. py'

MaplestoryAlen, 2024-08-15:
This is the command for writing：

   ./generate_esp32_chip_factory_bin.py -d 1083 -p 10831125 \
--product-name 81ma1 --product-id 0x8000 \
--vendor-name Testvendor --vendor-id 0xFFF1 \
--hw-ver 1009 --hw-ver-str 1.0.0.9  --serial-num 202408091125 \
--dac-cert /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-01.der \
--dac-key /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-Key-01.der \
--pai-cert /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-PAI-Cert.der \
--cd /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/esp_dac_fff1_8000.der --dac-in-secure-cert --target esp32s3 

But other information cannot be written, such as the Product URL and Product Label. Is there any way for me to write this information and burn it into the chip? I saw some API interfaces in "platform/ESP32/ESP32FactorySData Provider. h"

     CHIP_ERROR GetProductURL(char * buf, size_t bufSize) override;
     CHIP_ERROR GetProductLabel(char * buf, size_t bufSize) override;

   At present, I can only read a few, and I don't know how to write the others, so reading has failed

      ESP32FactoryDataProvider dataProvider;

	if(dataProvider.GetProductName(device_info.productName, sizeof(device_info.productName)) == CHIP_NO_ERROR)
	{
	   ESP_LOGI(TAG,"device_info.productName :%s",device_info.productName); 
	}
	if(dataProvider.GetHardwareVersionString(device_info.HardwareVersionString, sizeof(device_info.HardwareVersionString)) == CHIP_NO_ERROR)		
	{
	   ESP_LOGI(TAG,"device_info.HardwareVersionString :%s",device_info.HardwareVersionString);
	}

  May I ask if there is any way for me to write and read information from the Basic Information Cluster?

shubhamdp, 2024-08-17:
Those were recently added in https://github.com/project-chip/connectedhomeip/pull/34363, can you try the script from connectedhomeip's master branch?

Or you can use [esp-matter-mfg-tool](https://github.com/espressif/esp-matter-tools/tree/main/mfg_tool). It supports product label and product URL, and others will be added soon.

```
# to install
pip install esp-matter-mfg-tool

# usage
esp-matter-mfg-tool -h
```

MaplestoryAlen, 2024-08-20:
> Those were recently added in [project-chip/connectedhomeip#34363](https://github.com/project-chip/connectedhomeip/pull/34363), can you try the script from connectedhomeip's master branch?
> 
> Or you can use [esp-matter-mfg-tool](https://github.com/espressif/esp-matter-tools/tree/main/mfg_tool). It supports product label and product URL, and others will be added soon.
> 
> ```
> # to install
> pip install esp-matter-mfg-tool
> 
> # usage
> esp-matter-mfg-tool -h
> ```

I am using the factory mode, and there is a difference between the two tools "generate_esp32uchip_factory-bin" and "esp-mtter-mfg-tool". How can I use "esp-mtter-mfg-tool" to write the file "Esp-Development-PAI-Cert. der" into the chip? If the item '-- pai' is' -- pai/home/alen/esp-matter/connected homeip/connected homeip/out/host/Esp Development PaiCert.der ', an error will occur

1.generate_esp32_chip_factory_bin.py command:  

./generate_esp32_chip_factory_bin.py -d 1083 -p 10831125 \
--product-name 81ma1 --product-id 0x8000 \
--vendor-name Testvendor --vendor-id 0xFFF1 \
--hw-ver 1009 --hw-ver-str 1.0.0.9  --serial-num 202408091125 \
--dac-cert /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-01.der \
--dac-key /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-Key-01.der \
--pai-cert /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-PAI-Cert.der \
--cd /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/esp_dac_fff1_8000.der --dac-in-secure-cert --target esp32s3 

2. esp-matter-mfg-tool command  
 esp-matter-mfg-tool --discriminator 1083 --passcode 10831022 \
--product-name 81ma1 --product-id 0x8000 \
--vendor-name Testvendor --vendor-id 0xFFF1 \
--hw-ver 1009 --hw-ver-str 1.0.0.9  --serial-num 202408201022 \
--product-label xz.81ma1.202408201022.AirConditioner \
--dac-cert /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-01.der \
--dac-key /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-Key-01.der \
-pai  /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-PAI-Cert.der \
-cd /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/esp_dac_fff1_8000.der --dac-in-secure-cert --target esp32s3 

log:
[image]

shubhamdp, 2024-08-23:
@MaplestoryAlen sorry for a delayed response...

--pai do not take in a cert, for providing CA certs you have to use the `-c` option. please see `esp-matter-mfg-tool -h` for more details about the options. I have modified the command for you...

```
esp-matter-mfg-tool --discriminator 1083 --passcode 10831022
--product-name 81ma1 --product-id 0x8000
--vendor-name Testvendor --vendor-id 0xFFF1
--hw-ver 1009 --hw-ver-str 1.0.0.9 --serial-num 202408201022
--product-label xz.81ma1.202408201022.AirConditioner
--dac-cert /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-01.pem
--dac-key /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-Key-01.pem
--pai -C /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-PAI-Cert.pem
-cd /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/esp_dac_fff1_8000.der --dac-in-secure-cert --target esp32s3
```

If you have these certs in PEM format then that would be great, the utility right now works with PEM certs and need support for DER (will do that soon)

DER to PEM
```
// certs
openssl x509 -inform der -outform pem -in Esp-Development-DAC-01.der -out Esp-Development-DAC-01.pem
openssl x509 -inform der -outform pem -inEsp-Development-PAI-Cert.der -out Esp-Development-PAI-Cert.pem

//key
openssl ec -inform der -outform pem -in Esp-Development-DAC-Key-01.der -out Esp-Development-DAC-Key-01.pem
```

MaplestoryAlen, 2024-09-05:
> @MaplestoryAlen sorry for a delayed response...
> 
> --pai do not take in a cert, for providing CA certs you have to use the option. please see for more details about the options. I have modified the command for you...`-c``esp-matter-mfg-tool -h`
> 
> ```
> esp-matter-mfg-tool --discriminator 1083 --passcode 10831022
> --product-name 81ma1 --product-id 0x8000
> --vendor-name Testvendor --vendor-id 0xFFF1
> --hw-ver 1009 --hw-ver-str 1.0.0.9 --serial-num 202408201022
> --product-label xz.81ma1.202408201022.AirConditioner
> --dac-cert /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-01.pem
> --dac-key /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-DAC-Key-01.pem
> --pai -C /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/Esp-Development-PAI-Cert.pem
> -cd /home/alen/esp-matter/connectedhomeip/connectedhomeip/out/host/esp_dac_fff1_8000.der --dac-in-secure-cert --target esp32s3
> ```
> 
> If you have these certs in PEM format then that would be great, the utility right now works with PEM certs and need support for DER (will do that soon)
> 
> DER to PEM
> 
> ```
> // certs
> openssl x509 -inform der -outform pem -in Esp-Development-DAC-01.der -out Esp-Development-DAC-01.pem
> openssl x509 -inform der -outform pem -inEsp-Development-PAI-Cert.der -out Esp-Development-PAI-Cert.pem
> 
> //key
> openssl ec -inform der -outform pem -in Esp-Development-DAC-Key-01.der -out Esp-Development-DAC-Key-01.pem
> ```

I used a tool to generate bin files and tried testing mode, but both encountered problems

partition.csv

[image]

[image]

CSV file:
[image]

1.Write flash "partition.bin" and "esp_secure_cert.bin"
esptool.py -p /dev/ttyUSB0 write_flash  0xd000 /home/alen/esp-matter/connectedhomeip/connectedhomeip/scripts/tools/bin/801e465e-4c1b-45ba-b584-b6ef16972918_esp_secure_cert.bin
esptool.py -p /dev/ttyUSB0 write_flash 0x10000 /home/alen/esp-matter/connectedhomeip/connectedhomeip/scripts/tools/bin/801e465e-4c1b-45ba-b584-b6ef16972918-partition.bin
2.chip-tool read basicinformation
  /chip-tool basicinformation read product-name 0x11 0
log:
[image]

I wrote product name: 81ma1 using esp-mtter-mfg-tool, but the actual read product name :TEST PRODUCT

Set the factory mode and wrote 'XXX_ partition. bin' into fctry, which may result in the inability to distribute the network,
（esptool.py -p  /dev/ttyUSB0  write_flash   0xE40000   /home/alen/esp-matter/connectedhomeip/connectedhomeip/scripts/tools/bin/801e465e-4c1b-45ba-b584-b6ef16972918-partition.bin）

May I ask where I went wrong? I need to write some information into the chip now, and be able to read this information in cluster ->basic information, such as "productname", "product label" and other information.

shubhamdp, 2024-09-06:
Can you please share the sdkconfig. I suspect few config options are not set...

Please enable [CONFIG_ ENABLE_ESP32_FACTORY_DATA_PROVIDER](https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/Kconfig#L950) and then set

- [dac provider](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/Kconfig#L48C16-L48C37) option to `CONFIG_SEC_CERT_DAC_PROVIDER`
- [Commissionable data provider](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/Kconfig#L73) to `CONFIG_ FACTORY_COMMISSIONABLE_DATA_PROVIDER`
- [device instance info provider](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/Kconfig#L98) to `CONFIG_ FACTORY_DEVICE_INSTANCE_INFO_PROVIDER`

Also, I hope that [CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL](https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/Kconfig#L1224) is unchanged and its "nvs", cause you are flashing at "nvs".

I know, the config options are a bit messy, and we are trying to figure out some way to simplify them.

MaplestoryAlen, 2024-09-09:
> * FACTORY_DEVICE_INSTANCE_INFO_PROVIDER

Please check the log file
[log.txt](https://github.com/user-attachments/files/16927713/log.txt)

The Commissioning window is closed when the device is powered on because the Bluetooth broadcast failed
[image]

shubhamdp, 2024-09-12:
This happens when BLEManager does not find the discriminator. Can you please share the `sdkconfig` of the project. I think few options are not enabled.

MaplestoryAlen, 2024-09-13:
Please check the sdkconfig file 
[sdkconfig.txt](https://github.com/user-attachments/files/16986890/sdkconfig.txt)

MaplestoryAlen, 2024-09-13:
> This happens when BLEManager does not find the discriminator. Can you please share the `sdkconfig` of the project. I think few options are not enabled.

Please check the sdkconfig file
[sdkconfig.txt](https://github.com/user-attachments/files/16986896/sdkconfig.txt)

shubhamdp, 2024-09-13:
In your sdkconfig, all the configuration options are correctly configured, so it should work. `chip-factory` namespace is configured to be stored in `fctry` partition. Can you please re-flash the <uuid>-partition.bin to 0xE40000.

To try out at my end, I used few options from your sdkconfig and it worked perfectly fine, I was able to commission the device as well... (I used C3)
```
➜  light git:(main) ✗ cat sdkconfig.defaults.sec_cert
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y
CONFIG_FACTORY_COMMISSIONABLE_DATA_PROVIDER=y
CONFIG_FACTORY_DEVICE_INSTANCE_INFO_PROVIDER=y
CONFIG_NONE_DEVICE_INFO_PROVIDER=y
CONFIG_SEC_CERT_DAC_PROVIDER=y
CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL="fctry"
```

- configure and build firmware
```
idf.py -D SDKCONFIG_DEFAULTS=sdkconfig.defaults.sec_cert set-target esp32c3
idf.py build
```

- generate secure cert and factory partition
```
esp-matter-mfg-tool --discriminator 1083 --passcode 10831022 \
    --product-name 81ma1 --product-id 0x8000 \
    --vendor-name Testvendor --vendor-id 0xFFF1 \
    --hw-ver 1009 --hw-ver-str 1.0.0.9 --serial-num 202408201022 \
    --product-label xz.81ma1.202408201022.AirConditioner \
    --dac-in-secure-cert --target esp32c3 \
    --dac-cert $MATTER_SDK_PATH/credentials/test/attestation/Chip-Test-DAC-FFF1-8000-0001-Cert.pem \
    --dac-key $MATTER_SDK_PATH/test/attestation/Chip-Test-DAC-FFF1-8000-0001-Key.pem \
    --pai -c $MATTER_SDK_PATH/credentials/test/attestation/Chip-Test-PAI-FFF1-8000-Cert.pem \
    -cd Test-CD-0xFFF1-0x8000.der
```

- flash partitions and firmware
```
cd out/fff1_8000/68d843b1-ab12-4025-a704-a224ca5e039a
esptool.py -p /dev/cu.usbmodem1101 write_flash 0xd000 68d843b1-ab12-4025-a704-a224ca5e039a_esp_secure_cert.bin
esptool.py -p /dev/cu.usbmodem1101 write_flash 0x3e0000 68d843b1-ab12-4025-a704-a224ca5e039a-partition.bin
cd -
idf.py flash monitor
```

- commission
```
chip-tool pairing ble-wifi 1 SSID PASSPHRASE 10831022 1083
```

MaplestoryAlen, 2024-09-18:
> CHIP_FACTORY_NAMESPACE_PARTITION_LABEL

Thank you very much. The issue has been resolved

Solution (source code unchanged)
1. Create a new project in the examples to solve the issue of BLE Bluetooth broadcasting
2. Looking at the CSV file, I couldn't find the partition certificate in the distribution network when my esp_decure_cert was not encrypted. Later, I changed it to encrypted, but I don't know why this happened.

closed 2024-09-18 by MaplestoryAlen