espressif/esp-matter #1055: Operational Discovery

simonFloat, 2024-08-15:
How would I perform operational discovery? 
I was look at the _DiscoveryImplPlatform_ class which seems to have this functionality. I'm however unsure if there is a better higher level functionality to perform operational discovery. Moreover I get the error 
`undefined reference to chip::Dnssd::DiscoveryImplPlatform::GetInstance()`, and it seems the file is not included when compiling? Other files in the same directory seems to be included so I am a bit confused.

simonFloat, 2024-08-15:
Node-id 1886217941 might be my google nest hub, as when I plug out my other device only node-id 16 disappear. However this confuses me as I specified a fabric id as filter, and I am pretty sure they are not on the same fabric (google nest hub and chip-tool). Also node-id 16 does not correspond to the node-id I assigned to the device, it should have been id 10.

wqx6, 2024-08-16:
Looks like the Filter doesn't work for minimal mDNS resolver. This might need a fix for upstream repo of connectedhomeip. I am looking into this issue. Thanks for report this issue!

simonFloat, 2024-08-17:
Thanks, can I also ask about the minimal mDNS? as the context for the resolver says:

```
/**
 * Node discovery context class.
 *
 * This class enables multiple clients of the global DNS-SD resolver to start simultaneous
 * discovery operations.
 *
 * An object of this class is shared between a resolver client and the concrete resolver
 * implementation. The client is responsible for allocating the context and passing it to
 * the resolver when initiating a discovery operation. The resolver, in turn, is supposed to retain
 * the context until the operation is finished. This allows the client to release the ownership of
 * the context at any time without putting the resolver at risk of using a deleted object.
 */
```

But in the minimal mDNS resolver _StartDiscovery()_ it says:

```
 // minmdns currently supports only one discovery context at a time so override the previous context
    SetDiscoveryContext(&context);
```

What exactly does this mean / what are the effects of this. Do I have to be careful when I perform my own discovery, so that I do not override a context set by another discovery?

wqx6, 2024-08-19:
> What exactly does this mean / what are the effects of this. Do I have to be careful when I perform my own discovery, so that I do not override a context set by another discovery?

That means the resolver can only handle one browsing at a time. You need to stop current discovery when you start a new discovery.

simonFloat, 2024-08-19:
Is it safe to just stop any discovery and is there a safe way to see if a discovery is ongoing?

wqx6, 2024-08-20:
It is safe the call the stop function even if you don't start any discovery.

wqx6, 2024-08-20:
https://github.com/project-chip/connectedhomeip/pull/35063 Created a PR to fix this issue, could you try with this patch?

simonFloat, 2024-08-20:
> [project-chip/connectedhomeip#35063](https://github.com/project-chip/connectedhomeip/pull/35063) Created a PR to fix this issue, could you try with this patch?

Yes, fixes the issue for me.

I couldn't build when switching to the PR, some errors occurred from some esp32 component delegate, so I wrote your changes in manually to test it. I just wanted to be sure that this might be due to the compliance of the esp32-matter to the version of the upstream repo.

wqx6, 2024-08-21:
> I just wanted to be sure that this might be due to the compliance of the esp32-matter to the version of the upstream repo.

Yes, the esp-matter should be updated if you update the upstream repo.

closed 2024-08-21 by simonFloat