espressif/esp-matter #1057: Could not build using with arduino esp32 component

dzungpv, 2024-08-17:
**Describe the bug**
I want to use with Arduino esp32 as esp idf component but it could not build

**Environment**
- ESP-Matter Commit Id: 652b0e78294d0dcd68a1cedd5470f5ee0e609bf2
- ESP-IDF 5.1.x Commit Id: 446ec3a8990dfcb05a4d9abee250abaf68132cd2
- Arduino Esp32 3.04 Commit Id: 1a42b87fb5a92e25596cff10ebe563a85b847fab
- SoC (eg: ESP32 or ESP32-C3): ESP32

- Host Machine OS: MacOS on M2 
- Host Machine Python version: 3.12.5

**Any additional details**
I use sample light project and add Arduino esp32 component.
Add `CONFIG_FREERTOS_HZ=1000` to sdkconfig.defaults
Add `#include "Arduino.h"` add the top off main.cpp file then clean and build.

The error:
```
from /Users/vietdzung/dev/projects/esp32/thermostat/main/app_driver.cpp:13:
/Users/vietdzung/dev/projects/esp32/thermostat/build/esp-idf/chip/gen/include/system/SystemBuildConfig.h:31:38: fatal error: platform/1/CHIPPlatformConfig.h: No such file or directory
   31 | #define CHIP_PLATFORM_CONFIG_INCLUDE <platform/ESP32/CHIPPlatformConfig.h>
      |      
```

pimpalemahesh, 2024-09-02:
Hi @dzungpv,

The error you're encountering is due to a macro definition in the Arduino-ESP32 component. You can resolve this issue by modifying the target_compile_options() in the CMakeLists.txt file of this component.

Please follow these steps:

1. Open the CMakeLists.txt file located at managed_components/espressif__arduino-esp32/ in any text editor.
2. Find the following lines:
```
target_compile_options(${COMPONENT_TARGET} PUBLIC
    -DARDUINO=10812
    -DARDUINO_${idf_target_for_macro}_DEV
    -DARDUINO_ARCH_ESP32
    -DARDUINO_BOARD="${idf_target_caps}_DEV"
    -DARDUINO_VARIANT="${CONFIG_ARDUINO_VARIANT}"
    -DESP32)

```
3. Change -DESP32 to -DESP32=ESP32.

The updated section should look like this:
```
target_compile_options(${COMPONENT_TARGET} PUBLIC
    -DARDUINO=10812
    -DARDUINO_${idf_target_for_macro}_DEV
    -DARDUINO_ARCH_ESP32
    -DARDUINO_BOARD="${idf_target_caps}_DEV"
    -DARDUINO_VARIANT="${CONFIG_ARDUINO_VARIANT}"
    -DESP32=ESP32)
```
This change should resolve the issue. For more details, you can refer to this thread: https://github.com/espressif/arduino-esp32/issues/7432.

dzungpv, 2024-09-02:
It is OK, problem solve on the arduino-esp32 git.

closed 2024-09-02 by dzungpv