espressif/esp-matter #1058: Access Control - general question

simonFloat, 2024-08-19:
Hello, I have a question regarding Access Control in matter.

I am not quite understanding how devices will get the permissions they need. From my understanding the controller delegates permissions under and after commissioning, but does a device advertise what permissions they need / offer. 

I want to develop a monitoring device, so it needs read permissions of all (most) devices in the fabric. Does this mean that the user will have to manually update every ACL for other devices to allow my monitoring device to read attributes. Or what is the general procedure of updating ACLs, and how much does a user have control to configure specific permissions through their app?

simonFloat, 2024-08-19:
Follow up question:

When I perform a read request from device1 to device2 where I don't have access on device2 access control list, I get an error: `AccessControl: denied` on device2. However the callback to device1 directly terminates to OnDone, but I would have guessed OnError would occur as well. This does not seem to happen, so does a device get notified when perform a request to a device with insufficient permissions?

wqx6, 2024-08-20:
> Hello, I have a question regarding Access Control in matter.
>
> I am not quite understanding how devices will get the permissions they need. From my understanding the controller delegates permissions [...]

Matter provides a simple way to manage the Access Control which is CASE Authenticated Tag (CAT), as a subject DN of NOC.
If the controller/commissioner's NOCs includes a CAT and that CAT is in the ACL of the end-device(The ACL can be initialized with that ACL during commissioning step of AddNOC command). the controller can access all the devices in the Matter Fabric without any additional modification of end-devices' ACL.

wqx6, 2024-08-20:
> Follow up question:
>
> When I perform a read request from device1 to device2 where I don't have access on device2 access control list, I get an error: `AccessControl: denied` on device2. However the [...]

I didn't encounter this issue. I got `E (404688) read_command: Response Failure: Error IM:0x0000057E` when I use the controller to access am attribute which the controller didn't have the privilege. And `Error IM:0x0000057E` is the error `UnsupportedAccess`.

simonFloat, 2024-08-20:
> Matter provides a simple way to manage the Access Control which is CASE Authenticated Tag (CAT) (in SPEC 6.6.2.1.2), as a subject DN of NOC.
> If the controller/commissioner's NOCs include a CAT and that CAT is in the ACL of the end-device(The ACL can be initialized with that CAT during commissioning step of AddNOC command). the controller can access all the devices in the Matter Fabric without any additional modification of end-devices' ACL.

Is it then correctly understood that it is the commissioner that has full control over what CAT my monitoring device will get, if any? Ideally the monitoring device should not be a commissioner or an administrator, so it does not have any rights to manage the ACLs. So will it then fall upon the consumer include the monitoring device in a group / give the permissions to monitor devices. If there a way for my device to 'ask' for certain access?

simonFloat, 2024-08-20:
> I didn't encounter this issue. I got `E (404688) read_command: Response Failure: Error IM:0x0000057E` when I use the controller to access am attribute which the controller didn't have the privilege. [...]

Hm okay, I am running the lighting example on one esp device. On my other esp device I use my own class derived from ReadClient::Callback, just as the read_command component. I have defined the call back functions, and I can just see that the OnAttributeData only get called if I have permission to read the data, otherwise I get no response back. But maybe I have misconfigured something? Is the error received as a OnAttributeData call? in the case of reading attributes of course.

wqx6, 2024-08-27:
> Is it then correctly understood that it is the commissioner that has full control over what CAT my monitoring device will get, if any? Ideally the monitoring device should not be a commissioner or [...]

If you don't want your monitoring device get too high permissions. you need to modify other end-devices's ACL to add an entry with manage or operate privilege CAT, and you can use UpdateNOC command of operationalcredentials cluster to update the NOC of your monitoring device to an NOC with that manage/operate CAT. Currently there is no such way to ask for the access. Does the updateNOC command work for you?

simonFloat, 2024-08-27:
My issue is I can't seem to grasp how it would work for an end user. Is it correctly understood that it would be what ecosystem the user uses, which will be responsible for modifying end-devices ACL? etc. Google Home or HomeKit will have to modify the ACLs to add entries with manage/operate privilege.

wqx6, 2024-08-28:
> which will be responsible for modifying end-devices ACL? etc. Google Home or HomeKit will have to modify the ACLs to add entries with manage/operate privilege.

I think they will modify the ACL to add entries with manage/operate privilege. But currently these ecosystems don't support to configure a client so I am not clear about the detail how they will manage the devices' ACLs.

closed 2025-03-06 by simonFloat