espressif/esp-matter #1061: Why does the newly added content in the ZAP file not respond in the callbacks after adding a new Endpoint?

WOLEN6914183, 2024-08-21:
Using the official ZAP tool, no matter what type of Endpoint I add to the existing light-switch.zap file, The Endpoints 3 and 4 shown in the image below are the ones I added. I have implemented logging in the program to monitor callbacks for Endpoint 3 and Endpoint 4, and I found that home control does not trigger this callback function.---------

void AppDeviceCallbacks::PostAttributeChangeCallback(EndpointId endpointId, ClusterId clusterId, AttributeId attributeId,uint8_t mask, uint8_t type, uint16_t size, uint8_t * value)“

Have I missed any steps? I am using the official light-switch-app project.”
[image]
[image]

jadhavrohit924, 2024-08-28:
@WOLEN6914183 I have tried the same and it is working completely fine. Please tell me what steps you have followed to build the light-switch-app along with the commit ID.

WOLEN6914183, 2024-08-28:
I just solved it. I found that this callback function has an extra mask parameter and just remove it.

closed 2024-08-28 by WOLEN6914183