espressif/esp-matter #1063: How to get power consumption with Matter; watts, volts, amps, etc., etc.

JoseAntonioMG, 2024-08-23:
I have built a device with Esp32c6 and PZEM-004T to get the current and accumulated energy consumption in my house. This device works fine with arduino, however I want to implement Matter + thread technology in this device to connect it to other devices I have already created with esp matter.

I have developed a firmware from examples I have obtained from the internet with the "Electrical Energy Measurement" cluster, however it does not work.

**app_main.cpp**

```
#include "esp_matter_endpoint.h"
#include <esp_err.h>
#include <esp_log.h>
#include <nvs_flash.h>
#include <esp_matter.h>
#include <esp_matter_cluster.h>
#include <esp_matter_console.h>
#include <esp_matter_ota.h>
#include <common_macros.h>
#include <app_priv.h>
#include <app_reset.h>
#include <platform/ESP32/OpenthreadLauncher.h>
#include <app/clusters/diagnostic-logs-server/diagnostic-logs-server.h>
#include <app/server/Server.h>

static const char *TAG = "app_main";
using namespace esp_matter;
using namespace esp_matter::attribute;
using namespace esp_matter::endpoint;
using namespace chip::app::Clusters;
using namespace chip::app::Clusters::DiagnosticLogs;
constexpr auto k_timeout_seconds = 300;
uint16_t light_endpoint_id = 0;
extern uint16_t energy_sensor_endpoint_id = 0;

static void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)
{
  switch (event->Type)
  {
    case chip::DeviceLayer::DeviceEventType::kInterfaceIpAddressChanged:
      ESP_LOGI(TAG, "Interface IP Address changed");
      break;
    case chip::DeviceLayer::DeviceEventType::kCommissioningComplete:
      ESP_LOGI(TAG, "Commissioning complete");
      break;
    case chip::DeviceLayer::DeviceEventType::kFailSafeTimerExpired:
      ESP_LOGI(TAG, "Commissioning failed, fail safe timer expired");
      break;
    case chip::DeviceLayer::DeviceEventType::kCommissioningSessionStarted:
      ESP_LOGI(TAG, "Commissioning session started");
      break;
    case chip::DeviceLayer::DeviceEventType::kCommissioningSessionStopped:
      ESP_LOGI(TAG, "Commissioning session stopped");
      break;
    case chip::DeviceLayer::DeviceEventType::kCommissioningWindowOpened:
      ESP_LOGI(TAG, "Commissioning window opened");
      break;
    case chip::DeviceLayer::DeviceEventType::kCommissioningWindowClosed:
      ESP_LOGI(TAG, "Commissioning window closed");
      break;
    case chip::DeviceLayer::DeviceEventType::kFabricRemoved:
      ESP_LOGI(TAG, "Fabric removed successfully");
      if (chip::Server::GetInstance().GetFabricTable().FabricCount()==0)
      {
        chip::CommissioningWindowManager & commissionMgr = chip::Server::GetInstance().GetCommissioningWindowManager();
        constexpr auto kTimeoutSeconds = chip::System::Clock::Seconds16(k_timeout_seconds);
        if (!commissionMgr.IsCommissioningWindowOpen())
        {
          CHIP_ERROR err = commissionMgr.OpenBasicCommissioningWindow(kTimeoutSeconds, chip::CommissioningWindowAdvertisement::kDnssdOnly);
          if (err!=CHIP_NO_ERROR)
          {
            ESP_LOGE(TAG, "Failed to open commissioning window, err:%" CHIP_ERROR_FORMAT, err.Format());
          }
[... 633 lines cut ...]
I (2181) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000002A's Attribute 0x00000003 is null **********
I (21411) app_driver: === Get electrical data ===
I (41511) app_driver: === Get electrical data ===
I (61611) app_driver: === Get electrical data ===
I (81711) app_driver: === Get electrical data ===
```

As can be seen, the function for obtaining data from the PZEM-004T is running correctly, 

```
I (21411) app_driver: === Get electrical data ===
I (41511) app_driver: === Get electrical data ===
I (61611) app_driver: === Get electrical data ===
I (81711) app_driver: === Get electrical data ===
```

however the method:

```
esp_matter::attribute::update(energy_sensor_endpoint_id, ElectricalPowerMeasurement::Id, ElectricalPowerMeasurement::Attributes::Voltage::Id, &voltaje_value);
```

does not collect the attributes correctly.

Can someone help me build the Matter schematic for obtaining energy data from pzem004t please.

jadhavrohit924, 2024-08-26:
Could you please try to add an error check at `esp_matter::attribute::update`? So that we can debug what is the issue. Please use the `attribute::get_val` API instead of `voltaje_value.type = esp_matter_val_type_t::ESP_MATTER_VAL_TYPE_INT16;`  before updating the attribute to get the correct attribute type because the attribute type of the `Voltage` attribute is int64.

Please try the above suggestion so that we get clear understanding of the issue.

jadhavrohit924, 2024-08-28:
@JoseAntonioMG In your code, you have created a `PowerSource` device type and are updating the attribute of the `ElectricalPowerMeasurement` Cluster. 
1. You have to create an accurate device type first.
2. Add appropriate attributes, that you want to use and then try to update those attributes.

JoseAntonioMG, 2024-08-28:
esp_matter::attribute::update does not produce an error, I have changed the type esp_matter_val_type_t::ESP_MATTER_VAL_TYPE_INT16 to esp_matter_val_type_t::ESP_MATTER_VAL_TYPE_INT64, however I still have the same problem.

I have also tried the following device types:

**electrical_sensor**
```
    electrical_sensor::config_t electrical_sensor_config;
    endpoint_t *energy_sensor_endpoint = electrical_sensor::create(node, &electrical_sensor_config, ENDPOINT_FLAG_NONE, energy_sensor_handle);
    if (energy_sensor_endpoint)
    {
      energy_sensor_endpoint_id = endpoint::get_id(energy_sensor_endpoint);
      ESP_LOGI(TAG, "Power Device PZEM004T. Endpoint: %d", energy_sensor_endpoint_id);
    }
```
**electrical_power_measurement**
```
    cluster::electrical_power_measurement::config_t electrical_sensor_config;
    endpoint_t *energy_sensor_endpoint = cluster::electrical_power_measurement::create(node, &electrical_sensor_config, ENDPOINT_FLAG_NONE, energy_sensor_handle);
    if (energy_sensor_endpoint)
    {
      energy_sensor_endpoint_id = endpoint::get_id(energy_sensor_endpoint);
      ESP_LOGI(TAG, "Power Device PZEM004T. Endpoint: %d", energy_sensor_endpoint_id);
    }
```

**device_energy_management**
```
    device_energy_management::config_t electrical_sensor_config;
    endpoint_t *energy_sensor_endpoint = device_energy_management::create(node, &electrical_sensor_config, ENDPOINT_FLAG_NONE, energy_sensor_handle);
    if (energy_sensor_endpoint)
    {
      energy_sensor_endpoint_id = endpoint::get_id(energy_sensor_endpoint);
      ESP_LOGI(TAG, "Power Device PZEM004T. Endpoint: %d", energy_sensor_endpoint_id);
    }

```

and none of them work properly. Also, I have compiled and flashed the **All_devices_types_app** example, and created the **electrical_power_measurement** device, which is listed in the list of available devices, however it does not solve the problem.

I have the feeling that the devices for power measurements of different devices are not implemented correctly.

jadhavrohit924, 2024-08-28:
@JoseAntonioMG Did you create the `Voltage` attribute on the `electrical_power_measurement` cluster? Since Voltage is an optional attribute in Matter Spec, ESP-Matter doesn’t create it by default. You have to create it in your application explicitly before updating it. Please take a look at [esp-matter-components](https://github.com/espressif/esp-matter/tree/main/components/esp_matter) for better understanding.

JoseAntonioMG, 2024-08-28:
Does this mean I need to create the optional attributes for the Matter SDK?
How can I create the voltage attribute?

JoseAntonioMG, 2024-08-28:
It is possible to construct the attribute in this way:

```
  node::config_t node_config;
  node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);
  if (node!=nullptr)
  {
    on_off_light::config_t on_off_light_config;
    endpoint_t *light_endpoint = on_off_light::create(node, &on_off_light_config, ENDPOINT_FLAG_NONE, light_handle);
    if (light_endpoint)
    {
      light_endpoint_id = endpoint::get_id(light_endpoint);
      ESP_LOGI(TAG, "Light. Endpoint: %d", light_endpoint_id);
    }
    device_energy_management::config_t electrical_sensor_config;
    esp_matter::cluster_t *energy_cluster = device_energy_management::create(node, &electrical_sensor_config, ENDPOINT_FLAG_NONE, energy_sensor_handle);
    energy_sensor_endpoint_id = endpoint::get_id(energy_cluster);
    esp_matter_attr_val_t voltaje_value;
    voltaje_value = esp_matter_invalid(NULL);
    voltaje_value.type = esp_matter_val_type_t::ESP_MATTER_VAL_TYPE_INT64;
    esp_matter::attribute::create(energy_cluster, ElectricalPowerMeasurement::Attributes::Voltage::Id, ATTRIBUTE_FLAG_NONE, voltaje_value);
    ESP_LOGI(TAG, "Power Device PZEM004T. Endpoint: %d", energy_sensor_endpoint_id);
    esp_openthread_platform_config_t config = {.radio_config = ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG(), .host_config = ESP_OPENTHREAD_DEFAULT_HOST_CONFIG(), .port_config = ESP_OPENTHREAD_DEFAULT_PORT_CONFIG(),};
    set_openthread_platform_config(&config);
    err = esp_matter::start(app_event_cb);
    if(err==ESP_OK)
    {
      esp_matter::console::diagnostics_register_commands();
      esp_matter::console::init();
    }
    else
    {
      ESP_LOGE(TAG, "Fallo en el arranque de Matter, err:%d", err);
    }
  }

```

jadhavrohit924, 2024-08-29:
Please go through the docs of [create your own data model](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#defining-your-own-data-model).

JoseAntonioMG, 2024-08-29:
I have created a new Voltage attribute and it works:

```
electrical_power_measurement::config_t electrical_sensor_config;
esp_matter::cluster_t *energy_cluster = electrical_power_measurement::create(main_endpoint, &electrical_sensor_config, CLUSTER_FLAG_SERVER, electrical_power_measurement::feature::direct_current::get_id() | electrical_power_measurement::feature::alternating_current::get_id());
int64_t Voltaje = 0;
electrical_power_measurement::attribute::create_voltage(energy_cluster, Voltaje);
```
However, I need to create the "electrical power measurement" cluster on a new endpoint, independent of "main_endpoint", where I have other clusters.
In this example:

```
node::config_t node_config;
node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);
on_off_light::config_t on_off_light_config;
endpoint_t *light_endpoint = on_off_light::create(node, &on_off_light_config, ENDPOINT_FLAG_NONE, light_handle);
light_endpoint_id = endpoint::get_id(light_endpoint);
electrical_power_measurement::config_t electrical_sensor_config;
esp_matter::cluster_t *energy_cluster = electrical_power_measurement::create(node, &electrical_sensor_config, CLUSTER_FLAG_SERVER, electrical_power_measurement::feature::direct_current::get_id() | electrical_power_measurement::feature::alternating_current::get_id());
energy_endpoint_id = endpoint::get_id(energy_cluster);
int64_t Voltaje = 0;
cluster::electrical_power_measurement::attribute::create_voltage(energy_cluster, Voltaje);
```
"node" is the main node from which all the clusters hang.
In the electrical_power_measurement::create method I cannot specify a handler for the energy_cluster cluster, as I have been able to specify in the on_off_light cluster. This example not work.

I have created another example for a temperature sensor:

```
node::config_t node_config;
node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);
on_off_light::config_t on_off_light_config;
endpoint_t *light_endpoint = on_off_light::create(node, &on_off_light_config, ENDPOINT_FLAG_NONE, light_handle);
light_endpoint_id = endpoint::get_id(light_endpoint);
temperature_sensor::config_t temperature_sensor_config;
endpoint_t *temperature_sensor_endpoint = temperature_sensor::create(node, &temperature_sensor_config, ENDPOINT_FLAG_NONE, temperature_sensor_handle);
temperature_sensor_endpoint_id = endpoint::get_id(temperature_sensor_endpoint);

```
This example does work

jadhavrohit924, 2024-09-02:
> In the electrical_power_measurement::create method I cannot specify a handler for the energy_cluster cluster, as I have been able to specify in the on_off_light cluster. This example not work.

The handler can be set to an endpoint and not the cluster. So please create `ElectricalSensor` device type and create `electrical_power_measurement` cluster on same endpoint.

JoseAntonioMG, 2024-09-02:
@jadhavrohit924 I have created the "ElectricalSensor" endpoint and I have also created the "electrical_power_measurement" cluster, to which I have assigned the endpoint corresponding to ElectricalSensor, but the log shows that the cluster and also the attributes already exist, and therefore they are not created.

```
extern "C" void app_main()
{
  esp_err_t err = ESP_OK;
  nvs_flash_init();
  app_driver_handle_t electrical_sensor_handle = app_driver_PZEM004T_sensor_init();
  app_driver_handle_t light_handle = app_driver_light_init();
  app_driver_handle_t button_handle = app_driver_button_init();
  app_reset_button_register(button_handle);
  node::config_t node_config;
  node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);
  if (node!=nullptr)
  {
    on_off_light::config_t on_off_light_config;
    endpoint_t *light_endpoint = on_off_light::create(node, &on_off_light_config, ENDPOINT_FLAG_NONE, light_handle);
    if (light_endpoint)
    {
      light_endpoint_id = endpoint::get_id(light_endpoint);
      ESP_LOGI(TAG, "Light Endpoint: %d", light_endpoint_id);
    }
    electrical_sensor::config_t electrical_sensor_config;
    endpoint_t *electrical_endpoint = electrical_sensor::create(node, &electrical_sensor_config, ENDPOINT_FLAG_NONE, electrical_sensor_handle);
    if (electrical_endpoint)
    {
        electrical_endpoint_id = endpoint::get_id(electrical_endpoint);
        ESP_LOGI(TAG, "Electrical Sensor Endpoint: %d", electrical_endpoint_id);
    }
    cluster::electrical_power_measurement::config_t measurement_sensor_config;
    esp_matter::cluster_t *energy_cluster = cluster::electrical_power_measurement::create(electrical_endpoint, &measurement_sensor_config, CLUSTER_FLAG_SERVER, cluster::electrical_power_measurement::feature::direct_current::get_id() | cluster::electrical_power_measurement::feature::alternating_current::get_id());
    int64_t Voltaje = 0;
    int64_t Watios = 0;
    int64_t Amperios = 0;
    int64_t WatiosAcumulados = 0;
    cluster::electrical_power_measurement::attribute::create_voltage(energy_cluster, Voltaje);
    cluster::electrical_power_measurement::attribute::create_active_power(energy_cluster, Watios);
    cluster::electrical_power_measurement::attribute::create_active_current(energy_cluster, Amperios);
    cluster::electrical_power_measurement::attribute::create_reactive_power(energy_cluster, WatiosAcumulados);
    esp_openthread_platform_config_t config = {.radio_config = ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG(), .host_config = ESP_OPENTHREAD_DEFAULT_HOST_CONFIG(), .port_config = ESP_OPENTHREAD_DEFAULT_PORT_CONFIG(),};
    set_openthread_platform_config(&config);
    err = esp_matter::start(app_event_cb);
    if(err==ESP_OK)
    {
      esp_matter::console::diagnostics_register_commands();
      esp_matter::console::init();
    }
    else
    {
      ESP_LOGE(TAG, "Fallo en el arranque de Matter, err:%d", err);
    }
  }
  else
  {
    ESP_LOGE(TAG, "Fallo al crear el nodo de Matter");
  }
}
```

This is the log of the device

```
I (23) boot: ESP-IDF v5.2.2-dirty 2nd stage bootloader
I (23) boot: compile time Sep  2 2024 11:21:58
I (24) boot: chip revision: v0.0
I (26) boot.esp32c6: SPI Speed      : 80MHz
I (31) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 4MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
[... 117 lines cut ...]
I (21382) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000004 is 0 **********
I (21382) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000005 is 0 **********
I (21382) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000008 is 0 **********
I (21382) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000009 is 0 **********
I (31282) chip[DL]: bleAdv Timeout : Start slow advertisement
I (31282) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)
I (31282) chip[DL]: Device already advertising, stop active advertisement and restart
I (31282) NimBLE: GAP procedure initiated: stop advertising.
I (31282) NimBLE: GAP procedure initiated: advertise; 
I (31282) NimBLE: disc_mode=2
I (31282) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (31282) NimBLE: 
I (41482) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000004 is 0 **********
I (41482) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000005 is 0 **********
I (41482) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000008 is 0 **********
I (41482) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000009 is 0 **********
I (61582) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000004 is 0 **********
I (61582) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000005 is 0 **********
I (61582) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000008 is 0 **********
I (61582) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000009 is 0 **********
I (81682) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000004 is 0 **********
I (81682) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000005 is 0 **********
I (81682) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000008 is 0 **********
I (81682) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000009 is 0 **********
```

jadhavrohit924, 2024-09-03:
@JoseAntonioMG Yes `electrical_power_measurement` is created by default when you create `electrical_sensor ` so you don’t need to create it again. Please close the issue if it is solved.

JoseAntonioMG, 2024-09-03:
@jadhavrohit924 There are errors when updating the values ​​of the "Electrical Power Measurement" endpoint.

I have removed the code fragment corresponding to the creation of the "Electrical_Power_Measurement" cluster and also the creation of attributes, since if I include them I also get the same errors.

```
extern "C" void app_main()
{
  esp_err_t err = ESP_OK;
  nvs_flash_init();
  app_driver_handle_t electrical_sensor_handle = app_driver_PZEM004T_sensor_init();
  app_driver_handle_t light_handle = app_driver_light_init();
  app_driver_handle_t button_handle = app_driver_button_init();
  app_reset_button_register(button_handle);
  node::config_t node_config;
  node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);
  if (node!=nullptr)
  {
    on_off_light::config_t on_off_light_config;
    endpoint_t *light_endpoint = on_off_light::create(node, &on_off_light_config, ENDPOINT_FLAG_NONE, light_handle);
    if (light_endpoint)
    {
      light_endpoint_id = endpoint::get_id(light_endpoint);
      ESP_LOGI(TAG, "Light Endpoint: %d", light_endpoint_id);
    }
    electrical_sensor::config_t electrical_sensor_config;
    endpoint_t *electrical_endpoint = electrical_sensor::create(node, &electrical_sensor_config, ENDPOINT_FLAG_NONE, electrical_sensor_handle);
    if (electrical_endpoint)
    {
	electrical_endpoint_id = endpoint::get_id(electrical_endpoint);
      	ESP_LOGI(TAG, "Device energy management Endpoint: %d", electrical_endpoint_id);
    }
    esp_openthread_platform_config_t config = {.radio_config = ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG(), .host_config = ESP_OPENTHREAD_DEFAULT_HOST_CONFIG(), .port_config = ESP_OPENTHREAD_DEFAULT_PORT_CONFIG(),};
    set_openthread_platform_config(&config);
    err = esp_matter::start(app_event_cb);
    if(err==ESP_OK)
    {
      esp_matter::console::diagnostics_register_commands();
      esp_matter::console::init();
    }
    else
    {
      ESP_LOGE(TAG, "Fallo en el arranque de Matter, err:%d", err);
    }
  }
  else
  {
    ESP_LOGE(TAG, "Fallo al crear el nodo de Matter");
  }
}

```
This is the function that allows you to update attributes with energy consumption values.

```
static void PZEM004T(void *pvParameter)
{
  struct current_values pzValues;
  esp_matter_attr_val_t voltaje;
  esp_matter_attr_val_t Amperios;
  esp_matter_attr_val_t Watios;
  esp_matter_attr_val_t WatiosAcumulados;
  ESP_LOGI(TAG, "Starting PZEM004T ...");
  while (1==1)
  {
	PzemGetValues(&pzValues);
    voltaje = esp_matter_invalid(NULL);
    voltaje.type = esp_matter_val_type_t::ESP_MATTER_VAL_TYPE_INT64;
    voltaje.val.i64 = pzValues.voltage;
    Amperios = esp_matter_invalid(NULL);
    Amperios.type = esp_matter_val_type_t::ESP_MATTER_VAL_TYPE_INT64;
[... 152 lines cut ...]
I (31282) NimBLE: GAP procedure initiated: advertise; 
I (31282) NimBLE: disc_mode=2
I (31282) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (31282) NimBLE: 
I (41492) chip[ZCL]: 0x421475cc ep 2 clus 0x0000_0090 attr 0x0000_0004 not supported
E (41492) esp_matter_attribute: Error updating Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000004 to matter: 0x86
I (41492) chip[ZCL]: 0x421475cc ep 2 clus 0x0000_0090 attr 0x0000_0005 not supported
E (41492) esp_matter_attribute: Error updating Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000005 to matter: 0x86
I (41492) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000008 is 0 **********
I (41492) chip[ZCL]: 0x421475cc ep 2 clus 0x0000_0090 attr 0x0000_0009 not supported
E (41492) esp_matter_attribute: Error updating Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000009 to matter: 0x86
I (61592) chip[ZCL]: 0x421475cc ep 2 clus 0x0000_0090 attr 0x0000_0004 not supported
E (61592) esp_matter_attribute: Error updating Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000004 to matter: 0x86
I (61592) chip[ZCL]: 0x421475cc ep 2 clus 0x0000_0090 attr 0x0000_0005 not supported
E (61592) esp_matter_attribute: Error updating Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000005 to matter: 0x86
I (61592) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000008 is 0 **********
I (61592) chip[ZCL]: 0x421475cc ep 2 clus 0x0000_0090 attr 0x0000_0009 not supported
E (61592) esp_matter_attribute: Error updating Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000009 to matter: 0x86
I (81692) chip[ZCL]: 0x421475cc ep 2 clus 0x0000_0090 attr 0x0000_0004 not supported
E (81692) esp_matter_attribute: Error updating Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000004 to matter: 0x86
I (81692) chip[ZCL]: 0x421475cc ep 2 clus 0x0000_0090 attr 0x0000_0005 not supported
E (81692) esp_matter_attribute: Error updating Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000005 to matter: 0x86
I (81692) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000008 is 0 **********
I (81692) chip[ZCL]: 0x421475cc ep 2 clus 0x0000_0090 attr 0x0000_0009 not supported
```

jadhavrohit924, 2024-09-04:
@JoseAntonioMG, I suggest you first read the attribute list on that cluster and then add those attributes that are not on your device. Also while updating the attribute please refer to the code to [update the attribute](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_driver.cpp#L100-L103).

RichardPar, 2024-09-22:
@JoseAntonioMG  

Did you manage to get this working?  (and are willing to share your code?)

JoseAntonioMG, 2024-09-23:
@RichardPar, i have managed to get it working but not correctly. I am using the "electrical_sensor" device as described above, but in the device log I get the warnings:

```
W (661) esp_matter_core: Server Cluster 0x00000090 on endpoint 0x0002 already exists. Not creating again.
W (661) esp_matter_core: Attribute 0x0000FFFC on cluster 0x00000090 already exists. Not creating again.
W (661) esp_matter_core: Attribute 0x00000000 on cluster 0x00000090 already exists. Not creating again.
W (661) esp_matter_core: Attribute 0x00000001 on cluster 0x00000090 already exists. Not creating again.
W (661) esp_matter_core: Attribute 0x00000002 on cluster 0x00000090 already exists. Not creating again.
W (661) esp_matter_core: Attribute 0x00000008 on cluster 0x00000090 already exists. Not creating again.
W (661) esp_matter_core: Attribute 0x0000FFFD on cluster 0x00000090 already exists. Not creating again.
W (661) esp_matter_core: Attribute 0x00000008 on cluster 0x00000090 already exists. Not creating again.
```

With the following code in **app_main.cpp**:

```
extern "C" void app_main()
{
  esp_err_t err = ESP_OK;
  nvs_flash_init();
  app_driver_handle_t electrical_sensor_handle = app_driver_PZEM004T_sensor_init();
  app_driver_handle_t light_handle = app_driver_light_init();
  app_driver_handle_t button_handle = app_driver_button_init();
  app_reset_button_register(button_handle);
  node::config_t node_config;
  node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);
  if (node!=nullptr)
  {
    on_off_light::config_t on_off_light_config;
    endpoint_t *light_endpoint = on_off_light::create(node, &on_off_light_config, ENDPOINT_FLAG_NONE, light_handle);
    if (light_endpoint)
    {
      light_endpoint_id = endpoint::get_id(light_endpoint);
      ESP_LOGI(TAG, "Light Endpoint: %d", light_endpoint_id);
    }
    electrical_sensor::config_t electrical_sensor_config;
    endpoint_t *electrical_endpoint = electrical_sensor::create(node, &electrical_sensor_config, ENDPOINT_FLAG_NONE, electrical_sensor_handle);
    if (electrical_endpoint)
    {
      electrical_endpoint_id = endpoint::get_id(electrical_endpoint);
      ESP_LOGI(TAG, "Device energy management Endpoint: %d", electrical_endpoint_id);
    }
    cluster::electrical_power_measurement::config_t measurement_sensor_config;
    esp_matter::cluster_t *electrical_cluster = cluster::electrical_power_measurement::create(electrical_endpoint, &measurement_sensor_config, CLUSTER_FLAG_SERVER, cluster::electrical_power_measurement::feature::direct_current::get_id() | cluster::electrical_power_measurement::feature::alternating_current::get_id());
    int64_t Voltaje = 0;
    int64_t Watios = 0;
    int64_t Amperios = 0;
    int64_t WatiosAcumulados = 0;
    cluster::electrical_power_measurement::attribute::create_voltage(electrical_cluster, Voltaje);
    cluster::electrical_power_measurement::attribute::create_active_power(electrical_cluster, Watios);
    cluster::electrical_power_measurement::attribute::create_active_current(electrical_cluster, Amperios);
    cluster::electrical_power_measurement::attribute::create_reactive_power(electrical_cluster, WatiosAcumulados);
    esp_openthread_platform_config_t config = {.radio_config = ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG(), .host_config = ESP_OPENTHREAD_DEFAULT_HOST_CONFIG(), .port_config = ESP_OPENTHREAD_DEFAULT_PORT_CONFIG(),};
    set_openthread_platform_config(&config);
    err = esp_matter::start(app_event_cb);
    if(err==ESP_OK)
    {
      esp_matter::console::diagnostics_register_commands();
      esp_matter::console::init();
    }
    else
    {
      ESP_LOGE(TAG, "Fallo en el arranque de Matter, err:%d", err);
    }
  }
  else
  {
    ESP_LOGE(TAG, "Fallo al crear el nodo de Matter");
  }
}
[... 191 lines cut ...]
[1727108238.748975][46497:46497] CHIP:IN: SecureSession[0x7f892c006330]: Released - Type:2 LSID:38257
[1727108238.748987][46497:46497] CHIP:FP: Forgetting fabric 0x1
[1727108238.748996][46497:46497] CHIP:TS: Pending Last Known Good Time: 2023-10-14T01:16:48
[1727108238.749065][46497:46497] CHIP:TS: Previous Last Known Good Time: 2023-10-14T01:16:48
[1727108238.749072][46497:46497] CHIP:TS: Reverted Last Known Good Time to previous value
[1727108238.749086][46497:46497] CHIP:CTL: Shutting down the commissioner
[1727108238.749093][46497:46497] CHIP:CTL: Shutting down the controller
[1727108238.749101][46497:46497] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1727108238.749143][46497:46497] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1727108238.749172][46497:46497] CHIP:FP: Shutting down FabricTable
[1727108238.749184][46497:46497] CHIP:TS: Pending Last Known Good Time: 2023-10-14T01:16:48
[1727108238.749236][46497:46497] CHIP:TS: Previous Last Known Good Time: 2023-10-14T01:16:48
[1727108238.749243][46497:46497] CHIP:TS: Reverted Last Known Good Time to previous value
[1727108238.749325][46497:46497] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-yHVugJ)
[1727108238.749495][46497:46497] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1727108238.749513][46497:46497] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1727108238.749523][46497:46497] CHIP:DL: Inet Layer shutdown
[1727108238.749530][46497:46497] CHIP:DL: BLE shutdown
[1727108238.749570][46497:46497] CHIP:DL: System Layer shutdown

```
Although in this case it is an example where the voltage is 0, but I have tested it with an electrical device and it gives me the correct voltage.

**Home Assistant:**
[image]

lboue, 2024-09-24:
@JoseAntonioMG Electrical sensor is not supported yet in HomeAssistant, so the sensor data is not displayed

RichardPar, 2024-09-25:
THanks @JoseAntonioMG  and @lboue  !   Will be staying tuned!

lboue, 2024-09-26:
@JoseAntonioMG 

Beta [release-202410](https://rc.home-assistant.io/blog/2024/09/25/release-202410/#matter) is out. You can try.

jadhavrohit924, 2024-09-26:
Since the main issue is addressed can we close this issue?

JoseAntonioMG, 2024-09-28:
How can I get the accumulated energy in KWatts/hour or Watts/hour?
I am getting:
voltage, current and watts
But I can't get accumulated Watts/hour
I have implemented the cluster:

```
cluster::electrical_energy_measurement::config_t measurement_energy_config;
esp_matter::cluster_t *energy_cluster = cluster::electrical_energy_measurement::create(electrical_endpoint, &measurement_energy_config, CLUSTER_FLAG_SERVER, cluster::electrical_energy_measurement::feature::cumulative_energy::get_id() | cluster::electrical_energy_measurement::feature::imported_energy::get_id() | cluster::electrical_energy_measurement::feature::periodic_energy::get_id());

```

At the same endpoint as electrical_power_measurement, and I have created the corresponding attributes:

```
cluster::electrical_energy_measurement::attribute::create_cumulative_energy_imported(energy_cluster, int64_t(0), 100, 120);
cluster::electrical_energy_measurement::attribute::create_periodic_energy_imported(energy_cluster, int64_t(0), 100, 120);

```

Then in the app_driver.cpp file I added the lines to update the attributes:

```
esp_matter::attribute::update(electrical_endpoint_id, ElectricalEnergyMeasurement::Id, ElectricalEnergyMeasurement::Attributes::CumulativeEnergyImported::Id, &WatiosAcumulados);
esp_matter::attribute::update(electrical_endpoint_id, ElectricalEnergyMeasurement::Id, ElectricalEnergyMeasurement::Attributes::PeriodicEnergyImported::Id, &WatiosAcumulados);

```

However, the following errors appear:

```
I (61542) chip[EM]: <<< [E:53112i S:35998 M:10134001] (S) Msg TX to 2:000000000001B669 [5EE6] [UDP:[FDDE:AD00:BEEF:CAFE:AB3E:6390:CD80:EBF]:59500] --- Type 0001:05 (IM:ReportData)
I (61642) chip[EM]: >>> [E:53112i S:35998 M:244880974 (Ack:10134001)] (S) Msg RX from 2:000000000001B669 [5EE6] --- Type 0001:01 (IM:StatusResponse)
I (61642) chip[IM]: Received status response, status is 0x00
I (61642) chip[EM]: <<< [E:53112i S:35998 M:10134002 (Ack:244880974)] (S) Msg TX to 2:000000000001B669 [5EE6] [UDP:[FDDE:AD00:BEEF:CAFE:AB3E:6390:CD80:EBF]:59500] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (81592) PZEM-004T: CRC check OK for GetValues()
I (81592) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000004 is 234 **********
I (81602) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000005 is 0 **********
I (81602) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000008 is 6 **********
I (81602) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000091's Attribute 0x00000001 is 8 **********
I (81602) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000091's Attribute 0x00000003 is 8 **********
I (81602) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000008 is 6 **********
I (81602) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000004 is 234 **********
I (81602) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000005 is 0 **********
I (81602) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000091's Attribute 0x00000001 is <invalid type: 4> **********
E (81602) esp_matter_attribute: Insufficient space for reading Endpoint 0x0002's Cluster 0x00000091's Attribute 0x00000001: required: 4, max: 2
I (81602) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000091's Attribute 0x00000003 is <invalid type: 4> **********
E (81612) esp_matter_attribute: Insufficient space for reading Endpoint 0x0002's Cluster 0x00000091's Attribute 0x00000003: required: 4, max: 2
I (81612) chip[EM]: <<< [E:53113i S:35998 M:10134003] (S) Msg TX to 2:000000000001B669 [5EE6] [UDP:[FDDE:AD00:BEEF:CAFE:AB3E:6390:CD80:EBF]:59500] --- Type 0001:05 (IM:ReportData)
I (81712) chip[EM]: >>> [E:53113i S:35998 M:244880975 (Ack:10134003)] (S) Msg RX from 2:000000000001B669 [5EE6] --- Type 0001:01 (IM:StatusResponse)
I (81712) chip[IM]: Received status response, status is 0x00
I (81712) chip[EM]: <<< [E:53113i S:35998 M:10134004 (Ack:244880975)] (S) Msg TX to 2:000000000001B669 [5EE6] [UDP:[FDDE:AD00:BEEF:CAFE:AB3E:6390:CD80:EBF]:59500] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (101672) PZEM-004T: CRC check OK for GetValues()
I (101672) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000004 is 234 **********
I (101672) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000005 is 0 **********
I (101672) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000008 is 5 **********
I (101672) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000091's Attribute 0x00000001 is 8 **********
I (101672) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000091's Attribute 0x00000003 is 8 **********
I (101672) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000008 is 5 **********
I (101672) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000004 is 234 **********
I (101682) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000005 is 0 **********
I (101682) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000091's Attribute 0x00000001 is <invalid type: 4> **********
[... cut ...]
I (121752) chip[EM]: <<< [E:53115i S:35998 M:10134007] (S) Msg TX to 2:000000000001B669 [5EE6] [UDP:[FDDE:AD00:BEEF:CAFE:AB3E:6390:CD80:EBF]:59500] --- Type 0001:05 (IM:ReportData)
I (121842) chip[EM]: >>> [E:53115i S:35998 M:244880977 (Ack:10134007)] (S) Msg RX from 2:000000000001B669 [5EE6] --- Type 0001:01 (IM:StatusResponse)
I (121842) chip[IM]: Received status response, status is 0x00
I (121842) chip[EM]: <<< [E:53115i S:35998 M:10134008 (Ack:244880977)] (S) Msg TX to 2:000000000001B669 [5EE6] [UDP:[FDDE:AD00:BEEF:CAFE:AB3E:6390:CD80:EBF]:59500] --- Type 0000:10 (SecureChannel:StandaloneAck)

```

jadhavrohit924, 2024-11-27:
@JoseAntonioMG Are you still facing the issue? If yes, please share the state that you are in and if not, can you please close the issue?

jadhavrohit924, 2024-12-19:
@JoseAntonioMG Any update?

JoseAntonioMG, 2025-01-23:
Hi guys,

I have downloaded matter version 1.4 (connectedhomeip and esp matter) and also updated the espressif esp-idf version to 5.4

I have tried to test the electrical_energy_measurement cluster configuration to create a watts/hour attribute that can be updated through the PZEM004T driver, however it still doesn't work.

I can get the voltage, current and watts consumed attributes at a specific time, but not the watts/hour consumed in a period.

This obtained data can be viewed in Home Assistant, once the device has been commissioned.

Is there a way to configure some cluster and its attributes so that Home Assistant, once the device has been commissioned, can read the watts/hour consumed attribute?

lboue, 2025-05-07:
@JoseAntonioMG 
I have a working example for `solar_power` device type: [app_main.cpp](https://github.com/lboue/esp-matter/blob/9c02643be20c2f0e911b218ae1c9e36261fba258/examples/solar_power/main/app_main.cpp).
The values are static, but it works.
My case is for `ExportedEnergy` feature and `CumulativeEnergyExported `attribute.

[image]

jadhavrohit924, 2025-06-17:
The example for energy measurement is out in all-device-types app please check the latest commit.

jadhavrohit924, 2025-07-25:
Can we close the issue?

lboue, 2026-09-22:
https://github.com/espressif/esp-matter/tree/ff9f07ecb0bf01d7a2db5428149ef16bf601774c/examples/all_device_types_app/main/electrical_measurement