espressif/esp-matter #1064: Setting up desired permissions

simonFloat, 2024-08-26:
How does a controller (not commissioner), get required permissions to operate desired devices? The example of a light-switch to on-off light makes sense. But what if, in my case, the devices needs to be monitoring X amount of devices on-off cluster to read if they are on or not.

How would this be set up with regard to permissions? Would the monitoring device still make use of a on-off client cluster? And is there a way for a device to signal what permissions it desire to have on other devices?

wqx6, 2024-09-03:
During commissioning, the commissioner will send AddNOC command to the end-device. This command has a command field named CaseAdminSubject which will be the subject of the default ACL entry after commissioning. So if the commissioner set this command field to a node ID of CAT range(0xFFFF_FFFD_XXXX_XXXX). All the nodes with the NOC of that CAT will be able to access the end-device. So the only thing that you need to do is to assign an NOC of that CAT to your controller. You do not need to change the ACLs of the end-devices.
[image]

dhrishi, 2024-10-26:
@simonFloat Please close the issue if your question is answered

dhrishi, 2024-12-03:
Closing this now.

closed 2024-12-03 by dhrishi