espressif/esp-matter #1068: Issue with Re-commissioning Thread Devices Using Matter Example Code - Error CHIP:0x000000AC

vishnuprasadvp12345, 2024-08-29:
<0x1b>[0;32mI (31856) chip[EM]: >>> [E:2397i S:16172 M:231697154] (S) Msg RX from  node 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)<0x1b>[0m
<0x1b>[0;32mI (31876) chip[DMG]: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0005<0x1b>[0m
<0x1b>[0;32mI (31886) chip[CTL]: Received NetworkConfig response, networkingStatus=1<0x1b>[0m
<0x1b>[0;31mE (31886) chip[CTL]: Error on commissioning step 'ThreadNetworkSetup': 'Error CHIP:0x000000AC'<0x1b>[0m
<0x1b>[0;32mI (31896) chip[CTL]: Going from commissioning step 'ThreadNetworkSetup' with lastErr = 'Error CHIP:0x000000AC' -> 'Cleanup'<0x1b>[0m
<0x1b>[0;32mI (31916) chip[CTL]: Performing next commissioning step 'Cleanup' with completion status = 'Error CHIP:0x000000AC'<0x1b>[0m
<0x1b>[0;32mI (31926) chip[CTL]: Successfully finished commissioning step 'Cleanup'<0x1b>[0m
<0x1b>[0;32mI (31926) chip[CTL]: Commissioning complete for node ID 0x0000000000000001: Error CHIP:0x000000AC<0x1b>[0m
<0x1b>[0;32mI (31936) pairing_command: Device commissioning Failure: MatterError CHIP:0x000000AC<0x1b>[0m

After this error, if I erase the controller and re-commission, it works fine again, but the same issue occurs on the second attempt.

dhrishi, 2024-08-30:
@vishnuprasadvp12345 Please provide details about how are you exactly trying to re-commission the device. I hope the flow is like -
1. Start to commission a Thread device using <Ecosystem phone app + Hub/ Chip tool + TBR>
2. Commissioning is successful
3. Then how do you start re-commissioning on the same device? Who opens the commissioning window? Which phone app is used?

dhrishi, 2024-09-28:
I am closing this now. Please re-open with the details, if you are still facing the issue

closed 2024-09-28 by dhrishi