espressif/esp-matter #1071: Multiple endpoint with thermostat and on off switch in Homekit not working

dzungpv, 2024-09-01:
In Homekit I can create multiple node in master node, but I read from this #1304 can only add multiple endpoint to a single node.
So I add 1 thermostat and 2 on off switch (to simulate dry and fan mode) like this:

```
    /* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
    node::config_t node_config;

    // node handle can be used to add/modify other endpoints.
    node_t *node = node::create(&node_config, app_matter_attribute_update_cb, app_matter_identification_cb);
    ABORT_APP_ON_FAILURE(node != nullptr, ESP_LOGE(TAG, "Failed to create Matter node"));
    // Add enpoint to the node
    thermostat::config_t thermostat_config;
    thermostat_config.thermostat.local_temperature = 1900;  // 19 degree
    thermostat_config.thermostat.system_mode = 0;           // OFF state
    endpoint_t *thermostat_endpoint = thermostat::create(node, &thermostat_config, ENDPOINT_FLAG_NONE, NULL);

    thermostat_endpoint_id = endpoint::get_id(thermostat_endpoint);
    ESP_LOGI(TAG_MATTER, "Thermostat endpoint_id = %" PRIu32, static_cast<uint32_t>(thermostat_endpoint_id));

    // Switch for dry mode
    on_off_switch::config_t dry_switch_config;
    endpoint_t *dry_endpoint = on_off_switch::create(node, &dry_switch_config, ENDPOINT_FLAG_NONE, NULL);
    ABORT_APP_ON_FAILURE(dry_endpoint != nullptr, ESP_LOGE(TAG, "Failed to create on off switch endpoint"));

    // Add group cluster to the dry switch endpoint
    cluster::groups::config_t groups_config_d;
    cluster::groups::create(dry_endpoint, &groups_config_d, CLUSTER_FLAG_SERVER | CLUSTER_FLAG_CLIENT);

    dry_switch_endpoint_id = endpoint::get_id(dry_endpoint);
    ESP_LOGE(TAG, "Dry Switch created with endpoint_id %d", dry_switch_endpoint_id);

    // Switch for fan mode
    on_off_switch::config_t fan_switch_config;
    endpoint_t *fan_endpoint = on_off_switch::create(node, &fan_switch_config, ENDPOINT_FLAG_NONE, NULL);
    ABORT_APP_ON_FAILURE(dry_endpoint != nullptr, ESP_LOGE(TAG, "Failed to create on off switch endpoint"));

    // Add group cluster to the dry switch endpoint
    cluster::groups::config_t groups_config_f;
    cluster::groups::create(fan_endpoint, &groups_config_f, CLUSTER_FLAG_SERVER | CLUSTER_FLAG_CLIENT);

    fan_switch_endpoint_id = endpoint::get_id(fan_endpoint);
    ESP_LOGE(TAG, "Fan Switch created with endpoint_id %d", fan_switch_endpoint_id); 
```
But when I add it to homekit, only thermostat  device appear, could not see on off switch enywhere

pimpalemahesh, 2024-09-14:
We are working on it.

dzungpv, 2025-03-01:
> We are working on it.

I am back months later, what is the progress?

pimpalemahesh, 2025-03-10:
@dzungpv Your observation is correct—on_off_switch is not supported in HomeKit because it only has a client cluster.

You can refer to the [Supported Device Types in Apple HomeKit](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#supported-device-types-in-apple-home) for more details. _On_Off_Switch is only supported if it has both client and server cluster._

I am not sure about your exact use case, but instead of on_off_switch, you can use a generic switch. Check out the ESP-Matter [Generic Switch Example](https://github.com/espressif/esp-matter/tree/main/examples/generic_switch). You can add a thermostat endpoint there, and it should work as expected.

However, note that only momentary switches are supported in Apple HomeKit, not latching switches. So, you will need to handle the implementation accordingly

dzungpv, 2025-03-10:
> [@dzungpv](https://github.com/dzungpv) Your observation is correct—on_off_switch is not supported in HomeKit because it only has a client cluster.
>
> You can refer to the [Supported Device Types in [...]

The use-case here is Homekit not support Dry mode and Fan only mode in thermostat device type and they also not support new device type: Room Air Conditioner device. So I must use two switch to simulate Fan only mode and dry mode. If I remember correctly. I have try  `On/Off Light`, `Generic Switch` from that time but it not working. I will try current code base.

pimpalemahesh, 2025-03-20:
@dzungpv Have you tested your implementation with the current codebase?

dzungpv, 2025-05-07:
> [@dzungpv](https://github.com/dzungpv) Have you tested your implementation with the current codebase?

It is the same.

dzungpv, 2025-07-08:
@pimpalemahesh I try on_off_light and it work for now, thank you

closed 2025-07-08 by dzungpv