espressif/esp-matter #1072: Wrong switch type for Generic example in README

lboue, 2024-09-01:
**Describe the bug**
Wrong switch type for Generic example in README
https://github.com/espressif/esp-matter/blob/22fad1243ba0121845c8eff648eb57f0a99a13f5/examples/generic_switch/README.md?plain=1#L80

But in the project file it's the other way around:
https://github.com/espressif/esp-matter/blob/22fad1243ba0121845c8eff648eb57f0a99a13f5/examples/generic_switch/main/Kconfig.projbuild#L5

**Environment**
- ESP-Matter Commit Id: 22fad1243ba0121845c8eff648eb57f0a99a13f5
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3):
- Device Logs (Please attach the log file):

- Host Machine OS: 
- Host Machine Python version:

- Commissioner app and versions if present:
- Home hub app and versions if present:
- Commissioner's logs if present:

shripad621git, 2024-09-02:
@lboue , thanks for reporting the issue. We have added the fix.

lboue, 2024-09-02:
Thansk. I think it would also be useful to add a few lines explaining how to change the type with the idf.py menuconfig command.

This would avoid wasting time searching through the code as I had to do.

closed 2024-09-03 by espressif-bot