espressif/esp-matter #1073: Commissioning failed

lboue, 2024-09-01:
**Describe the bug**
A clear and concise description of what the bug is.

Confirmation packets seem to be received, but the commisioning is running down.

```
[22:05:32]I (69552) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
[22:05:35]I (72052) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 16
[22:05:37]I (74562) NimBLE: GATT procedure initiated: indicate; 
[22:05:37]I (74562) NimBLE: att_handle=18
[22:05:37]
[22:05:37]I (74612) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
[22:05:40]E (76902) chip[SVR]: Commissioning failed (attempt 1): 32
[22:05:40]I (76902) chip[BLE]: Releasing end point's BLE connection back to application.
[22:05:40]I (76912) chip[DL]: Closing BLE GATT connection (con 1)
[22:05:40]I (76912) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19
[22:05:40]
[22:05:40]I (76932) chip[DIS]: Updating services using commissioning mode 1
[22:05:40]E (76932) chip[DIS]: Failed to remove advertised services: 3
[22:05:40]I (76932) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3841/15 cm=1 cp=0
[22:05:40]E (76952) chip[DIS]: Failed to advertise commissionable node: 3
[22:05:40]E (76952) chip[DIS]: Failed to finalize service update: 3
[22:05:40]I (76952) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 1 ) indicate = 0
[22:05:40]I (76982) chip[DL]: CHIPoBLE unsubscribe received
[22:05:40]I (76982) chip[DL]: BLE GAP connection terminated (con 1 reason 0x216)
[22:05:40]I (76962) app_main: Commissioning window opened
[22:05:40]I (76992) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
[22:05:40]I (77002) NimBLE: GAP procedure initiated: advertise; 
[22:05:40]I (77012) NimBLE: disc_mode=2
[22:05:40]I (77012) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
```
Could you help me to fix this issue?

**Environment**
- ESP-Matter Commit Id: 22fad1243ba0121845c8eff648eb57f0a99a13f5
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3): ESP32-H2
- Device Logs (Please attach the log file):

- Host Machine OS:
- Host Machine Python version:

- Commissioner app and versions if present: iOS 17.6.1
- Home hub app and versions if present: HomePodOS 17.6.1
- Commissioner's logs if present:

**Any additional details**
Log file
[generic-switch-esp32h2_Commissioning_failed.log](https://github.com/user-attachments/files/16830316/generic-switch-esp32h2_Commissioning_failed.log)

shripad621git, 2024-09-03:
@lboue , can you please confirm that during commissioning using apple-home, have you scanned the qrcode specified in the README [here](https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/README.md).
If not, please use the one to commission the device as the generic switch example uses different qrcode than standard matter qr code due to factory partition flashed.

lboue, 2024-09-03:
Yes, exactly. That's exactly what I did.

lboue, 2024-09-03:
I noticed several errors in the log file. Is there any way to correct them?

[22:04:24]E (932) esp_matter_core: Cannot set bounds because of val type mismatch: expected: 8, min: 15, max: 15
[22:04:24]E (1482) chip[DIS]: Failed to remove advertised services: 3
[22:04:24]E (1502) chip[DIS]: Failed to advertise commissionable node: 3
[22:04:24]E (1512) chip[DIS]: Failed to finalize service update: 3

[22:04:40]E (17192) chip[DL]: ble_gap_terminate() failed: Error CHIP:0x00000048
[22:05:40]E (76902) chip[SVR]: Commissioning failed (attempt 1): 32
[22:05:40]E (76932) chip[DIS]: Failed to remove advertised services: 3
[22:05:40]E (76952) chip[DIS]: Failed to advertise commissionable node: 3
[22:05:40]E (76952) chip[DIS]: Failed to finalize service update: 3
[22:06:15]E (112342) chip[DL]: ble_gap_terminate() failed: Error CHIP:0x00000048

lboue, 2024-09-09:
I noticed that `CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER` is enabled in `sdkconfig.defaults`
https://github.com/espressif/esp-matter/blob/a4739acde12cd3e8c845e449986c7540d1a83307/examples/generic_switch/sdkconfig.defaults#L40-L47

But not in [sdkconfig.defaults.esp32h2](https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/sdkconfig.defaults.esp32h2)

Is this the expected behavior? Should the file `sdkconfig.defaults.esp32h2` be modified?

[image]

shripad621git, 2024-09-10:
@lboue , if you notice the sdkconfig file of your project after setting the target to esp32h2, it has the required options including `CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y`. The `sdkconfig.defaults.esp32h2` has configs specific to esp32h2 or the configs that should be overriden for the values already set in `sdkconfig.defaults `when target is esp32h2. For e.g.: `CONFIG_ESP_SECURE_CERT_DS_PERIPHERAL=n`' in  `sdkconfig.defaults` and it is enabled in `sdkconfig.defaults.esp32h2`. If you have a look at the `sdkconfig` file, it is enabled.

shripad621git, 2024-09-10:
@lboue , when we tried at our end, the commissioning seemed to work fine. Can you please mention about what is the frequency of the commissioning failure at your end?

lboue, 2024-09-10:
> @lboue , when we tried at our end, the commissioning seemed to work fine. Can you please mention about what is the frequency of the commissioning failure at your end?

It fails every time. I've never succeeded in commissioning. I've tested with several C6 and H2 boards.

shripad621git, 2024-09-12:
@lboue , can you please run the command `esptool.py -p <port> read_flash 0x10000 0xc000 /tmp/nvs.bin` and provide the nvs bin.

closed 2024-09-16 by lboue

hvolpers, 2024-12-18:
@shripad621git Hi, I'm facing the same issue with a Waveshare ESP32-H2-DEV-KIT-N4 Board. I tried the Launchpad example and I tried the Espressif Zerocode solution. Both showing the same behaviour.
In my case Launchpad and Zerocode works fine with my ESP32-C6-DEV-KIT-N8 (also Waveshare iirc), I still have to test the matter-sdk example.
I connected one WS2812, not sure if that matters, just trying to provide a full overview :)

Should I open a new issue or is it ok to continue here?