espressif/esp-matter #1074: How to set accessory/device name in Homekit?

dzungpv, 2024-09-03:
I read all the issue but not found a way to set display name of the accessory / node in the Homekit.
basic_information does nothing and the bridged_device_basic_information I could not make it work.
I create a thermostat, a fan and two lights. I try with multiple endpoint in a node and It can add to Homekit, but I could not found a way to set name for light, it show Light and Light2.
I try bridge but could not add more than 2 devices, it show 

> PacketBuffer: pool EMPTY

Seem like bridge is the only way but I could not make it work 

I follow some hint here but not working https://github.com/espressif/esp-matter/issues/201#issuecomment-1416053877
https://github.com/espressif/esp-matter/issues/356#issuecomment-1549936762

I don' know why a basic feature like set the name off a device/node/accessory does not include in the simple way

dzungpv, 2024-09-03:
I add only two light on the bridge, and add label like this:
```
    char fan_mode_name[] = "Fan mode";
    cluster::bridged_device_basic_information::config_t cfg_fan_md_br;
    cluster_t *cluster_fan_md_br = cluster::bridged_device_basic_information::create(aggregator, &cfg_fan_md_br, CLUSTER_FLAG_SERVER);
    cluster::bridged_device_basic_information::attribute::create_node_label(cluster_fan_md_br, fan_mode_name, strlen(fan_mode_name));
```

and 

```
    // dev2->dev->endpoint;  // bridged endpoint for device
    char name[] = "ROOM1 LIGHT1";
    cluster_t *cluster = cluster::get(dev2->dev->endpoint, chip::app::Clusters::BridgedDeviceBasicInformation::Id);
    attribute::create(cluster, chip::app::Clusters::BridgedDeviceBasicInformation::Attributes::NodeLabel::Id, ATTRIBUTE_FLAG_WRITABLE, esp_matter_char_str(name, strlen(name)));
    attribute::create(cluster, chip::app::Clusters::BridgedDeviceBasicInformation::Attributes::ProductName::Id, ATTRIBUTE_FLAG_WRITABLE, esp_matter_char_str(name, strlen(name)));
```

But it still show: "Matter Accessory, 2" and "Matter Accessory, 3", iOS 17.6.1

jadhavrohit924, 2024-09-04:
Hi @dzungpv Please check https://github.com/espressif/esp-matter/issues/1046#issuecomment-2277085835.

herculesp17, 2025-01-20:
@jadhavrohit924 your comment is not helping me at all. I found out that it is fairly easy to change the label of the root node, but I still can't figure out how to change the label of the endpoint.
```
/* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
node::config_t node_config;
snprintf(node_config.root_node.basic_information.node_label, sizeof(node_config.root_node.basic_information.node_label), "Root Label Name");
```

jadhavrohit924, 2025-01-21:
@herculesp17 If you read the node label from chip-tool, you'll get a string that you have set for your node. Ecosystems don’t obey the node label set in the application. This is an issue with the ecosystem, not with the SDK.

herculesp17, 2025-01-29:
@jadhavrohit924 thanks for the hint. I just ask myself if someone want to cert their device and have multiple lights for example as different endpoints they can't name it differently. That is inconvenient. So it seems that is an issue with the matter integration of the Ecosystems.