espressif/esp-matter #1075: How to write value to ElectricalEnergyMeasurement::Attributes::Accuracy::Id

svn-matterize, 2024-09-04:
Trying to write a array value  to the attribute `ElectricalEnergyMeasurement::Attributes::Accuracy::Id`
using esp_matter_array
```C
ElectricalEnergyMeasurement::Structs::MeasurementAccuracyRangeStruct::Type energyAccuracyRanges[] = {
		{ .rangeMin = 0,
		  .rangeMax = 1'000'000'000'000'000, // 1 million Mwh
		  .percentMax = chip::MakeOptional(static_cast<chip::Percent100ths>(500)),
		  .percentMin = chip::MakeOptional(static_cast<chip::Percent100ths>(50)) }
	};

	ElectricalEnergyMeasurement::Structs::MeasurementAccuracyStruct::Type accuracy = {
		.measurementType = chip::app::Clusters::detail::MeasurementTypeEnum::kElectricalEnergy,
		.measured = true,
		.minMeasuredValue = 0,
		.maxMeasuredValue = 1'000'000'000'000'000, // 1 million Mwh
		.accuracyRanges = chip::app::DataModel::List<
			const ElectricalEnergyMeasurement::Structs::MeasurementAccuracyRangeStruct::Type>(
			energyAccuracyRanges)
	};
	arr_val = esp_matter_array((uint8_t *)&accuracy, sizeof(arr_val), 1);
	// err = attribute::set_val(attribute, &arr_val);
	err = attribute::update(Endpoint.kPlug, ElectricalEnergyMeasurement::Id,
				ElectricalEnergyMeasurement::Attributes::Accuracy::Id, &arr_val);
	if (err != ESP_OK) {
		ESP_LOGE(TAG, "Updated EnergyAccuracy failed");
	}

```
Device log during attribute update
```C
I (2862) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000091's Attribute 0x00000000 is <invalid type: 4> **********
```
Chip tool read back 
```
[1725446179.843262][8020:8022] CHIP:DMG: ReportDataMessage =
[1725446179.843366][8020:8022] CHIP:DMG: {
[1725446179.843492][8020:8022] CHIP:DMG: 	AttributeReportIBs =
[1725446179.843806][8020:8022] CHIP:DMG: 	[
[1725446179.843979][8020:8022] CHIP:DMG: 		AttributeReportIB =
[1725446179.844116][8020:8022] CHIP:DMG: 		{
[1725446179.844245][8020:8022] CHIP:DMG: 			AttributeStatusIB =
[1725446179.844411][8020:8022] CHIP:DMG: 			{
[1725446179.844566][8020:8022] CHIP:DMG: 				AttributePathIB =
[1725446179.845094][8020:8022] CHIP:DMG: 				{
[1725446179.845569][8020:8022] CHIP:DMG: 					Endpoint = 0x1,
[1725446179.846278][8020:8022] CHIP:DMG: 					Cluster = 0x91,
[1725446179.846903][8020:8022] CHIP:DMG: 					Attribute = 0x0000_0000,
[1725446179.847421][8020:8022] CHIP:DMG: 				}
[1725446179.847961][8020:8022] CHIP:DMG: 					
[1725446179.848284][8020:8022] CHIP:DMG: 				StatusIB =
[1725446179.848313][8020:8022] CHIP:DMG: 				{
[1725446179.848338][8020:8022] CHIP:DMG: 					status = 0x89 (RESOURCE_EXHAUSTED),
[1725446179.848442][8020:8022] CHIP:DMG: 				},
[1725446179.848470][8020:8022] CHIP:DMG: 				
[1725446179.848557][8020:8022] CHIP:DMG: 			},
[1725446179.848583][8020:8022] CHIP:DMG: 			
[1725446179.848651][8020:8022] CHIP:DMG: 		},
[1725446179.849047][8020:8022] CHIP:DMG: 		
[1725446179.849136][8020:8022] CHIP:DMG: 	],
[1725446179.849159][8020:8022] CHIP:DMG: 	
[1725446179.849173][8020:8022] CHIP:DMG: 	SuppressResponse = true, 
[1725446179.849190][8020:8022] CHIP:DMG: 	InteractionModelRevision = 11
[1725446179.849264][8020:8022] CHIP:DMG: }
[1725446179.849314][8020:8022] CHIP:TOO: Response Failure: IM Error 0x00000589: General error: 0x89 (RESOURCE_EXHAUSTED)
```
while reading the attribute device log is
```
I (13352) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000091's Attribute 0x00000000 is <invalid type: 4> **********
E (13362) esp_matter_attribute: Insufficient space for reading Endpoint 0x0001's Cluster 0x00000091's Attribute 0x00000000: required: 4, max: 2
```

lboue, 2024-09-12:
Any news about this issue?

jadhavrohit924, 2024-09-16:
@svn-matterize Sorry for the delayed response, you cannot use esp_matter_array for storage because esp-matter doesn’t support array attribute storage.
There is API to set Accuracy in the connectedhomeip SDK [SetMeasurementAccuracy](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/electrical-energy-measurement-server/electrical-energy-measurement-server.cpp#L164). You have to use this.

svn-matterize, 2024-09-16:
@jadhavrohit924 , attribute storage you are mentioning `(ATTRIBUTE_FLAG_NONVOLATILE)` nvs storage?
Is it possible to write/set attribute value using esp_matter_array, if the flag is `(ATTRIBUTE_FLAG_NONE)`?
How to form the `struct value to array` using `esp_matter_array` any example.

jadhavrohit924, 2024-09-16:
Independent of attribute quality, storage of arrays and structs are not possible using esp-matter because for such attributes there is delegate implementation in the connectedhomeip SDK. But delegate is missing for ElectricalEnergyMeasurement cluster so you have to reach to the SDK to update/set the attribute of this cluster.

HamzaHajeir, 2024-10-10:
> @svn-matterize Sorry for the delayed response, you cannot use esp_matter_array for storage because esp-matter doesn’t support array attribute storage. There is API to set Accuracy in the [...]

Hi there,

Calling this API emits an error `CHIP_ERROR_INVALID_ARGUMENT`, it seems there's a missing link between EndpointID and [`emberAfGetClusterServerEndpointIndex()`](https://github.com/project-chip/connectedhomeip/blob/86c1c71236cd97f1c3d871c27cd7cc3670638ac8/src/app/util/attribute-storage.cpp#L840) function, as it does not find the corresponding endpoint. 

What's really missing?

HamzaHajeir, 2024-10-13:
> Hi there,
>
> Calling this API emits an error `CHIP_ERROR_INVALID_ARGUMENT`, it seems there's a missing link between EndpointID and [...]

Queued the function call to post initializing time solves the issue.

jadhavrohit924, 2024-11-27:
@svn-matterize Can you please close the issue if it is resolved?

closed 2024-11-28 by svn-matterize