espressif/esp-matter #1076: CHIP Watchdog crash during pairing

law-ko, 2024-09-07:
**Describe the bug**
Watchdog occurred when commissioning via BLE.

**Environment**
- ESP-Matter Commit Id: 15519d13
- ESP-IDF Commit Id: V5.2.1(a322e6bdad4b6675d4597fb2722eea2851ba88cb)
- SoC (eg: ESP32 or ESP32-C3): ESP32-PICO-MINI-02
- Device Logs (Please attach the log file):
```
E (62150) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (62150) task_wdt:  - IDLE0 (CPU 0)
E (62150) task_wdt: Tasks currently running:
E (62150) task_wdt: CPU 0: CHIP
E (62150) task_wdt: CPU 1: IDLE1
E (62150) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x401B8533:0x3FFC00B0 0x4008F645:0x3FFC00E0 0x40008541:0x3FFEFB20 0x4014FF32:0x3FFEFB40 0x401572B5:0x3FFEFB60 0x40138F48:0x3FFEFB80 0x40138F56:0x3FFEFBA0 0x401396A4:0x3FFEFBC0 0x401397AA:0x3FFEFC20 0x401384E5:0x3FFEFC40 0x4013881E:0x3FFEFC60 0x40138851:0x3FFEFCD0 0x40093C9E:0x3FFEFCF0
0x401b8533: task_wdt_timeout_handling at /Users/lawrenceko/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:441
 (inlined by) task_wdt_isr at /Users/lawrenceko/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:515
0x4008f645: _xt_lowint1 at /Users/lawrenceko/esp/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x40008541: ets_delay_us in ROM
0x4014ff32: esp_bt_controller_disable at /Users/lawrenceko/esp/esp-idf/components/bt/controller/esp32/bt.c:1717
0x401572b5: nimble_port_deinit at /Users/lawrenceko/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:216
0x40138f48: chip::DeviceLayer::Internal::BLEManagerImpl::DeinitBLE() at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:995
0x40138f56: chip::DeviceLayer::Internal::BLEManagerImpl::DeinitESPBleLayer() at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:953
0x401396a4: chip::DeviceLayer::Internal::BLEManagerImpl::DriveBLEState() at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:824
0x401397aa: chip::DeviceLayer::Internal::BLEManagerImpl::DriveBLEState(int) at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:1691
0x401384e5: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.ipp:290
0x4013881e: chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:503
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:257
0x40138851: chip::DeviceLayer::PlatformManager::RunEventLoop() at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:403
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /Users/lawrenceko/esp-git/esp-matter/examples/im/outlet/build/esp-idf/chip/../../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:280
0x40093c9e: vPortTaskWrapper at /Users/lawrenceko/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

law-ko, 2024-09-07:
#748

law-ko, 2024-09-19:
@shubhamdp any idea on this?

shubhamdp, 2024-09-20:
@law-ko Is this with PSRAM?

law-ko, 2024-09-20:
> @law-ko Is this with PSRAM?

@shubhamdp Yes, both BLE and Matter is on PSRAM

pimpalemahesh, 2024-09-25:
@law-ko Could you please share the logs since bootup and `sdkconfig` file? 
Also, is it possible to provide the minimum possible patch on the ESP-Matter light example that can help us to reproduce the current issue?

shubhamdp, 2024-10-15:
@law-ko We are unable to reproduce this. Can you please let us know the frequency of occurrence of this.

Please share the `sdkconfig` file for the project.

pimpalemahesh, 2024-10-17:
@law-ko I attempted to reproduce this issue many times but was unable to do so.

Please use this patch to help us narrow down the cause of the crash: 
[esp32(9d81f09).zip](https://github.com/user-attachments/files/17412697/esp32.9d81f09.zip)

1. Unzip the attached file.
2. Replace the libbtdm_app.a file in components/bt/controller/lib_esp32/esp32/.
3. Apply the patch 0001-add-for-bt-sleep-debug.patch to the IDF.

This should assist in further debugging the issue. 

Please enable the debug logging mode and share complete logs.

shubhamdp, 2026-10-15:
Closing, please re-open if still facing an issue

closed 2026-10-15 by shubhamdp