espressif/esp-matter #1077: Light Control Using Occupancy Sensing Cluster by binding

mortezaMoezzifar, 2024-09-08:
### Description:

I would like to request a feature where an Occupancy Sensing device can be easily bound with an On/Off light device using the Binding Cluster in the Matter protocol. This would allow automatic light control based on room occupancy.

### Use Case:

When a person enters a room, the Occupancy Sensing device detects their presence and sends a signal to turn on the light automatically. Similarly, when the room becomes unoccupied, the light should turn off. This behavior should be configurable using the existing clusters and binding mechanism in the Matter protocol.

### Expected Behavior:

- The Occupancy Sensing device should trigger the On/Off Cluster of a light device when occupancy status changes.
- The binding between these devices should be straightforward, possibly via a configuration tool or automated setup script.

### Current Challenges:

Currently, the process to bind an Occupancy Sensing device with a light device is not clearly documented, and it seems to require manual configuration steps that are not user-friendly. There should be an easier way to achieve this functionality.

### Proposed Solution:

Provide an example or a predefined configuration setup in the Matter SDK or the chip-tool that demonstrates how to bind an Occupancy Sensing device to a light device effectively. This will make it easier for developers and integrators to implement such functionality in their smart home projects.

jadhavrohit924, 2024-09-09:
@mortezaMoezzifar We have demonstrated binding with [light](https://github.com/espressif/esp-matter/tree/main/examples/light) and [light_switch](https://github.com/espressif/esp-matter/tree/main/examples/light_switch) examples. Could you please refer to that to create your device?

mortezaMoezzifar, 2024-09-10:
I have used the examples for binding with the switch and light switch, but the clusters are different in this case. The Occupancy Sensing cluster acts as a server in this situation. I have tried several methods, but I have not been successful. Could you please provide more guidance on this scenario or more examples that include binding the Occupancy Sensing cluster with a light device?

jadhavrohit924, 2024-09-12:
You have two devices, right? One is Light and the other is an Occupancy Sensor. The sensor will have an Occupancy Sensing server cluster, and you must use an Occupancy Sensing client on the Light. So that you can bind the Light to the Sensor and in the binding callback you will subscribe to the Sensor by the Light.
Please refer the [binding](https://github.com/espressif/esp-matter/tree/main/examples/light_switch) and [subscription](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html?highlight=subscription#subscribe-commands).
Please note that we don’t get a callback when we are done binding, so you will have to store the fabric+node_id in the nvs when we get the first connect and use that node_id for a subscription. It's an issue with matter SDK, we will fix it.

mortezaMoezzifar, 2024-09-17:
Based on the explanation, I understand that the "Subscription" process takes place on the controller side. Does this mean we should consider the Occupancy node as the controller?
Is this approach correct?
I managed to achieve the result using the following method:
```c++
#include "occupancy_sensor.h"
#include "logger.h"

OccupancySensor::OccupancySensor() {}

OccupancySensor::~OccupancySensor() {}

bool OccupancySensor::init()
{
    esp_matter::node_t *root = system->systemNode;
    esp_matter::endpoint::occupancy_sensor::config_t config_endpoint;
    config_endpoint.occupancy_sensing.occupancy = 0;
    config_endpoint.occupancy_sensing.occupancy_sensor_type = 0;           // 0 = PIR, 1 = Ultrasonic, 2 = PIRAndUltrasonic, 3 = PhysicalContact
    config_endpoint.occupancy_sensing.occupancy_sensor_type_bitmap = 0x00; // Bit 0 = PIR, Bit 1 = Ultrasonic, Bit 2 = PhysicalContact
    // get min max from sensor details TODO
    // ENDPOINT_FLAG_DESTROYABLE|
    uint8_t flags = ENDPOINT_FLAG_DESTROYABLE;
    endpoint = esp_matter::endpoint::occupancy_sensor::create(root, &config_endpoint, flags, nullptr);
    if (!endpoint)
    {
        LOG(LogType::Error, "Failed to create endpoint");
        return false;
    }

    esp_matter::cluster_t *cluster = esp_matter::cluster::get(endpoint, chip::app::Clusters::OccupancySensing::Id);

    // create <PIROccupiedToUnoccupiedDelay> attribute
    uint32_t attribute_id1 = chip::app::Clusters::OccupancySensing::Attributes::PIROccupiedToUnoccupiedDelay::Id;
    esp_matter::attribute_t *attribute1 = esp_matter::attribute::get(cluster, attribute_id1);
    if (!attribute1)
    {
        flags = esp_matter::attribute_flags::ATTRIBUTE_FLAG_NULLABLE |
                esp_matter::attribute_flags::ATTRIBUTE_FLAG_WRITABLE |
                esp_matter::attribute_flags::ATTRIBUTE_FLAG_NONVOLATILE;
        attribute1 = esp_matter::attribute::create(cluster, attribute_id1, flags, esp_matter_nullable_uint16(nullable<uint16_t>()));
        if (!attribute1)
        {
            LOG(LogType::Error, "Failed to create <PIR Occupied To Unoccupied Delay> attribute");
            return false;
        }
    }

    // create <PIRUnoccupiedToOccupiedDelay> attribute
    uint32_t attribute_id2 = chip::app::Clusters::OccupancySensing::Attributes::PIRUnoccupiedToOccupiedDelay::Id;
    esp_matter::attribute_t *attribute2 = esp_matter::attribute::get(cluster, attribute_id2);
    if (!attribute2)
    {
        flags = esp_matter::attribute_flags::ATTRIBUTE_FLAG_NULLABLE |
                esp_matter::attribute_flags::ATTRIBUTE_FLAG_WRITABLE |
                esp_matter::attribute_flags::ATTRIBUTE_FLAG_NONVOLATILE;
        attribute2 = esp_matter::attribute::create(cluster, attribute_id2, flags, esp_matter_nullable_uint16(nullable<uint16_t>()));
        if (!attribute2)
        {
            LOG(LogType::Error, "Failed to create <PIR Unoccupied To Occupied Delay> attribute");
            return false;
        }
    }

    // create <PIRUnoccupiedToOccupiedThreshold> attribute
    uint32_t attribute_id3 = chip::app::Clusters::OccupancySensing::Attributes::PIRUnoccupiedToOccupiedThreshold::Id;
    esp_matter::attribute_t *attribute3 = esp_matter::attribute::get(cluster, attribute_id3);
    if (!attribute3)
    {
        flags = esp_matter::attribute_flags::ATTRIBUTE_FLAG_NULLABLE |
                esp_matter::attribute_flags::ATTRIBUTE_FLAG_WRITABLE |
                esp_matter::attribute_flags::ATTRIBUTE_FLAG_NONVOLATILE;
[... 561 lines cut ...]
        if (req_handle->command_path.mCommandId == chip::app::Clusters::Identify::Commands::Identify::Id)
        {
            if (((char *)req_handle->request_data)[0] != 1)
            {
                LOG(LogType::Error, "Number of parameters error");
                return;
            }
            sprintf(command_data_str, "{\"0:U16\": %ld}",
                    strtoul((const char *)(req_handle->request_data) + 1, NULL, 16));
        }
        else
        {
            LOG(LogType::Error, "Unsupported command");
            return;
        }
    }
    else
    {
        LOG(LogType::Error, "Unsupported cluster");
        return;
    }
    client::interaction::invoke::send_group_request(fabric_index, req_handle->command_path, command_data_str);
}

```

closed 2024-10-12 by mortezaMoezzifar