espressif/esp-matter #1078: Commissioning over Thread, communication over Ethernet afterwards

bramvanneerven, 2024-09-08:
**Describe the question/query that you have**
Hi! I am searching for some guidance on how enabling/disabling certain transports for commissioning/communication at **runtime** should be implemented. As far as I have seen, there are only straightforward options for enabling/disabling transports using the configuration files (which can only be changed at build time).

It would be nice if there was some official documentation, and preferably examples, available that show how this use case can be implemented. I have found https://github.com/espressif/esp-matter/issues/271, but to me it is not clear how the final implementation was achieved.

**Additional context**
I want to use an ESP32C6, which does not have a MAC. Therefore, I will add Ethernet capability using the W5500 and SPI. Since the use of ESP32C6 is quite common for Matter devices, I, and likely many others, would also appreciate it if there was some documentation, and preferably examples, on how to correctly apply a different Ethernet driver in a Matter application. I have found [an example using esp-idf](https://github.com/espressif/esp-idf/blob/master/examples/ethernet/basic/main/ethernet_example_main.c), but I am not sure how this should be integrated into esp-matter.

FHFS, 2024-09-09:
Have you seen the [esp-thread-br example](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html#ethernet-based-thread-border-router). This example has support for the W5500 ethernet. It is not a Matter example but probably will help you figure out how the W5500 should be configured in esp-idf.

A Matter example for this device would be great!

pimpalemahesh, 2024-09-23:
@bramvanneerven The multiple network feature will be supported in Matter 1.4. An example demonstrating this functionality will be available after its release.