espressif/esp-matter #1079: Crash on start when commissioned

law-ko, 2024-09-09:
**Describe the bug**
The ESP32 is commissioned and when boot up it will randomly crash and reboot.

**Environment**
- ESP-Matter Commit Id: `release/v1.2` [066791d](https://github.com/espressif/esp-matter/commit/066791d61fe1802be7f9b6fd925efdcf4cdd516a)
- ESP-IDF Commit Id: v5.2.1
- SoC (eg: ESP32 or ESP32-C3): ESP32-PICO-MINI-02 with PSRAM enabled
- Device Logs (Please attach the log file):
```
I (4488) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000006's Attribute 0x0000FFFC is 1 **********
I (4488) chipâZCLê: Off completed. reset OnTime to  0
I (4498) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000006's Attribute 0x00004001 is 0 **********
I (4508) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000005's Attribute 0x00000003 is 0 **********
I (4518) esp_matter_core: Dynamic endpoint 2 added
I (4518) wifi:<ba-add>idx:0 (ifx:0, 42:98:b5:96:60:22), tid:0, ssn:0, winSize:64
E (4538) chipâDLê: Long dispatch time: 1563 ms, for event type 2
I (4538) chipâDLê: WIFI_EVENT_STA_START
I (4538) chipâDLê: Done driving station state, nothing else to do...
I (4548) SYSTEM: mDNS hostname set to: 4c7525f05d04.local
I (4558) app_main: fctry successfully read
I (4568) app_main: esp_secure_cert successfully read
I (4568) app_main: VID in DAC: 4891, PID in DAC: 32799
I (4568) app_main: DAC VID is Espressif's VID, no need to update DAC
W (4578) SYSTEM: mDNS TXT record set
I (4578) chipâDLê: CHIPoBLE advertising stopped
I (4598) chipâIMê: Resuming 1 subscriptions in 0 seconds
I (4598) chipâDLê: WIFI_EVENT_STA_CONNECTED
I (4598) chipâDLê: WiFi station state change: Connecting -> Connecting_Succeeded
I (4608) chipâDLê: WiFi station state change: Connecting_Succeeded -> Connected
I (4618) chipâDLê: WiFi station interface connected
I (4618) chipâZCLê: WiFiDiagnosticsDelegate: OnConnectionStatusChanged
I (4628) chipâDLê: Done driving station state, nothing else to do...
I (4638) chipâDLê: Updating advertising data
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4014ac75  PS      : 0x00060130  A0      : 0x8014b3c4  A1      : 0x3ffbabf0  
0x4014ac75: npl_freertos_eventq_put at /Users/lawrenceko/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:304

A2      : 0x3ffd895c  A3      : 0x3ffd8a78  A4      : 0x80093eb8  A5      : 0x3fff1bf0  
A6      : 0x00000003  A7      : 0x3ffd8c54  A8      : 0x00000000  A9      : 0x007bee68  
A10     : 0x003fffff  A11     : 0x00000005  A12     : 0x00702ae7  A13     : 0x3ff5f078  
A14     : 0x3ff5f07c  A15     : 0x3ff5f080  SAR     : 0x00000001  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400832e9  LEND    : 0x400832f1  LCOUNT  : 0x00000027  
0x400832e9: esp_timer_impl_get_counter_reg at /Users/lawrenceko/esp/esp-idf/components/esp_timer/src/esp_timer_impl_lac.c:118
0x400832f1: esp_timer_impl_get_counter_reg at /Users/lawrenceko/esp/esp-idf/components/esp_timer/src/esp_timer_impl_lac.c:128

Backtrace: 0x4014ac72:0x3ffbabf0 0x4014b3c1:0x3ffbac30 0x400d67d9:0x3ffbac50 0x40093ca6:0x3ffbac80
0x4014ac72: npl_freertos_eventq_put at /Users/lawrenceko/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:302
0x4014b3c1: ble_npl_eventq_put at /Users/lawrenceko/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:173
 (inlined by) ble_npl_event_fn_wrapper at /Users/lawrenceko/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:653
0x400d67d9: timer_process_alarm at /Users/lawrenceko/esp/esp-idf/components/esp_timer/src/esp_timer.c:453
 (inlined by) timer_task at /Users/lawrenceko/esp/esp-idf/components/esp_timer/src/esp_timer.c:479
0x40093ca6: vPortTaskWrapper at /Users/lawrenceko/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

- Host Machine OS:
- Host Machine Python version:

- Commissioner app and versions if present: Apple Home
- Home hub app and versions if present: Apple HomePod mini
- Commissioner's logs if present:

**Any additional details**
...

law-ko, 2024-09-10:
```
assert failed: xQueueGenericSend queue.c:936 (pxQueue)

Backtrace: 0x40081952:0x3ffbaa50 0x400938c5:0x3ffbaa70 0x40098541:0x3ffbaa90 0x4021171a:0x3ffbabb0 0x4015848e:0x3ffbabf0 0x40158b9d:0x3ffbac30 0x400d653d:0x3ffbac50 0x40093c9e:0x3ffbac80
0x40081952: panic_abort at /Users/lawrenceko/esp/esp-idf/components/esp_system/panic.c:472
0x400938c5: esp_system_abort at /Users/lawrenceko/esp/esp-idf/components/esp_system/port/esp_system_chip.c:93
0x40098541: __assert_func at /Users/lawrenceko/esp/esp-idf/components/newlib/assert.c:81
0x4021171a: xQueueGenericSend at /Users/lawrenceko/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:937 (discriminator 2)
0x4015848e: npl_freertos_eventq_put at /Users/lawrenceko/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:316
0x40158b9d: ble_npl_eventq_put at /Users/lawrenceko/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:173
 (inlined by) ble_npl_event_fn_wrapper at /Users/lawrenceko/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:653
0x400d653d: timer_process_alarm at /Users/lawrenceko/esp/esp-idf/components/esp_timer/src/esp_timer.c:453
 (inlined by) timer_task at /Users/lawrenceko/esp/esp-idf/components/esp_timer/src/esp_timer.c:479
0x40093c9e: vPortTaskWrapper at /Users/lawrenceko/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

```

shubhamdp, 2024-09-12:
@law-ko can you please share the logs since boot, if possible can you enable the debug log level and share the crash logs.

pimpalemahesh, 2024-10-17:
@law-ko I accidentally reproduce this issue. I reviewed my sdkconfig and found out that `CONFIG_SPIRAM=y` and `CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_INTERNAL=y` was enabled. Then I changed memory allocation mode for BT to external and crash was fixed.
```
CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL=y
```

Can you please give it a try and let us know. Also can you please share your sdkconfig.

alberk8, 2025-06-17:
I am having this randomly,  Is it due to running out of internal memory or memory contention.