espressif/esp-matter #1080: BLE-Thread Pairing Failure on nRF52840 DK for Matter Device Commissioning (Error: CHIP:0x00000032)

vishnuprasadvp12345, 2024-09-13:
Issue Description:

I am using the Matter ESP32 controller to commission a Thread device. The commissioning process works fine on the ESP32-C6-based device, but when I try to commission a Nordic nRF52840 DK running the Matter light Thread example, I get a BLE pairing failure with the following error:

chip[BLE]: Failed to cancel connection rc=2<0x1b>
chip[DIS]: Closing all BLE connections
pairing_command: Pairing Failure: Matter-Error CHIP:0x00000032

command on the ESP32 Matter controller to commission the Thread device:
ble-thread 0x03 0e080000000000010000000300000f35060004001fffe00208fc4ce2fc3cb4228d0708fd595fb741bc308d0510043170cb1f8627587c836b640fefdc0f030f4f70656e5468726561642d663439310102f49104100b876026d280697e21a47e1bc95e96110c0402a0f7f8 20202021 3840

wqx6, 2024-09-13:
@vishnuprasadvp12345  Hi, Could you send the the complete logs of the controller, Thread BR, and nordic chip so that we can analyze the reason of this error?

vishnuprasadvp12345, 2024-09-13:
okke

vishnuprasadvp12345, 2024-09-13:
[boar router .txt](https://github.com/user-attachments/files/16992055/boar.router.txt)
[esp controller.txt](https://github.com/user-attachments/files/16992056/esp.controller.txt)
[devices.txt](https://github.com/user-attachments/files/16992057/devices.txt)

wqx6, 2024-09-14:
Hi, I have tested the BLE pairing with the nrf52840 but found no issue. Could you also test the matter pairing with the nrf52840 using chip-tool?

vishnuprasadvp12345, 2024-09-14:
In the case of the ESP-C6 Thread example code, everything works fine. However, with the NRF device, I repeatedly encounter the same error after multiple attempts.

E (1337382) chip[BLE]: Failed to cancel connection, rc=2
I (1337382) chip[DIS]: Closing all BLE connections
I (1337382) pairing_command: Pairing Failure: Matter-Error CHIP:0x00000032
E (1347602) chip[BLE]: GATT discovery failed; status: 13
I (1347602) NimBLE: GAP procedure initiated: terminating connection; conn_handle=1, hci_reason=19
I (1347672) chip[DL]: BLE GAP connection terminated (conn 1, reason 0x216)

vishnuprasadvp12345, 2024-09-18:
I checked the connection of the nRF52840 DK to the Google Hub, and it is working fine. However, I see an error in the ESP controller. How can I solve this error?

<0x1b>[0;31mE (34302) chip[BLE]: Failed to cancel connection rc=2<0x1b>[0m
<0x1b>[0;32mI (34302) chip[DIS]: Closing all BLE connections<0x1b>[0m
<0x1b>[0;32mI (34302) pairing_command: Pairing Failure: Matter-Error CHIP:0x00000032<0x1b>[0m
<0x1b>[0;31mE (44362) chip[BLE]: GATT discovery failed; status:13<0x1b>[0m
<0x1b>[0;32mI (44362) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19
<0x1b>[0m

matter esp controller pairing ble-thread 0x01 0e080000000000010000000300000f35060004001fffe0020843e147d703597acc0708fdde41f81e84340105101f837282db4441eb581031b7e7f1d5bf030f4f70656e5468726561642d3761303301027a030410f63f2ca5c0381ac1aeb4194ab9720d9d0c0402a0f7f8 20202021 3840
any change this command i include

wqx6, 2024-09-20:
Could you tell me the commit of esp-matter you are using for esp-controller? There were an issue for the ble controller in the nimble code and it was fixed in the latest esp-matter. You can try to update the code of HandleRxNotify() https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/nimble/BLEManagerImpl.cpp#L1749-L1756 and  try again.

vishnuprasadvp12345, 2024-09-20:
commit f21874cbfba575d5325c7cafe04bfdda491ed962

ESP-IDF v5.2.2-dirty

vishnuprasadvp12345, 2024-09-20:
[ble error nrf.txt](https://github.com/user-attachments/files/17071487/ble.error.nrf.txt)

giangnv1993, 2024-12-15:
> @wqx6 In the case of the ESP-C6 Thread example code, everything works fine. However, with the NRF Thread device, I repeatedly encounter the same error after multiple attempts. I am working with Thread device commissioning using the ESP controller, and I get this error during that process
> 
> E (1337382) chip[BLE]: Failed to cancel connection, rc=2 I (1337382) chip[DIS]: Closing all BLE connections I (1337382) pairing_command: Pairing Failure: Matter-Error CHIP:0x00000032 E (1347602) chip[BLE]: GATT discovery failed; status: 13 I (1347602) NimBLE: GAP procedure initiated: terminating connection; conn_handle=1, hci_reason=19 I (1347672) chip[DL]: BLE GAP connection terminated (conn 1, reason 0x216) im also using same nrf52840 board

Hi @vishnuprasadvp12345 : not related to this topic, but I'm trying  to build matter over thread. I ran light_switch example and use ble-thread to commissioning, however, commissioning was always failed. could you give me some hints? thank you very much!