espressif/esp-matter #1081: Group cluster 0xfffc and TC_IDM_10_2

david0215, 2024-09-13:
Hi All,
I am preparing a thermostat for the Matter 1.3 certification.
While I do the test harness I have an issue with the case TC_IDM_10_2.

The test harness log shows:
--------------------------------------------------------------------
[MatterTest] 09-13 07:03:00.011 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
        Endpoint: 1,
        Cluster:  4 (0x04) Groups,
        Attribute:65532 (0xfffc)
    problem: Unknown feature with mask 0x80
    spec_location: 

[MatterTest] 09-13 07:03:00.012 INFO 
Problem: ProblemSeverity.ERROR
    test_name: IDM-10.2
    location: 
        Endpoint: 2,
        Cluster:  4 (0x04) Groups,
        Attribute:65532 (0xfffc)
    problem: Unknown feature with mask 0x80
    spec_location: 
--------------------------------------------------------------------

It said the groups cluster fail to pass the test and I don't know how to fix it.

In my program I didn't handle anything about the groups, just created two endpoints, one is thermostat another one is fan
the esp32c3 log shows:
-----------------------------------------------------------------------
I (1652) esp_matter_core: Dynamic endpoint 0 added
I (1652) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000003's Attribute 0x00000001 is 0 **********
I (1672) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000004's Attribute 0x00000000 is 128 **********
I (1682) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000004's Attribute 0x0000FFFC is <invalid type: 0> **********
I (1692) esp_matter_core: Dynamic endpoint 1 added
I (1702) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000003's Attribute 0x00000001 is 0 **********
I (1712) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000004's Attribute 0x00000000 is 128 **********
I (1722) esp_matter_attribute: ********** W : Endpoint 0x0002's Cluster 0x00000004's Attribute 0x0000FFFC is <invalid type: 0> **********
I (1732) esp_matter_core: Dynamic endpoint 2 added
-----------------------------------------------------------------------

How can I make the FFFC become valid and pass the test? Thanks a lot for your help!

shubhamdp, 2024-09-13:
@david0215 we need more information to debug this problem, can you help with the below template

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3):
- Device Logs (Please attach the log file): <please attach device logs if you have them>

- TH version or commit-id of the TH: 
- Commissioner's logs if present: <here, chip-tool logs, i.e. logs from TH will help>

**Any additional details**
...

david0215, 2024-09-13:
Hi Shubamdp,

Thanks for your reply, here is the information:

ESP-IDF: a322e6bdad (HEAD, tag: v5.2.1) change(version): Update version to 5.2.1

ESP-Matter : 67dda26 (HEAD -> main, origin/main, origin/HEAD) Merge branch 'esp_matter/simplification_core_cpp' into 'main'

TH version : origin/v2.10.1+spring2024

SoC: ESP32-C3

esp32c3 log:
[esp32c3_log.txt](https://github.com/user-attachments/files/16992680/esp32c3_log.txt)

TH log:
[UI_Test_Run_2024_09_13_10_40_23.log](https://github.com/user-attachments/files/16992687/UI_Test_Run_2024_09_13_10_40_23.log)

shubhamdp, 2024-09-16:
@david0215 thanks for reporting, can you please try applying this patch and see if it works https://github.com/shubhamdp/esp-matter/commit/50a40b9ac5695d6a2b3390df73897dd2373a3f05

```
curl https://github.com/shubhamdp/esp-matter/commit/50a40b9ac5695d6a2b3390df73897dd2373a3f05.patch | git am
```

david0215, 2024-09-17:
It works! Thank you @shubhamdp !
[image]

It also help to pass the TC_IDM_10_2 test

closed 2024-09-17 by david0215

shubhamdp, 2024-09-17:
@david0215 Thanks for the verification