espressif/esp-matter #1082: When adding a color temperature light under the bridge device, the light brightness cannot be set.I am using the official beidge-app code modification

wolen666, 2024-09-14:
Below is the initialization of my color temperature lamp
/----------------------- onoff attribute---------------------/

DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(onOffAttrs)
DECLARE_DYNAMIC_ATTRIBUTE(OnOff::Attributes::OnOff::Id, BOOLEAN, 1, ATTRIBUTE_MASK_WRITABLE), /* on/off /
DECLARE_DYNAMIC_ATTRIBUTE(OnOff::Attributes::GlobalSceneControl::Id, BOOLEAN, 1, ATTRIBUTE_MASK_WRITABLE), / on/off /
DECLARE_DYNAMIC_ATTRIBUTE(OnOff::Attributes::OnTime::Id, INT16U, 2, ATTRIBUTE_MASK_WRITABLE), / on/off /
DECLARE_DYNAMIC_ATTRIBUTE(OnOff::Attributes::OffWaitTime::Id, INT16U, 2, ATTRIBUTE_MASK_WRITABLE), / on/off /
DECLARE_DYNAMIC_ATTRIBUTE(OnOff::Attributes::StartUpOnOff::Id, ENUM8, 1, ATTRIBUTE_MASK_WRITABLE), / on/off /
DECLARE_DYNAMIC_ATTRIBUTE(OnOff::Attributes::FeatureMap::Id, BITMAP32, 4, ATTRIBUTE_MASK_WRITABLE), / on/off /
DECLARE_DYNAMIC_ATTRIBUTE(OnOff::Attributes::ClusterRevision::Id, INT16U, 2, ATTRIBUTE_MASK_WRITABLE), / on/off */

DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();
/----------------------- levlecontrol attribute---------------------/

DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(LevelcontrolAttrs)
DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::CurrentLevel::Id, INT8U,1,ATTRIBUTE_MASK_WRITABLE),
DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::RemainingTime::Id, INT16U,2,ATTRIBUTE_MASK_WRITABLE),
DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::MinLevel::Id, INT8U,1,ATTRIBUTE_MASK_WRITABLE),
DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::Options::Id, BITMAP8,1,ATTRIBUTE_MASK_WRITABLE),
DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::OnLevel::Id, INT8U,1,ATTRIBUTE_MASK_NULLABLE),
DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::StartUpCurrentLevel::Id, INT8U,1,ATTRIBUTE_MASK_NULLABLE),
DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::ClusterRevision::Id, INT16U,2,ATTRIBUTE_MASK_WRITABLE),
DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::FeatureMap::Id, BITMAP32,4,ATTRIBUTE_MASK_WRITABLE),
DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

/----------------------- levlecontrol command---------------------/

constexpr CommandId LevelLightCommands[] = {
app::Clusters::LevelControl::Commands::MoveToLevel::Id,
app::Clusters::LevelControl::Commands::Move::Id,
app::Clusters::LevelControl::Commands::Step::Id,
app::Clusters::LevelControl::Commands::Stop::Id,
app::Clusters::LevelControl::Commands::MoveToLevelWithOnOff::Id,
app::Clusters::LevelControl::Commands::MoveWithOnOff::Id,
app::Clusters::LevelControl::Commands::StepWithOnOff::Id,
app::Clusters::LevelControl::Commands::StopWithOnOff::Id,
kInvalidCommandId,
};

/-----------------------onoff command----------------------/

constexpr CommandId onOffIncomingCommands[] = {
app::Clusters::OnOff::Commands::Off::Id,
app::Clusters::OnOff::Commands::On::Id,
app::Clusters::OnOff::Commands::Toggle::Id,
app::Clusters::OnOff::Commands::OffWithEffect::Id,
app::Clusters::OnOff::Commands::OnWithRecallGlobalScene::Id,
app::Clusters::OnOff::Commands::OnWithTimedOff::Id,
kInvalidCommandId,
};

/-----------------------endpoint----------------------/

DECLARE_DYNAMIC_CLUSTER_LIST_BEGIN(LightCluster)
DECLARE_DYNAMIC_CLUSTER(OnOff::Id, onOffAttrs,onOffIncomingCommands, nullptr),
DECLARE_DYNAMIC_CLUSTER(LevelControl::Id, LevelcontrolAttrs,LevelLightCommands, nullptr),
DECLARE_DYNAMIC_CLUSTER(ColorControl::Id, ColorcontrolAttrs,temperLightCommands, nullptr),
DECLARE_DYNAMIC_CLUSTER(Descriptor::Id, descriptorAttrs, nullptr, nullptr),
DECLARE_DYNAMIC_CLUSTER(BridgedDeviceBasicInformation::Id, bridgedDeviceBasicAttrs, nullptr, nullptr),
DECLARE_DYNAMIC_CLUSTER_LIST_END;

/-----------------------level read fun----------------------/

EmberAfStatus HandleReadLevelcontrolAttribute(Device * dev, chip::AttributeId attributeId, uint8_t * buffer, uint16_t maxReadLength)
{
ChipLogProgress(DeviceLayer, "HandleReadLevlecontorlAttribute: attrId=%" PRIu32 ", maxReadLength=%u", attributeId, maxReadLength);
if(attributeId == LevelControl::Attributes::CurrentLevel::Id)
{
uint8_t recv = *buffer;
printf("level%d\r\n",recv);
}
else if(attributeId == LevelControl::Attributes::MinLevel::Id)
{
uint8_t recv =1;
dev->LightOn();
memcpy(buffer, &recv, sizeof(recv));
}
else if (attributeId == LevelControl::Attributes::Options::Id)
{
uint8_t recv =1;
dev->LightOn();
memcpy(buffer, &recv, sizeof(recv));
}
else if(attributeId == LevelControl::Attributes::RemainingTime::Id)
{
uint8_t ret =0;
memcpy(buffer, &ret, sizeof(ret));
}
else if(LevelControl::Attributes::StartUpCurrentLevel::Id)
{
uint8_t recv =8;
dev->LightOn();
memcpy(buffer, &recv, sizeof(recv));
}
else if(attributeId ==LevelControl::Attributes::OnLevel::Id)
{
dev->LightOn();
uint8_t ret = 0;
memcpy(buffer, &ret, sizeof(ret));
printf("On level\r\n");

}
else if(attributeId ==LevelControl::Attributes::FeatureMap::Id)
{
    dev->LightOn();
    uint16_t recv =3;
    printf("FeatureMap\r\n");
    memcpy(buffer, &recv, sizeof(recv));
}
else if(attributeId ==LevelControl::Attributes::ClusterRevision::Id)
{
    dev->LightOn();
    uint16_t recv =5;
    printf("FeatureMap\r\n");
    memcpy(buffer, &recv, sizeof(recv));
}
else
{
    return EMBER_ZCL_STATUS_FAILURE;
}

return EMBER_ZCL_STATUS_SUCCESS;
/-----------------------level write fun----------------------/

EmberAfStatus HandleWriteLevelcontrolAttribute(Device * dev, chip::AttributeId attributeId, uint8_t * buffer)
{
ChipLogProgress(DeviceLayer, "Level control write: attrId=%" PRIu32, attributeId);
printf("attribute%lX--值%d\r\n",attributeId,*buffer);
if(attributeId==LevelControl::Attributes::OnLevel::Id)
{
return EMBER_ZCL_STATUS_SUCCESS;
}
else if(attributeId==LevelControl::Attributes::CurrentLevel::Id)
{
dev->SetBrightness(*buffer);
return EMBER_ZCL_STATUS_SUCCESS;
}
return EMBER_ZCL_STATUS_FAILURE;
}

2.The following is the log of the code running on the device. When controlling the brightness at home, only reading the currentlevel attribute is triggered, and this value is 0. I thought it should trigger a write operation but it didn't actually happen.I hope you can help me answer it when you have time. Thank you.

[image]

pimpalemahesh, 2024-10-05:
@wolen666 I’m trying to reproduce this issue.

pimpalemahesh, 2024-10-08:
@wolen666 I tried your changes and found that the write operation is not triggering. 
However, with just these changes, it's difficult to pinpoint the exact problem. Could you please provide a **patch** of all your changes.

wolen666, 2024-10-08:
Sorry, I have solved this problem in the past few days. This requires adding memcpy(buffer, &level, sizeof(level)); after reading currentLevel. This will solve the above problem.

closed 2024-10-08 by wolen666