espressif/esp-matter #1084: Handling of console logging

sdrasti, 2024-09-18:
What is the best approach to filter the different types of console log messages and redirect them into the appropriate slots of a messaging system?

```
I (1513) wifi:<ba-add>idx:0 (ifx:0, d4:24:dd:91:83:05), tid:0, ssn:1, winSize:64
I (1533) chip[ZCL]: Using ZAP configuration...
I (1533) esp_matter_cluster: Cluster plugin init common callback
I (1533) chip[DMG]: AccessControlCluster: initializing
I (1583) esp_matter_core: Dynamic endpoint 0 added
I (1613) chip[DL]: WIFI_EVENT_STA_START
I (1673) NimBLE: GAP procedure initiated: stop advertising.
```

jonsmirl, 2024-09-18:
Check this out
https://github.com/espressif/esp-matter/issues/999

sdrasti, 2024-09-20:
@jonsmirl Thanks!

sdrasti, 2024-09-20:
@jonsmirl Do you know how one could disable CHIP logging or portions of it via Kconfig? I don't find related entries in the esp-matter sdkconfig, but in the connectedhomeip platform-configs there are defines like CONFIG_CHIP_USE_LOGGING, CONFIG_CHIP_DETAIL_LOGGING, CONFIG_CHIP_PROGRESS_LOGGING or CONFIG_CHIP_ERROR_LOGGING. I wonder how to set them within the esp-matter SDK?

jonsmirl, 2024-09-20:
It is possible to control the CHIP logging, however, I have forgotten how to do it. The Espressif people know how, they should post an answer when they wake up.

closed 2024-09-24 by sdrasti