espressif/esp-matter #1085: Using the Matter controller example together with chip-tool on the same network

pavel808, 2024-09-19:
Up until now I have been using chip-tool running on a Linux computer for commissioning my Thread end devices and sending cluster commands etc.

Now I am starting to use the Matter controller example at `esp-matter/examples/controller` which I have running on the ESP32-S3 on the ESP Thread Border Router board.

I can commission a Thread end device successfully with it, it seems with such a command. The commissioning appears successful :  

`matter esp controller pairing ble-thread 1515 0e080000000000010000000300001135060004001fffe00208782e413e1b5a59c00708fd4397689a751a420510386d0687a3427476540f88d18064410d030f4f70656e5468726561642d663662660102f6bf0410f06225d3c5eff0e0eb72e0821067591d0c0402a0f7f8 20202021 3840 `

On main thing I don't understand here : 

If I then have chip-tool running on a different computer, but it wasn't used to commission the Thread end device, should I still now be able to send cluster commands using it, to the previously commissioned device as follows ? : 

`chip-tool any command-by-id 0x131bfc00 0x0 0x2 1515 0x1`

It cannot find the end device. Only when I commission the end device using that chip-tool, am I able to send commands from it. 

I guess this makes sense, as how would chip-tool know what to communicate with initially? How to solve that?  

Another issue i'm having is this. In trying to send cluster commands using the ESP Matter controller as follows : 

`matter esp controller invoke-cmd 1515 1 320601088  0 0 `

```
I (945139) chip[DIS]: Found an existing secure session to [1:00000000000005EB]!
E (945139) esp_matter_client: Failed to convert json string to TLV

```

Thanks in advance.

wqx6, 2024-09-20:
For our controller, it will issue a self-signed Matter Root CA Certificate(RCAC)&Key upon initialization and then issue a Node Operational Certificate(NOC) for itself. So as a Matter node, it will have a NOC chain and be able to issue NOC for other Matter nodes.  During commissioning a Matter End-device, it will receive Certificate Sign Request(CSR) from that Matter End device, issue a NOC for that node, and send that NOC with the RCAC to the end device. So both the controller and the end device have the same RCAC, which means that they are in the same fabric. The controller will used the {compressed-fabric-id}-{node-id} to resolve the address of remote device.

Chip-tool is similar to the esp-controller. So the end-device commissioned by esp-controller is not in the Matter Fabric of chip-tool and the compressed fabric ids for the two fabrics are different. That is the reason why the chip-tool cannot find the end-device.

The command should be `matter esp controller invoke-cmd <node-id> <endpoint-id> <cluster-id> <command-id> <command-data>` and the command-data should be in json format. please refer to [this](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#cluster-commands) to use the `invoke-cmd`

pavel808, 2024-09-20:
> For our controller, it will issue a self-signed Matter Root CA Certificate(RCAC)&Key upon initialization and then issue a Node Operational Certificate(NOC) for itself. So as a Matter node, it will [...]

@wqx6 Thanks a lot for the explanation. 

So in that case, I have to use the same controller which was used in commissioning, for sending the commands? Or what is the way around this?

In the end, my goal is to have a separate commissioner application running as a phone application, but then it will be the ESP Matter controller running on the TBR which should be communicating with all my Thread end devices after they are commissioned . I'm not sure how to solve that.

sayondeep, 2024-10-09:
@pavel808 

> In the end, my goal is to have a separate commissioner application running as a phone application, but then it will be the ESP Matter controller running on the TBR which should be communicating with all my Thread end devices after they are commissioned . I'm not sure how to solve that.

Can you please try out matter_controller example in esp-rainmaker https://github.com/espressif/esp-rainmaker/tree/master/examples/matter/matter_controller