espressif/esp-matter #1092: Something is wrong with the electrical measurement clusters

jonsmirl, 2024-09-24:
I created them like this:
~~~
        electrical_sensor::config_t config;
        power_topology::create(ep, &(config.power_topology), CLUSTER_FLAG_SERVER,
                                power_topology::feature::set_topology::get_id());
        electrical_power_measurement::create(ep, &(config.electrical_power_measurement), CLUSTER_FLAG_SERVER,
                                electrical_power_measurement::feature::alternating_current::get_id());
        electrical_energy_measurement::config_t econfig;
        electrical_energy_measurement::create(ep, &econfig, CLUSTER_FLAG_SERVER,
                                electrical_energy_measurement::feature::imported_energy::get_id() | electrical_energy_measurement::feature::periodic_energy::get_id());
~~~
Then when I read the clusters from a subscription
~~~
I (86300961) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x0000009C's Attribute 0x0000FFFC is 4 **********
I (86300962) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x0000009C's Attribute 0x0000FFFD is 1 **********
I (86300963) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x0000009C's Attribute 0x00000000 is <invalid type: 4> **********
E (86300965) chip[DMG]: Attribute type 0x48 not handled
E (86300967) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_009C, attributeId: 0x0000_0000err = 501
I (86300968) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x0000FFFC is 2 **********
I (86300970) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000000 is 0 **********
I (86300972) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000001 is 0 **********
I (86300973) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000002 is <invalid type: 4> **********
E (86300975) chip[DMG]: Attribute type 0x48 not handled
E (86300975) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0090, attributeId: 0x0000_0002err = 501
I (86300977) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x00000008 is 0 **********
I (86300978) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000090's Attribute 0x0000FFFD is 1 **********
I (86300979) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000091's Attribute 0x0000FFFC is 9 **********
I (86300981) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000091's Attribute 0x00000000 is <invalid type: 4> **********
E (86300982) chip[DMG]: Attribute type 0x48 not handled
E (86300982) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0091, attributeId: 0x0000_0000err = 501
I (86300984) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000091's Attribute 0x0000FFFD is 1 **********
I (86300985) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000091's Attribute 0x00000003 is <invalid type: 4> **********
E (86300986) chip[DMG]: Attribute type 0x48 not handled
E (86300987) chip[DMG]: Fail to retrieve data, roll back and encode status on clusterId: 0x0000_0091, attributeId: 0x0000_0003err = 501

~~~

jadhavrohit924, 2024-09-25:
Please check [esp-matter docs](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/app_guide.html) once. These attributes are of either Structs or Lists, the application will need to implement delegate for such attributes.

lboue, 2025-05-06:
I am also facing issues with electrical_energy_measurement cluster feature-map attribute:
`I (8920) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000091's Attribute 0x0000FFFC is 6 **********`

jadhavrohit924, 2025-06-17:
The example is out in all-device-types app please check the latest commit.

closed 2025-07-25 by jonsmirl