espressif/esp-matter #1093: Can't find the `ThreadNetworkDiagnostics` cluster on Endpoint 0 for multiple_on_off_plugin_units example

lboue, 2024-09-25:
**Describe the question/query that you have**
Can't find the `ThreadNetworkDiagnostics` cluster on Endpoint 0 for [Multiple On Off plugin unit](https://github.com/espressif/esp-matter/tree/main/examples/multiple_on_off_plugin_units) example.

How can I add this missing cluster?

**Additional context**
...
I using a ESP32-H2 board.

shubhamdp, 2024-09-25:
You can find the documentations for creating the cluster and a few examples here: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#clusters

And here is the thread network diagnostic cluster: https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_cluster.h#L160

But, in-a-nutshell this should work
```
// to get the endpoint from node_t
esp_matter::endpoint_t *root_ep = esp_matter::endpoint::get(node, 0); // 0 being the root endpoint

esp_matter::cluster::thread_network_diagnostics::config_t thread_network_diag_config;
esp_matter::cluster_t *thread_network_diag_cluster = esp_matter::cluster::thread_network_diagnostics::create(root_ep, &thread_network_diag_config, CLUSTER_FLAG_SERVER);
```

lboue, 2024-09-25:
Thank you. So I have to implement each `ThreadNetworkDiagnostics` cluster attribute manually?

lboue, 2024-09-25:
Thank you. Now it works.

[image]

shubhamdp, 2024-09-25:
can you please close if problem is resolved!

closed 2024-09-25 by lboue