espressif/esp-matter #1094: attribute update

stefano055415, 2024-09-25:
I created a custom uint16 attribute and set it to 4000.
I updated it whitout error (ESP_OK)
but if I try to read it I still get the old value

cluster_t *test_cluster = cluster::create(endpoint_zero, 0x10000001, CLUSTER_FLAG_SERVER);
    attribute_t *test_attribute = attribute::create(test_cluster, 0x0, ATTRIBUTE_FLAG_WRITABLE | ATTRIBUTE_FLAG_NONVOLATILE, esp_matter_uint16(4000));
    esp_matter_attr_val_t test_attribute_val = esp_matter_invalid(NULL);
    attribute::get_val(test_attribute, &test_attribute_val);
    printf("Attribute value: %d\n", test_attribute_val.val.u16);
    
    esp_matter_attr_val_t test = esp_matter_uint16(5000);
    esp_err_t err = attribute::update(endpoint::get_id(endpoint_zero), 0x10000001, 0x0, &test);   
    printf("Error: %d\n", err);

    esp_matter_attr_val_t test_attribute_val_mod = esp_matter_invalid(NULL);
    attribute::get_val(test_attribute, &test_attribute_val_mod);
    printf("Attribute mod value: %d\n", test_attribute_val_mod.val.u16);

Attribute value: 4000
Error: 1
Attribute mod value: 4000

shubhamdp, 2024-09-26:
@stefano055415 I think you may have updated your attribute before starting the matter stack and thats why you are seeing the stale value.

Can you please update the variable after `esp_matter::start()` and you should get a `5000`.

Or do you have any other use case for this?

stefano055415, 2024-09-26:
@shubhamdp 

Yes, you're right, that was the problem.

Thank you

closed 2024-09-26 by stefano055415