espressif/esp-matter #1095: Device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time

bilalmalik76, 2024-09-25:
Getting this error while commissioning: device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
[image]

lboue, 2024-09-25:
Same issue for me:

```
[22:22:08]I (98750) chip[SC]: Received Sigma1 msg
[22:22:08]I (98760) chip[SC]: CASE matched destination ID: fabricIndex 3, NodeID 0x0000000000000029
[22:22:08]I (98800) chip[EM]: <<< [E:28076r S:0 M:136439739 (Ack:220325966)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[2A01:E0A:2B3:3990:202F:C4C4:BB61:7A01]:56595] --- Type 0000:31 (SecureChannel:CASE_Sigma2) (B:751)
[22:22:08]I (98820) chip[SC]: Sent Sigma2 msg
[22:22:08]I (98820) chip[EM]: >>> [E:34652r S:27836 M:200432397 (Ack:96387416)] (S) Msg RX from 1:00000000DA28A110 [FE0B] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
[22:22:08]I (99400) chip[EM]: >>> [E:28076r S:0 M:220325967 (Ack:136439739)] (U) Msg RX from 0:551205375136E426 [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3) (B:598)
[22:22:08]I (99410) chip[EM]: <<< [E:28076r S:0 M:136439740 (Ack:220325967)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[2A01:E0A:2B3:3990:202F:C4C4:BB61:7A01]:56595] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:26)
[22:22:08]I (99420) chip[SC]: Received Sigma3 msg
[22:22:08]E (99440) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
[22:22:09]I (99540) chip[EM]: <<< [E:28076r S:0 M:136439741 (Ack:220325967)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[2A01:E0A:2B3:3990:202F:C4C4:BB61:7A01]:56595] --- Type 0000:40 (SecureChannel:StatusReport) (B:34)
[22:22:09]I (99560) chip[SC]: SecureSession[0x408189f8, LSID:27839]: State change 'kEstablishing' --> 'kActive'
[22:22:09]I (99570) chip[IN]: CASE Session established to peer: <000000000001B669, 3>
[22:22:09]I (99640) chip[EM]: >>> [E:28077r S:27839 M:16763786] (S) Msg RX from 3:000000000001B669 [7C78] --- Type 0001:08 (IM:InvokeCommandRequest) (B:59)
[22:22:09]I (99650) esp_matter_command: Received command 0x00000004 for endpoint 0x0000's cluster 0x00000030
[22:22:09]I (99650) chip[FS]: GeneralCommissioning: Received CommissioningComplete
```

jadhavrohit924, 2024-09-26:
Yes, by default we don’t support real-time on the device. If you need you can use SNTP for time synchronization. We have APIs to do that, as well as an example that demonstrates this.
https://github.com/project-chip/connectedhomeip/blob/master/examples/platform/esp32/time/TimeSync.h
https://github.com/project-chip/connectedhomeip/blob/master/examples/energy-management-app/esp32/main/main.cpp

jadhavrohit924, 2024-09-26:
Please do not use `pool.ntp.org` as your server, we have used it just for demonstrating.

bilalmalik76, 2024-09-30:
Which server should be used if not pool.ntp.org?

jadhavrohit924, 2024-10-01:
Ideally, it would be best if you use your NTP server.
> Is using time synchronization mandatory if not how can skip it?
Time synchronization is not mandatory. If you don’t need real-time, don’t initialize the time synchronization

dhrishi, 2024-10-05:
Closing as the question is answered.

closed 2024-10-05 by dhrishi