espressif/esp-matter #1096: Can't add additional sensors to commissioned deivce

barsa4ever, 2024-09-26:
I build a weather station with ESP32C6 and BME680.
As starting point I added temperature and humidity. I flashed a firmware and was able to add device to Apple Home.
Now I added air pressure sensor to my code. I rebuilt and flush a firmware but I can't see air pressure in my Apple Home.
I removed sensors and device itself from keychain and added it. It went through onboarding screens but showed me that only two sensors are available for this device.
However I can see in logs that humidity sensor was added to matter and that value is updated:
```esp_matter_attribute: ********** W : Endpoint 0x0003's Cluster 0x00000403's Attribute 0x00000000 is 1014```
How can I reset/update esp32 so it will be recognized as a new device with new set of sensors by Apple Home?

shubhamdp, 2024-09-26:
@barsa4ever you can find the list of supported device types in Apple Home here: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#supported-device-types-in-apple-home

At this moment pressure sensor device type is not supported in Apple Home and hence won't be recognized by Home app.

barsa4ever, 2024-09-28:
actual list of characteristics supported by apple homekit: https://developer.apple.com/documentation/homekit/characteristic-types

closed 2024-09-28 by barsa4ever

shubhamdp, 2024-09-28:
> actual list of characteristics supported by apple homekit: https://developer.apple.com/documentation/homekit/characteristic-types

This is the list of devices supported through Homekit and not the Matter.
What you are trying is, creating a pressure sensor. This device type is supported in Matter but not in Apple Home.

barsa4ever, 2024-09-29:
@shubhamdp thanks, I got it.
I just left a comment for future references re which Matter devices/sensors are supported by Apple.