espressif/esp-matter #1099: smoke_co_alarm and illuminance_measurement

stefano055415, 2024-09-30:
I'm trying to create two sensors as a bridge device and display them on the Apple Home app.

The brightness sensor is seen but the value does not match the one I set

light_sensor::config_t light_sensor_conf;
err = light_sensor::add(ep, &light_sensor_conf);

esp_matter_attr_val_t current_ambient_light_level_val = esp_matter_nullable_uint16(20000);
    attribute::update(endpoint_id, IlluminanceMeasurement::Id, IlluminanceMeasurement::Attributes::MeasuredValue::Id, &current_ambient_light_level_val);

I expected to see 200 lux, but I see 0.00001

The smoke sensor is not displayed

endpoint::smoke_co_alarm::config_t smoke_co_alarm_conf;
err = endpoint::smoke_co_alarm::add(ep, &smoke_co_alarm_conf);

sayondeep, 2024-10-01:
@stefano055415 

> esp_matter_attr_val_t current_ambient_light_level_val = esp_matter_nullable_uint16(20000);
attribute::update(endpoint_id, IlluminanceMeasurement::Id, IlluminanceMeasurement::Attributes::MeasuredValue::Id, &current_ambient_light_level_val);

Please make sure that the above lines of code are after `esp_matter::start` as changes to the attributes only takes effect after the matter_thread is initialized as per configurations specified.

sayondeep, 2024-10-15:
@stefano055415 
Please check this patch:
[add_light_sensor.txt](https://github.com/user-attachments/files/17375431/add_light_sensor.txt)

I was able to get updated light sensor attribute value
[image]

sayondeep, 2024-11-11:
@stefano055415 Please feel free to close the issue if it is resolved.

dhrishi, 2024-11-13:
Closing this now

closed 2024-11-13 by dhrishi