espressif/esp-matter #1101: Unable to read attribute value from matter esp controller

pavel808, 2024-09-30:
I have an ESP32-C6 Thread end device which is reading temperature and voltages and updating attributes. 

I have attached my app_main source code file which shows how the attributes are getting updated etc. 

When I issue the following matter controller command to read the temperature from the device, this works fine and the temperature value gets displayed : 

`matter esp controller read-attr 1666 1 1026 0`

However, when trying to read my voltage values as follows : 

`matter esp controller read-attr 144 1 1026 4`
`matter esp controller read-attr 144 1 1026 11`

I get the following error back on the Matter controller stating that it doesn't know how to log the attribute value : 

```
I (4681896) chip[DIS]: Found an existing secure session to [1:0000000000000682]!
I (4681896) chip[EM]: <<< [E:21637i S:20710 M:69950749] (S) Msg TX to 1:0000000000000682 [8014] [UDP:[FD26:4944:1C37:1:E0BD:E60E:A609:C60A]:5540] --- Type 0001:02 (IM:ReadRequest)
Done
> I (4682006) chip[EM]: >>> [E:21637i S:20710 M:16836634 (Ack:69950749)] (S) Msg RX from 1:0000000000000682 [8014] --- Type 0001:05 (IM:ReportData)
I (4682006) chip[TOO]: Endpoint: 1 Cluster: 0x0000_0090 Attribute 0x0000_000B DataVersion: 876342339
I (4682006) chip[TOO]:   Don't know how to log atribute value
I (4682006) read_command: read done
I (4682006) chip[EM]: <<< [E:21637i S:20710 M:69950750 (Ack:16836634)] (S) Msg TX to 1:0000000000000682 [8014] [UDP:[FD26:4944:1C37:1:E0BD:E60E:A609:C60A]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)

```

Here is what happens on my end device wen the command is issued : 

```
I (2571704) chip[EM]: >>> [E:21638r S:15999 M:69950751] (S) Msg RX from 1:000000000001B669 [8014] --- Type 0001:02 (IM:ReadRequest)
I (2571704) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000090's Attribute 0x0000000B is 1 **********
I (2571724) chip[EM]: <<< [E:21638r S:15999 M:16836635 (Ack:69950751)] (S) Msg TX to 1:000000000001B669 [8014] [UDP:[FD26:4944:1C37:1:35C9:E685:E766:D812]:5580] --- Type 0001:05 (IM:ReportData)
I (2571804) chip[EM]: >>> [E:21638r S:15999 M:69950752 (Ack:16836635)] (S) Msg RX from 1:000000000001B669 [8014] --- Type 0000:10 (SecureChannel:StandaloneAck)
```

sayondeep, 2024-10-01:
@pavel808 
I am able to fetch voltage correctly using `matter esp controller read-attr 0x7283 1 144 11` 0x7283 being the node-id.

```
> matter esp controller read-attr 0x7283 1 144 11

I (309954) chip[DIS]: Found an existing secure session to [1:0000000000007283]!
I (309964) chip[DMG]: 0 data version filters provided, 0 not relevant, 0 encoded, 0 skipped due to lack of space
I (309974) chip[EM]: <<< [E:5458i S:25178 M:117799969] (S) Msg TX to 1:0000000000007283 [A3EB] [UDP:192.168.0.139:5540] --- Type 0001:02 (IM:ReadRequest) (B:51)
Done
> I (310134) chip[EM]: >>> [E:5458i S:25178 M:72517627 (Ack:117799969)] (S) Msg RX from 1:0000000000007283 [A3EB] --- Type 0001:05 (IM:ReportData) (B:70)
I (310154) chip[TOO]: Endpoint: 1 Cluster: 0x0000_0090 Attribute 0x0000_000B DataVersion: 2182737241
I (310154) chip[TOO]:   RMSVoltage: 0
I (310164) read_command: read done
I (310164) chip[EM]: <<< [E:5458i S:25178 M:117799970 (Ack:72517627)] (S) Msg TX to 1:0000000000007283 [A3EB] [UDP:192.168.0.139:5540] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
```

```
> matter esp controller read-attr 0x7283 1 144 4

I (394884) chip[DIS]: Found an existing secure session to [1:0000000000007283]!
I (394884) chip[DMG]: 0 data version filters provided, 0 not relevant, 0 encoded, 0 skipped due to lack of space
I (394894) chip[EM]: <<< [E:5459i S:25178 M:117799971] (S) Msg TX to 1:0000000000007283 [A3EB] [UDP:192.168.0.139:5540] --- Type 0001:02 (IM:ReadRequest) (B:51)
Done
> I (395024) chip[EM]: >>> [E:5459i S:25178 M:72517628 (Ack:117799971)] (S) Msg RX from 1:0000000000007283 [A3EB] --- Type 0001:05 (IM:ReportData) (B:70)
I (395044) chip[TOO]: Endpoint: 1 Cluster: 0x0000_0090 Attribute 0x0000_0004 DataVersion: 2182737241
I (395044) chip[TOO]:   Voltage: 0
I (395054) read_command: read done
I (395054) chip[EM]: <<< [E:5459i S:25178 M:117799972 (Ack:72517628)] (S) Msg TX to 1:0000000000007283 [A3EB] [UDP:192.168.0.139:5540] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
```

pavel808, 2024-10-01:
@sayondeep ok that's strange.

I tried it again now and I get this : 

 ```
matter esp controller read-attr 1666 1 144 11

I (55496) chip[DIS]: Resolving F55262D93AE58014:0000000000000682 ...
Done
> I (55696) chip[DIS]: Checking node lookup status after 201 ms
I (58526) chip[DIS]: Node ID resolved for F55262D93AE58014:0000000000000682
I (58526) chip[DIS]: UDP:[FD26:4944:1C37:1:E0BD:E60E:A609:C60A%st1]:5540: new be                                                 st score: 5
I (58526) chip[DIS]: Checking node lookup status after 3039 ms
I (58536) chip[SC]: Initiating session on local FabricIndex 1 from 0x00000000000                                                 1B669 -> 0x0000000000000682
I (58596) chip[EM]: <<< [E:49294i S:0 M:40479249] (U) Msg TX to 0:00000000000000                                                 00 [0000] [UDP:[FD26:4944:1C37:1:E0BD:E60E:A609:C60A]:5540] --- Type 0000:30 (Se                                                 cureChannel:CASE_Sigma1)
I (58606) chip[SC]: Sent Sigma1 msg
I (58776) chip[EM]: >>> [E:49294i S:0 M:27123418 (Ack:40479249)] (U) Msg RX from                                                  0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (58926) chip[EM]: >>> [E:49294i S:0 M:27123419 (Ack:40479249)] (U) Msg RX from                                                  0:0000000000000000 [0000] --- Type 0000:31 (SecureChannel:CASE_Sigma2)
I (58936) chip[EM]: <<< [E:49294i S:0 M:40479250 (Ack:27123419)] (U) Msg TX to 0                                                 :0000000000000000 [0000] [UDP:[FD26:4944:1C37:1:E0BD:E60E:A609:C60A]:5540] --- T                                                 ype 0000:10 (SecureChannel:StandaloneAck)
I (58936) chip[SC]: Received Sigma2 msg
E (59106) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.                                                   Falling back to Last Known Good UTC Time
I (59876) chip[EM]: <<< [E:49294i S:0 M:40479251 (Ack:27123419)] (U) Msg TX to 0                                                 :0000000000000000 [0000] [UDP:[FD26:4944:1C37:1:E0BD:E60E:A609:C60A]:5540] --- T                                                 ype 0000:32 (SecureChannel:CASE_Sigma3)
I (59876) chip[SC]: Sent Sigma3 msg
E (59886) chip[DL]: Long dispatch time: 953 ms, for event type 3
I (60096) chip[EM]: >>> [E:49294i S:0 M:27123420 (Ack:40479251)] (U) Msg RX from                                                  0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (60246) chip[EM]: >>> [E:49294i S:0 M:27123421 (Ack:40479251)] (U) Msg RX from                                                  0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (60246) chip[SC]: Success status report received. Session was established
I (60256) chip[SC]: SecureSession[0x3c2113a8, LSID:13864]: State change 'kEstabl                                                 ishing' --> 'kActive'
I (60266) chip[EM]: <<< [E:49295i S:13864 M:65343025] (S) Msg TX to 1:0000000000                                                 000682 [8014] [UDP:[FD26:4944:1C37:1:E0BD:E60E:A609:C60A]:5540] --- Type 0001:02                                                  (IM:ReadRequest)
I (60266) chip[EM]: <<< [E:49294i S:0 M:40479252 (Ack:27123421)] (U) Msg TX to 0                                                 :0000000000000000 [0000] [UDP:[FD26:4944:1C37:1:E0BD:E60E:A609:C60A]:5540] --- T                                                 ype 0000:10 (SecureChannel:StandaloneAck)
I (60366) chip[EM]: >>> [E:49295i S:13864 M:206507511 (Ack:65343025)] (S) Msg RX                                                  from 1:0000000000000682 [8014] --- Type 0001:05 (IM:ReportData)
I (60366) chip[TOO]: Endpoint: 1 Cluster: 0x0000_0090 Attribute 0x0000_000B Data                                                 Version: 3226250552
I (60366) chip[TOO]:   Don't know how to log atribute value
I (60366) read_command: read done
I (60376) chip[EM]: <<< [E:49295i S:13864 M:65343026 (Ack:206507511)] (S) Msg TX                                                  to 1:0000000000000682 [8014] [UDP:[FD26:4944:1C37:1:E0BD:E60E:A609:C60A]:5540]                                                  --- Type 0000:10 (SecureChannel:StandaloneAck)

```

sayondeep, 2024-10-01:
@pavel808 The above shared result were from esp-matter `commit-id: commit 9350d9d5f948d3b7c61c8659c4d6990d0ff00ea4 ` 
Can you please pull the latest changes of esp-matter if you are on older commit.

pavel808, 2024-10-01:
> @pavel808 The above shared result were from esp-matter `commit-id: commit 9350d9d5f948d3b7c61c8659c4d6990d0ff00ea4 ` Can you please pull the latest changes of esp-matter if you are on older commit.

That's a good point. I pulled the latest changes for esp-matter and all is good now:) Thanks for your help.

closed 2024-10-01 by pavel808