espressif/esp-matter #1103: max dynamic endpoint

stefano055415, 2024-10-02:
I'm creating a bridge that adds many devices.

By modifying the CONFIG_ESP_MATTER_MAX_DYNAMIC_ENDPOINT_COUNT configuration I can add up to 140 devices

If I increase the value again > 140 I get this error:
esp-aes: Failed to allocate memory

shubhamdp, 2024-10-07:
@stefano055415 I would need more details to debug the issue. I'm trying this on ESP32-C3 [blemesh_bridge](http://github.com/espressif/esp-matter/tree/main/examples/blemesh_bridge) with different values for `CONFIG_ESP_MATTER_MAX_DYNAMIC_ENDPOINT_COUNT`, and here are my observations.

- esp-idf : v5.3.1
- esp-matter: 42b6412e011c6d0fea431123cd77a3b138e3ae4d (I have a patch on top to fix some ble-mesh issues)

| Endpoint count | Remark|
|-|-|
| 16, 64 | works fine, able to commission the bridge |
| 140, 150 | crash on boot during ble-init |
| 100 | threading related crash on boot |

Can you please help with
- esp-idf version
- esp-matter version
- SoC used
- PSRAM enabled/disabled?

Can you elaborate a bit more on your workflow when you hit this problem.
eg:
- you build the app with 140 and it just works and with 141st it does not.
-  Build successful, flash, and you see this problem.
- or after commissioning
- or after you start adding the 140 bridged devices to the Matter bridge and when you add the 141st you see this error?

stefano055415, 2024-10-07:
@shubhamdp,

- esp-matter docker release 1.3 (35765f30e5b2)
- PSRAM enabled
- esp32s3

I create a bridge without any device, but only with the configuration set to > 140 devices and the problem occurs immediately at boot.

The flash is successful.

To solve the problem I set the psram in this way:

#
# ESP SPI
#
CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=0
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_FETCH_INSTRUCTIONS=y
# end ESP SPI

If I set "CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=0" it works.

shubhamdp, 2024-10-10:
`malloc()` always allocates from the internal RAM, and since you are running low on internal RAM, you are seeing this problem.
Setting `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=0` will allocate from the external RAM.
(ref: https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32s3/api-reference/kconfig.html#config-spiram-malloc-alwaysinternal).

I am also scanning the code for any allocations that sits inside the static RAM, and these are many big buffers. I forgot to keep the direct links to the code from where they originated but heres the summary. (I'll try to update these with actual code pointers).

static-dram --> emberAfEndpoint  32 * 140
static-dram --> 2 * no-of-devices (esp-matter)
static-dram --> each cluster allocates few bytes per endpoint

And whenever a new bridged device joins your bridge there's some more heap overhead. 

Good to know that this option works for you.

closed 2024-10-10 by stefano055415