espressif/esp-matter #1104: ESP32-C6 for both Matter over Thread and Matter over WiFi in single chip

bilalmalik76, 2024-10-04:
Is this possible to use ESP32-C6 for both Matter over Thread and Matter over WiFi in single chip? we want to include a button which can be selected before bootup to select either matter over Thread or Matter over WiFi. Is this possible?

Also is there any document for ESP32-C6 for battery life estimation for Matter over Thread case?

shubhamdp, 2024-10-04:
@bilalmalik76 It is possible, and we have a proof of concept. You can find the documentation here: https://gist.github.com/shubhamdp/76723cdedf6cdf88d7e0c49602ccab2. This PoC allows you to choose the interface after each factory reset and not on every boot-up.

> Also is there any document for ESP32-C6 for battery life estimation for Matter over Thread case?

We have some sampled power consumption data, which you can find here: https://github.com/espressif/esp-matter/tree/main/examples/icd_app#4-power-usage. These samples are with Matter ICD feature. Please note that these numbers are not sampled using a battery power source.

bilalmalik76, 2024-10-04:
Hi @shubhamdp Thanks for sharing but this is not accessible: https://gist.github.com/shubhamdp/76723cdedf6cdf88d7e0c49602ccab2

bilalmalik76, 2024-10-04:
@shubhamdp power consumption figures shown on this link are seems to be very high for a battery power device.
https://github.com/espressif/esp-matter/tree/main/examples/icd_app#4-power-usage

shubhamdp, 2024-10-04:
> Hi @shubhamdp Thanks for sharing but this is not accessible: https://gist.github.com/shubhamdp/76723cdedf6cdf88d7e0c49602ccab2

Sorry for the incorrect link, please check https://gist.github.com/shubhamdp/76723dcedf6cdfe88d7e0c49602ccab2

I have also updated the link in the above comment.

shubhamdp, 2024-10-10:
@bilalmalik76 please feel free to close the issue if question is answered.

closed 2024-10-20 by bilalmalik76