espressif/esp-matter #1108: Matter controller using QR or pairing coes

pavel808, 2024-10-04:
I have recently just started using the ESP Matter controller instead of chip-tool directly.  

For commissioning, Is there no way to use the QR code or pairing code directly instead of the pincode and discriminator? 

`matter esp controller pairing ble-thread <node_id> <dataset_tlvs> <pincode> <discriminator> `

I was thinking that this command would be possible : 

`matter esp controller pairing code-thread <node_id> <dataset_tlvs> <QR code> `

shubhamdp, 2024-10-10:
HI @pavel808 right now only `onnetwork`, `ble-wifi,` and `ble-thread` is supported https://github.com/espressif/esp-matter/blob/main/components/esp_matter_controller/commands/esp_matter_controller_pairing_command.cpp#L116.

Yeah, we will add the pairing code parsing support to our controllers.

closed 2024-10-22 by espressif-bot

shubhamdp, 2024-10-22:
@pavel808 We've added the `code`, `code-wifi`, `code-thread`, and code-wifi-thread` in https://github.com/espressif/esp-matter/commit/4490e9aa41f9b966962f74cc3109cb976187b0c2.