espressif/esp-matter #1109: thermostat systemMode

stefano055415, 2024-10-07:
I am adding a thermostat to a bridge.

endpoint::thermostat::config_t thermostat_conf;
err = endpoint::thermostat::add(ep, &thermostat_conf);

I have two questions:
1. How do I create a thermostat that has only heating or only cooling?
2. How can I limit the commands accepted by system mode (for example only 3(cool) and 6(precooling))

jonsmirl, 2024-10-11:
Matter thermostats have feature bits, section 4.3.3 of the Cluster manual.  HEAT and COOL are separate features.

As to commands, first note in the conformance column that many of the commands are controlled by feature bits. If the feature is not turned on, the command is invalid.  Next note in the Access column that many of the commands are optional.

Finally if there are still required commands that you don't want to implement, just stub them out and make them do nothing. In general you shouldn't do that because the remaining required command will be the minimal set needed to implement the function.

jadhavrohit924, 2024-10-11:
> 1. How do I create a thermostat that has only heating or only cooling?

By default, we have created a thermostat with heat and cool features [check this](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_endpoint.cpp#L668). You can remove any one of them as required.

> 2. How can I limit the commands accepted by system mode (for example only 3(cool) and 6(precooling))

This is a known issue, and there is no way to limit SystemMode values. We will try to figure out something to support it.