espressif/esp-matter #1112: Crash on calling esp_matter::controller::send_invoke_cluster_command

pavel808, 2024-10-09:
I am trying to invoke a cluster command from my application running on an ESP32-S3 as follows : 

```
uint64_t commandID = 1871730051;
std::string commandData =  "{\\\"0:U16\\\": " +  std::to_string(commandID) + "}";
ESP_LOGI(TAG, "Command data is %s", commandData.c_str());

// nodeID, endpointID, clusterID, commandID commandData
esp_matter::controller::send_invoke_cluster_command(0x7777, 0x1, 0x131bfc00, 0x1, commandData.c_str());
```
See the crash below. There seems to be some locking / race condition issue. I can't find other working examples of using this send_invoke_cluster_command function. Do I need to call some locking function prior to this? 

```
I (17703) device_commands: Command data is {\"0:U16\": 1871730051}
I (17703) chip[DIS]: Resolving 8C2A5B7D8C06A3DE:0000000000007777 ...
E (17703) chip[DL]: Chip stack locking error at 'src/system/SystemLayerImplFreeRTOS.cpp:106'. Code is unsafe/racy
E (17703) chip[-]: chipDie chipDie chipDie

abort() was called at PC 0x420695eb on core 0
--- 0x420695eb: chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:506
 (inlined by) chipDie at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:516
 (inlined by) chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/LockTracker.cpp:36

Backtrace: 0x40375ada:0x3fcf6a60 0x40380881:0x3fcf6a80 0x4038796e:0x3fcf6aa0 0x420695eb:0x3fcf6b10 0x4209f1c5:0x3fcf6b30 0x4208a6e8:0x3fcf6b50 0x4208a7fd:0x3fcf6b80 0x42085425:0x3fcf6bb0 0x4208585a:0x3fcf6bf0 0x420858a9:0x3fcf6c10 0x42082306:0x3fcf6c30 0x42082336:0x3fcf6ca0 0x4208ffff:0x3fcf6cd0 0x4201252a:0x3fcf6d10 0x42012670:0x3fcf6d40 0x42009951:0x3fcf6d80 0x42008f05:0x3fcf6ed0 0x42177926:0x3fcf7060 0x4217691b:0x3fcf70a0 0x42176f7d:0x3fcf7150 0x42175974:0x3fcf7170 0x421cd6cd:0x3fcf7190 0x42175b78:0x3fcf71b0 0x403812ee:0x3fcf7240
--- 0x40375ada: panic_abort at /home/user/esp-idf/components/esp_system/panic.c:472
0x40380881: esp_system_abort at /home/user/esp-idf/components/esp_system/port/esp_system_chip.c:93
0x4038796e: abort at /home/user/esp-idf/components/newlib/abort.c:38
0x420695eb: chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:506
 (inlined by) chipDie at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:516
 (inlined by) chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/LockTracker.cpp:36
0x4209f1c5: chip::System::LayerImplFreeRTOS::CancelTimer(void (*)(chip::System::Layer*, void*), void*) at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/system/SystemLayerImplFreeRTOS.cpp:106
0x4208a6e8: chip::AddressResolve::Impl::Resolver::ReArmTimer() at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:398
0x4208a7fd: chip::AddressResolve::Impl::Resolver::LookupNode(chip::AddressResolve::NodeLookupRequest const&, chip::AddressResolve::Impl::NodeLookupHandle&) at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:196
 (inlined by) chip::AddressResolve::Impl::Resolver::LookupNode(chip::AddressResolve::NodeLookupRequest const&, chip::AddressResolve::Impl::NodeLookupHandle&) at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:186
0x42085425: chip::OperationalSessionSetup::LookupPeerAddress() at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/OperationalSessionSetup.cpp:612 (discriminator 1)
0x4208585a: chip::OperationalSessionSetup::Connect(chip::Callback::Callback<void (*)(void*, chip::Messaging::ExchangeManager&, chip::SessionHandle const&)>*, chip::Callback::Callback<void (*)(void*, chip::ScopedNodeId const&, chip::ChipError)>*, chip::Callback::Callback<void (*)(void*, chip::OperationalSessionSetup::ConnnectionFailureInfo const&)>*, chip::TransportPayloadCapability) at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/OperationalSessionSetup.cpp:125
0x420858a9: chip::OperationalSessionSetup::Connect(chip::Callback::Callback<void (*)(void*, chip::Messaging::ExchangeManager&, chip::SessionHandle const&)>*, chip::Callback::Callback<void (*)(void*, chip::ScopedNodeId const&, chip::ChipError)>*, chip::TransportPayloadCapability) at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/OperationalSessionSetup.cpp:190
0x42082306: chip::CASESessionManager::FindOrEstablishSessionHelper(chip::ScopedNodeId const&, chip::Callback::Callback<void (*)(void*, chip::Messaging::ExchangeManager&, chip::SessionHandle const&)>*, chip::Callback::Callback<void (*)(void*, chip::ScopedNodeId const&, chip::ChipError)>*, chip::Callback::Callback<void (*)(void*, chip::OperationalSessionSetup::ConnnectionFailureInfo const&)>*, unsigned char, chip::Callback::Callback<void (*)(void*, chip::ScopedNodeId const&, chip::ChipError, std::chrono::duration<unsigned short, std::ratio<1ll, 1ll> >)>*, chip::TransportPayloadCapability) at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/CASESessionManager.cpp:125
0x42082336: chip::CASESessionManager::FindOrEstablishSession(chip::ScopedNodeId const&, chip::Callback::Callback<void (*)(void*, chip::Messaging::ExchangeManager&, chip::SessionHandle const&)>*, chip::Callback::Callback<void (*)(void*, chip::ScopedNodeId const&, chip::ChipError)>*, unsigned char, chip::Callback::Callback<void (*)(void*, chip::ScopedNodeId const&, chip::ChipError, std::chrono::duration<unsigned short, std::ratio<1ll, 1ll> >)>*, chip::TransportPayloadCapability) at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/CASESessionManager.cpp:44
0x4208ffff: chip::Controller::DeviceController::GetConnectedDevice(unsigned long long, chip::Callback::Callback<void (*)(void*, chip::Messaging::ExchangeManager&, chip::SessionHandle const&)>*, chip::Callback::Callback<void (*)(void*, chip::ScopedNodeId const&, chip::ChipError)>*, chip::TransportPayloadCapability) at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/controller/CHIPDeviceController.h:252 (discriminator 3)
 (inlined by) chip::Controller::DeviceController::GetConnectedDevice(unsigned long long, chip::Callback::Callback<void (*)(void*, chip::Messaging::ExchangeManager&, chip::SessionHandle const&)>*, chip::Callback::Callback<void (*)(void*, chip::ScopedNodeId const&, chip::ChipError)>*, chip::TransportPayloadCapability) at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/co [... cut ...]
 (inlined by) httpd_process_session at /home/user/esp-idf/components/esp_http_server/src/httpd_main.c:240
0x421cd6cd: httpd_sess_enum at /home/user/esp-idf/components/esp_http_server/src/httpd_sess.c:50 (discriminator 1)
0x42175b78: httpd_server at /home/user/esp-idf/components/esp_http_server/src/httpd_main.c:305
 (inlined by) httpd_thread at /home/user/esp-idf/components/esp_http_server/src/httpd_main.c:327
0x403812ee: vPortTaskWrapper at /home/user/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 210bcbe37

Rebooting...

```

wqx6, 2024-10-10:
`send_invoke_cluster_command()` should be called in Matter context. You should lock the Matter stack before calling this function or post it to Matter task using `DeviceLayer::PlatformMgr().ScheduleWork()`.

pavel808, 2024-10-10:
> `send_invoke_cluster_command()` should be called in Matter context. You should lock the Matter stack before calling this function or post it to Matter task using `DeviceLayer::PlatformMgr().ScheduleWork()`.

@wqx6 Interesting thanks. I am calling it from a different context indeed, from HTTP handler code. Can I just lock it from there? Are there any coding examples of this?

wqx6, 2024-10-11:
> @wqx6 Interesting thanks. I am calling it from a different context indeed, from HTTP handler code. Can I just lock it from there? Are there any coding examples of this?

Yes, you can call `esp_matter::lock::chip_stack_lock()` before call `send_invoke_cluster_command()`.

dhrishi, 2024-10-29:
@pavel808 Please close this is your issue is resolved

pavel808, 2024-10-29:
> `send_invoke_cluster_command()` should be called in Matter context. You should lock the Matter stack before calling this function or post it to Matter task using `DeviceLayer::PlatformMgr().ScheduleWork()`.

Hi @wqx6 I'm just back from vacation working on this. In fact dealing with the Matter controller commissioning function first : 

```
// NOTE : thread_dataset is defined elsewhere in a uint8_t []

esp_matter::lock::chip_stack_lock(portMAX_DELAY);

err = esp_matter::controller::pairing_code_thread(0x6F4, "34970112332", thread_dataset, sizeof(thread_dataset));

if (err != ESP_OK)
{
    ESP_LOGE(TAG, "ERROR! Unable to commission device : %d", err);
}

esp_matter::lock::chip_stack_unlock();
```

This causes a stack overflow from within the HTTP context that I am calling it from, a POST handler. That is probably expected. 

I tried calling it from somewhere else. The commissioning seems to be successful but then the application crashes due to an unhanded exception : 

```
I (11462) chip[CTL]: Providing certificate chain to the commissioner
I (11462) chip[CTL]: Received callback from the CA for NOC Chain generation. Status Success
I (11462) chip[CTL]: Successfully finished commissioning step 'GenerateNOCChain'
I (11462) chip[CTL]: Performing next commissioning step 'SendTrustedRootCert'
I (11462) chip[CTL]: Sending root certificate to the device
I (11472) chip[EM]: <<< [E:29980i S:8248 M:99794831] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [BLE] --- Type 0001:08 (IM:InvokeCommandRequest) (B:293)
I (11472) NimBLE: GATT procedure initiated: write; 
I (11472) NimBLE: att_handle=16 len=244

E (11472) BLE_INIT: Malloc failed
I (11472) chip[CTL]: Sent root certificate to the device
I (12332) chip[DL]: Posting ESPSystemEvent: IP Event with eventId : 3
I (12332) chip[DL]: HandlePlatformSpecificBLEEvent 49152
I (12332) chip[DL]: IP_EVENT_GOT_IP6
I (12332) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fd00:0000:0000:0000:ce8d:a2ff:fe29:a094
I (12332) chip[DL]: HandlePlatformSpecificBLEEvent 32782
I (12332) app_main: Interface IP Address changed
I (12332) chip[DIS]: Updating services using commissioning mode 0
I (12332) chip[DIS]: Advertise operational node BC407F51F976B0CA-000000000001B669
I (12332) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=0 cp=0
I (12342) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=0000/00 cm=0 cp=0
I (25652) chip[DL]: I (25652) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)HandlePlatformSpecificBLEEvent 16387

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4207582f  PS      : 0x00060930  A0      : 0x82075854  A1      : 0x3fce0b40  
--- 0x4207582f: peer_svc_find_uuid at /home/user/MatterEdgeServer/controller/build/esp-idf/chip/../../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/peer.c:515

A2      : 0x00000000  A3      : 0x3c2275fe  A4      : 0x3fccaf28  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80380d12  A9      : 0x3fce0b10  
A10     : 0x3fccaf28  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00060923  
A14     : 0x00000001  A15     : 0x3fccaedc  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x40056f08  LEND    : 0x40056f12  LCOUNT  : 0x00000000  
--- 0x40056f08: __memcpy_aux in ROM
0x40056f12: __memcpy_aux in ROM

```

Do you understand what is wrong there? 

I am thinking that perhaps the best way to to do this is by using `DeviceLayer::PlatformMgr().ScheduleWork() `as you mentioned?

Are there any examples available of how to use ScheduleWork() to commission a device from another context as a reference?

Thanks in advance.

closed 2024-10-29 by pavel808