espressif/esp-matter #1114: Build fails due to recent release of esp_insights

h2zero, 2024-10-10:
**Describe the bug**
Compilation fails due to new version of esp_insights by [commit](https://github.com/espressif/esp-insights/commit/bbe13d1897b8272dc7bc350bd805e8852722e48d)

**Environment**
- ESP-Matter Commit Id: v1.3.0-d76d4857
- ESP-IDF Commit Id:  v5.1.1-577-g6b1f40b9bf
- SoC (eg: ESP32 or ESP32-C3): esp32
- Device Logs (Please attach the log file): `managed_components/espressif__esp_insights/src/esp_insights_cbor_encoder.c:176:22: error: 'SHA_SIZE' undeclared here`

- Host Machine OS: linux
- Host Machine Python version: 3.10

I fixed this locally by adding `espressif/esp_diag_data_store: "1.0.1"` to my idf_component.yaml in the project folder but should be addressed here.

shripad621git, 2024-10-11:
@h2zero , thanks for reporting the issue.It is a known reported issue in connectedhomeip. The fix for it is up in the [connectedhomeip](https://github.com/project-chip/connectedhomeip/pull/36025). This change will require a submodule update on esp-matter.

h2zero, 2024-10-11:
Thanks, but that doesn't resolve the compile issue when using esp-matter as a component. That same fix should be applied to the idf_component.yaml here and in the esp-matter component release as well.

shubhamdp, 2024-10-11:
@h2zero until we push the new esp-matter component to registry, please check https://github.com/espressif/esp-matter/issues/1115#issuecomment-2406955672 that would help solve the problem till then.

h2zero, 2024-10-11:
@shubhamdp Thanks, I had already done something similar in my project as mentioned in the first post which will hopefully help others that will encounter this.

shubhamdp, 2024-10-15:
This was fixed in https://github.com/espressif/esp-matter/commit/704bed84cf7aac87c21d098b4cdb9237644d272d

closed 2024-10-15 by dhrishi