espressif/esp-matter #1115: Build Failure : 'SHA_SIZE' and 'rtc_store_non_critical_data_hdr_t' Errors (Due to recent esp-Insights Release)

iamiha, 2024-10-11:
**Description**
Attempting to build the `light` example from the esp-matter sdk results in a build failure due to undeclared identifiers and missing structure members. This issue occurs with the latest release of esp-Insights, which introduces changes that cause the `SHA_SIZE` and `rtc_store_non_critical_data_hdr_t.dg` identifiers to not be recognized during compilation. The build fails when using ESP-IDF v5.2.1 and the latest ESP-Matter version.

**Environment**
- **ESP-Matter Commit ID**: v3.1
- **ESP-IDF Commit ID**: v5.2.1
- **SoC**: ESP32
- **Host Machine OS**: Linux
- **Host Machine Python version**: Python 3.10

**Device Logs**
```shell
/home/../light/managed_components/espressif__esp_insights/src/esp_insights_cbor_encoder.c:176:22: error: 'SHA_SIZE' undeclared here (not in a function)
  176 |     char sha_sum[2 * SHA_SIZE + 1];
      |                      ^~~~~~~~
/home/../light/managed_components/espressif__esp_insights/src/esp_insights_cbor_encoder.c: In function 'encode_data_points':
/home/../light/managed_components/espressif__esp_insights/src/esp_insights_cbor_encoder.c:587:20: error: 'rtc_store_non_critical_data_hdr_t' has no member named 'dg'
  587 |         if (!header.dg || !esp_ptr_in_drom(header.dg) || !header.len) {
      |                    ^
/home/../light/managed_components/espressif__esp_insights/src/esp_insights_cbor_encoder.c:587:50: error: 'rtc_store_non_critical_data_hdr_t' has no member named 'dg'
  587 |         if (!header.dg || !esp_ptr_in_drom(header.dg) || !header.len) {
      |                                                  ^
...
ninja: build stopped: subcommand failed.
```

Could you please provide guidance on how to overcome this issue and successfully build the example with the latest esp-Insights? Is there a workaround or specific change required to fix this compatibility issue?

shubhamdp, 2024-10-11:
Duplicate of https://github.com/espressif/esp-matter/issues/1114.

@iamiha There is one workaround mentioned in #1114.

But, I'd suggest the simpler one, please run the below command in your example.

```
idf.py add-dependency "espressif/esp_diag_data_store==1.0.1"

# resume build
idf.py. build
```

iamiha, 2024-10-11:
@shubhamdp  thankyou , it is solved

closed 2024-10-11 by iamiha