espressif/esp-matter #1116: Groups on bridged endpoints

jonsmirl, 2024-10-11:
When I create a group on a bridged endpoint, the group is not getting listed in the group table on the groupkeymanagement cluster. This works correctly on non-bridged endpoints.  I am trying to debug what is going on....

I make the group, then look at the table, the group is not there...
~~~
>>> groups add-group 0x0001 grp1 0x7283 2        
[1728657884.336] [2902752:2902752] [TOO] Command: groups add-group 0x0001 grp1 0x7283 2 
[1728657884.337] [2902752:2902754] [TOO] Sending command to node 0x7283
[1728657884.338] [2902752:2902754] [CSM] FindOrEstablishSession: PeerId = [1:0000000000007283]
[1728657884.338] [2902752:2902754] [CSM] FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1728657884.338] [2902752:2902754] [DIS] Found an existing secure session to [1:0000000000007283]!
[1728657884.338] [2902752:2902754] [DIS] OperationalSessionSetup[1:0000000000007283]: State change 1 --> 5
[1728657884.338] [2902752:2902754] [TOO] Sending cluster (0x00000004) command (0x00000000) on endpoint 2
[1728657884.338] [2902752:2902754] [DMG] ICR moving to [AddingComm]
[1728657884.338] [2902752:2902754] [DMG] ICR moving to [AddedComma]
[1728657884.339] [2902752:2902754] [EM] <<< [E:36146i S:22217 M:112546868] (S) Msg TX to 1:0000000000007283 [889F] [UDP:[fe80::7204:1dff:fecd:9ce4%enp3s0f1]:5540] --- Type 0001:08 (IM:InvokeCommandRequest) (B:69)
[1728657884.339] [2902752:2902754] [DMG] ICR moving to [AwaitingRe]
[1728657884.464] [2902752:2902754] [EM] >>> [E:36146i S:22217 M:19530949 (Ack:112546868)] (S) Msg RX from 1:0000000000007283 [889F] --- Type 0001:09 (IM:InvokeCommandResponse) (B:70)
[1728657884.464] [2902752:2902754] [EM] Found matching exchange: 36146i, Delegate: 0x782258008758
[1728657884.464] [2902752:2902754] [EM] Rxd Ack; Removing MessageCounter:112546868 from Retrans Table on exchange 36146i
[1728657884.464] [2902752:2902754] [DMG] ICR moving to [ResponseRe]
[1728657884.464] [2902752:2902754] [DMG] InvokeResponseMessage =
[1728657884.464] [2902752:2902754] [DMG] {
[1728657884.464] [2902752:2902754] [DMG] 	suppressResponse = false, 
[1728657884.464] [2902752:2902754] [DMG] 	InvokeResponseIBs =
[1728657884.464] [2902752:2902754] [DMG] 	[
[1728657884.464] [2902752:2902754] [DMG] 		InvokeResponseIB =
[1728657884.464] [2902752:2902754] [DMG] 		{
[1728657884.464] [2902752:2902754] [DMG] 			CommandDataIB =
[1728657884.464] [2902752:2902754] [DMG] 			{
[1728657884.464] [2902752:2902754] [DMG] 				CommandPathIB =
[1728657884.464] [2902752:2902754] [DMG] 				{
[1728657884.464] [2902752:2902754] [DMG] 					EndpointId = 0x2,
[1728657884.464] [2902752:2902754] [DMG] 					ClusterId = 0x4,
[1728657884.464] [2902752:2902754] [DMG] 					CommandId = 0x0,
[1728657884.464] [2902752:2902754] [DMG] 				},
[1728657884.464] [2902752:2902754] [DMG] 				
[1728657884.464] [2902752:2902754] [DMG] 				CommandFields = 
[1728657884.464] [2902752:2902754] [DMG] 				{
[1728657884.464] [2902752:2902754] [DMG] 					0x0 = 126 (unsigned), 
[1728657884.464] [2902752:2902754] [DMG] 					0x1 = 1 (unsigned), 
[1728657884.464] [2902752:2902754] [DMG] 				},
[1728657884.464] [2902752:2902754] [DMG] 			},
[1728657884.464] [2902752:2902754] [DMG] 			
[1728657884.464] [2902752:2902754] [DMG] 		},
[1728657884.464] [2902752:2902754] [DMG] 		
[1728657884.464] [2902752:2902754] [DMG] 	],
[1728657884.464] [2902752:2902754] [DMG] 	
[1728657884.464] [2902752:2902754] [DMG] 	InteractionModelRevision = 11
[1728657884.464] [2902752:2902754] [DMG] },
[1728657884.464] [2902752:2902754] [DMG] Received Command Response Data, Endpoint=2 Cluster=0x0000_0004 Command=0x0000_0000
[1728657884.464] [2902752:2902754] [TOO] Endpoint: 2 Cluster: 0x0000_0004 Command 0x0000_0000
[1728657884.464] [2902752:2902754] [TOO]   AddGroupResponse: {
[1728657884.464] [2902752:2902754] [TOO]     status: 126
[1728657884.464] [2902752:2902754] [TOO]     groupID: 1
[1728657884.464] [2902752:2902754] [TOO]    }
[1728657884.464] [2902752:2902754] [DMG] ICR moving to [AwaitingDe]
[1728657884.464] [2902752:2902754] [EM] <<< [E:36146i S:22217 M:112546869 (Ack:19530949)] (S) Msg TX to 1:0000000000007283 [889F] [UDP:[fe80::7204:1dff:fecd:9ce4%enp3s0f1]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
[1728657884.464] [2902752:2902754] [EM] Flushed pending ack for MessageCounter:19530949 on exchange 36146i
>>> groupkeymanagement read group-table 0x7283 0
[1728657895.227] [2902752:2902752] [TOO] Command: groupkeymanagement read group-table 0x7283 0 
[1728657895.227] [2902752:2902754] [TOO] Sending command to node 0x7283
[1728657895.228] [2902752:2902754] [CSM] FindOrEstablishSession: PeerId = [1:0000000000007283]
[1728657895.228] [2902752:2902754] [CSM] FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1728657895.229] [2902752:2902754] [DIS] Found an existing secure session to [1:0000000000007283]!
[1728657895.229] [2902752:2902754] [DIS] OperationalSessionSetup[1:0000000000007283]: State change 1 --> 5
[1728657895.229] [2902752:2902754] [TOO] Sending ReadAttribute to:
[1728657895.229] [2902752:2902754] [TOO] 	cluster 0x0000_003F, attribute: 0x0000_0001, endpoint 0
[1728657895.229] [2902752:2902754] [DMG] SendReadRequest ReadClient[0x78226003b000]: Sending Read Request
[1728657895.229] [2902752:2902754] [DMG] 0 data version filters provided, 0 not relevant, 0 encoded, 0 skipped due to lack of space
[1728657895.229] [2902752:2902754] [EM] <<< [E:36147i S:22217 M:112546870] (S) Msg TX to 1:0000000000007283 [889F] [UDP:[fe80::7204:1dff:fecd:9ce4%enp3s0f1]:5540] --- Type 0001:02 (IM:ReadRequest) (B:51)
[1728657895.229] [2902752:2902754] [DMG] MoveToState ReadClient[0x78226003b000]: Moving to [AwaitingIn]
[1728657895.306] [2902752:2902754] [EM] >>> [E:36147i S:22217 M:19530950 (Ack:112546870)] (S) Msg RX from 1:0000000000007283 [889F] --- Type 0001:05 (IM:ReportData) (B:70)
[1728657895.306] [2902752:2902754] [EM] Found matching exchange: 36147i, Delegate: 0x78226003b010
[1728657895.306] [2902752:2902754] [EM] Rxd Ack; Removing MessageCounter:112546870 from Retrans Table on exchange 36147i
[1728657895.306] [2902752:2902754] [DMG] ReportDataMessage =
[1728657895.306] [2902752:2902754] [DMG] {
[1728657895.306] [2902752:2902754] [DMG] 	AttributeReportIBs =
[1728657895.306] [2902752:2902754] [DMG] 	[
[1728657895.306] [2902752:2902754] [DMG] 		AttributeReportIB =
[1728657895.306] [2902752:2902754] [DMG] 		{
[1728657895.306] [2902752:2902754] [DMG] 			AttributeDataIB =
[1728657895.306] [2902752:2902754] [DMG] 			{
[1728657895.306] [2902752:2902754] [DMG] 				DataVersion = 0x556391a8,
[1728657895.306] [2902752:2902754] [DMG] 				AttributePathIB =
[1728657895.306] [2902752:2902754] [DMG] 				{
[1728657895.306] [2902752:2902754] [DMG] 					Endpoint = 0x0,
[1728657895.306] [2902752:2902754] [DMG] 					Cluster = 0x3f,
[1728657895.306] [2902752:2902754] [DMG] 					Attribute = 0x0000_0001,
[1728657895.306] [2902752:2902754] [DMG] 				}
[1728657895.306] [2902752:2902754] [DMG] 					
[1728657895.306] [2902752:2902754] [DMG] 				Data = [
[1728657895.306] [2902752:2902754] [DMG] 						
[1728657895.306] [2902752:2902754] [DMG] 				],
[1728657895.306] [2902752:2902754] [DMG] 			},
[1728657895.306] [2902752:2902754] [DMG] 			
[1728657895.306] [2902752:2902754] [DMG] 		},
[1728657895.306] [2902752:2902754] [DMG] 		
[1728657895.306] [2902752:2902754] [DMG] 	],
[1728657895.306] [2902752:2902754] [DMG] 	
[1728657895.306] [2902752:2902754] [DMG] 	SuppressResponse = true, 
[1728657895.306] [2902752:2902754] [DMG] 	InteractionModelRevision = 11
[1728657895.306] [2902752:2902754] [DMG] }
[1728657895.306] [2902752:2902754] [TOO] Endpoint: 0 Cluster: 0x0000_003F Attribute 0x0000_0001 DataVersion: 1432588712
[1728657895.306] [2902752:2902754] [TOO]   GroupTable: 0 entries
[1728657895.307] [2902752:2902754] [EM] <<< [E:36147i S:22217 M:112546871 (Ack:19530950)] (S) Msg TX to 1:0000000000007283 [889F] [UDP:[fe80::7204:1dff:fecd:9ce4%enp3s0f1]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
[1728657895.307] [2902752:2902754] [EM] Flushed pending ack for MessageCounter:19530950 on exchange 36147i
~~~

jonsmirl, 2024-10-11:
The key map is correct:
~~~
>>> groupkeymanagement read group-key-map 0x7283 0
[1728661384.384] [2902752:2902752] [TOO] Command: groupkeymanagement read group-key-map 0x7283 0 
[1728661384.385] [2902752:2902754] [TOO] Sending command to node 0x7283
[1728661384.386] [2902752:2902754] [CSM] FindOrEstablishSession: PeerId = [1:0000000000007283]
[1728661384.386] [2902752:2902754] [CSM] FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1728661384.386] [2902752:2902754] [DIS] Found an existing secure session to [1:0000000000007283]!
[1728661384.386] [2902752:2902754] [DIS] OperationalSessionSetup[1:0000000000007283]: State change 1 --> 5
[1728661384.386] [2902752:2902754] [TOO] Sending ReadAttribute to:
[1728661384.386] [2902752:2902754] [TOO] 	cluster 0x0000_003F, attribute: 0x0000_0000, endpoint 0
[1728661384.386] [2902752:2902754] [DMG] SendReadRequest ReadClient[0x78226003b000]: Sending Read Request
[1728661384.386] [2902752:2902754] [DMG] 0 data version filters provided, 0 not relevant, 0 encoded, 0 skipped due to lack of space
[1728661384.387] [2902752:2902754] [EM] <<< [E:36149i S:22217 M:112546874] (S) Msg TX to 1:0000000000007283 [889F] [UDP:[fe80::7204:1dff:fecd:9ce4%enp3s0f1]:5540] --- Type 0001:02 (IM:ReadRequest) (B:51)
[1728661384.387] [2902752:2902754] [DMG] MoveToState ReadClient[0x78226003b000]: Moving to [AwaitingIn]
[1728661384.409] [2902752:2902754] [EM] >>> [E:36149i S:22217 M:19530952 (Ack:112546874)] (S) Msg RX from 1:0000000000007283 [889F] --- Type 0001:05 (IM:ReportData) (B:81)
[1728661384.409] [2902752:2902754] [EM] Found matching exchange: 36149i, Delegate: 0x78226003b010
[1728661384.409] [2902752:2902754] [EM] Rxd Ack; Removing MessageCounter:112546874 from Retrans Table on exchange 36149i
[1728661384.409] [2902752:2902754] [DMG] ReportDataMessage =
[1728661384.409] [2902752:2902754] [DMG] {
[1728661384.409] [2902752:2902754] [DMG] 	AttributeReportIBs =
[1728661384.409] [2902752:2902754] [DMG] 	[
[1728661384.409] [2902752:2902754] [DMG] 		AttributeReportIB =
[1728661384.409] [2902752:2902754] [DMG] 		{
[1728661384.410] [2902752:2902754] [DMG] 			AttributeDataIB =
[1728661384.410] [2902752:2902754] [DMG] 			{
[1728661384.410] [2902752:2902754] [DMG] 				DataVersion = 0x556391a8,
[1728661384.410] [2902752:2902754] [DMG] 				AttributePathIB =
[1728661384.410] [2902752:2902754] [DMG] 				{
[1728661384.410] [2902752:2902754] [DMG] 					Endpoint = 0x0,
[1728661384.410] [2902752:2902754] [DMG] 					Cluster = 0x3f,
[1728661384.410] [2902752:2902754] [DMG] 					Attribute = 0x0000_0000,
[1728661384.410] [2902752:2902754] [DMG] 				}
[1728661384.410] [2902752:2902754] [DMG] 					
[1728661384.410] [2902752:2902754] [DMG] 				Data = [
[1728661384.410] [2902752:2902754] [DMG] 						
[1728661384.410] [2902752:2902754] [DMG] 					{
[1728661384.410] [2902752:2902754] [DMG] 						0x1 = 1 (unsigned), 
[1728661384.410] [2902752:2902754] [DMG] 						0x2 = 42 (unsigned), 
[1728661384.410] [2902752:2902754] [DMG] 						0xfe = 1 (unsigned), 
[1728661384.410] [2902752:2902754] [DMG] 					},
[1728661384.410] [2902752:2902754] [DMG] 				],
[1728661384.410] [2902752:2902754] [DMG] 			},
[1728661384.410] [2902752:2902754] [DMG] 			
[1728661384.410] [2902752:2902754] [DMG] 		},
[1728661384.410] [2902752:2902754] [DMG] 		
[1728661384.410] [2902752:2902754] [DMG] 	],
[1728661384.410] [2902752:2902754] [DMG] 	
[1728661384.410] [2902752:2902754] [DMG] 	SuppressResponse = true, 
[1728661384.410] [2902752:2902754] [DMG] 	InteractionModelRevision = 11
[1728661384.410] [2902752:2902754] [DMG] }
[1728661384.410] [2902752:2902754] [TOO] Endpoint: 0 Cluster: 0x0000_003F Attribute 0x0000_0000 DataVersion: 1432588712
[1728661384.410] [2902752:2902754] [TOO]   GroupKeyMap: 1 entries
[1728661384.410] [2902752:2902754] [TOO]     [1]: {
[1728661384.411] [2902752:2902754] [TOO]       GroupId: 1
[1728661384.411] [2902752:2902754] [TOO]       GroupKeySetID: 42
[1728661384.411] [2902752:2902754] [TOO]       FabricIndex: 1
[1728661384.411] [2902752:2902754] [TOO]      }
[1728661384.411] [2902752:2902754] [EM] <<< [E:36149i S:22217 M:112546875 (Ack:19530952)] (S) Msg TX to 1:0000000000007283 [889F] [UDP:[fe80::7204:1dff:fecd:9ce4%enp3s0f1]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
[1728661384.411] [2902752:2902754] [EM] Flushed pending ack for MessageCounter:19530952 on exchange 36149i

~~~

jonsmirl, 2024-10-11:
I can view the group on the end point...
So why is it missing from the the table on groupkeymanagement?
~~~
>>> groups view-group 1 0x7283 2
[1728661491.184] [2902752:2902752] [TOO] Command: groups view-group 1 0x7283 2 
[1728661491.185] [2902752:2902754] [TOO] Sending command to node 0x7283
[1728661491.186] [2902752:2902754] [CSM] FindOrEstablishSession: PeerId = [1:0000000000007283]
[1728661491.186] [2902752:2902754] [CSM] FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1728661491.186] [2902752:2902754] [DIS] Found an existing secure session to [1:0000000000007283]!
[1728661491.186] [2902752:2902754] [DIS] OperationalSessionSetup[1:0000000000007283]: State change 1 --> 5
[1728661491.186] [2902752:2902754] [TOO] Sending cluster (0x00000004) command (0x00000001) on endpoint 2
[1728661491.186] [2902752:2902754] [DMG] ICR moving to [AddingComm]
[1728661491.186] [2902752:2902754] [DMG] ICR moving to [AddedComma]
[1728661491.186] [2902752:2902754] [EM] <<< [E:36150i S:22217 M:112546876] (S) Msg TX to 1:0000000000007283 [889F] [UDP:[fe80::7204:1dff:fecd:9ce4%enp3s0f1]:5540] --- Type 0001:08 (IM:InvokeCommandRequest) (B:62)
[1728661491.186] [2902752:2902754] [DMG] ICR moving to [AwaitingRe]
[1728661491.211] [2902752:2902754] [EM] >>> [E:36150i S:22217 M:19530953 (Ack:112546876)] (S) Msg RX from 1:0000000000007283 [889F] --- Type 0001:09 (IM:InvokeCommandResponse) (B:73)
[1728661491.211] [2902752:2902754] [EM] Found matching exchange: 36150i, Delegate: 0x782258008758
[1728661491.211] [2902752:2902754] [EM] Rxd Ack; Removing MessageCounter:112546876 from Retrans Table on exchange 36150i
[1728661491.211] [2902752:2902754] [DMG] ICR moving to [ResponseRe]
[1728661491.211] [2902752:2902754] [DMG] InvokeResponseMessage =
[1728661491.211] [2902752:2902754] [DMG] {
[1728661491.211] [2902752:2902754] [DMG] 	suppressResponse = false, 
[1728661491.211] [2902752:2902754] [DMG] 	InvokeResponseIBs =
[1728661491.211] [2902752:2902754] [DMG] 	[
[1728661491.211] [2902752:2902754] [DMG] 		InvokeResponseIB =
[1728661491.212] [2902752:2902754] [DMG] 		{
[1728661491.212] [2902752:2902754] [DMG] 			CommandDataIB =
[1728661491.212] [2902752:2902754] [DMG] 			{
[1728661491.212] [2902752:2902754] [DMG] 				CommandPathIB =
[1728661491.212] [2902752:2902754] [DMG] 				{
[1728661491.212] [2902752:2902754] [DMG] 					EndpointId = 0x2,
[1728661491.212] [2902752:2902754] [DMG] 					ClusterId = 0x4,
[1728661491.212] [2902752:2902754] [DMG] 					CommandId = 0x1,
[1728661491.212] [2902752:2902754] [DMG] 				},
[1728661491.212] [2902752:2902754] [DMG] 				
[1728661491.212] [2902752:2902754] [DMG] 				CommandFields = 
[1728661491.212] [2902752:2902754] [DMG] 				{
[1728661491.212] [2902752:2902754] [DMG] 					0x0 = 139 (unsigned), 
[1728661491.212] [2902752:2902754] [DMG] 					0x1 = 1 (unsigned), 
[1728661491.212] [2902752:2902754] [DMG] 					0x2 = "" (0 chars), 
[1728661491.212] [2902752:2902754] [DMG] 				},
[1728661491.212] [2902752:2902754] [DMG] 			},
[1728661491.212] [2902752:2902754] [DMG] 			
[1728661491.212] [2902752:2902754] [DMG] 		},
[1728661491.212] [2902752:2902754] [DMG] 		
[1728661491.212] [2902752:2902754] [DMG] 	],
[1728661491.212] [2902752:2902754] [DMG] 	
[1728661491.212] [2902752:2902754] [DMG] 	InteractionModelRevision = 11
[1728661491.212] [2902752:2902754] [DMG] },
[1728661491.212] [2902752:2902754] [DMG] Received Command Response Data, Endpoint=2 Cluster=0x0000_0004 Command=0x0000_0001
[1728661491.212] [2902752:2902754] [TOO] Endpoint: 2 Cluster: 0x0000_0004 Command 0x0000_0001
[1728661491.212] [2902752:2902754] [TOO]   ViewGroupResponse: {
[1728661491.212] [2902752:2902754] [TOO]     status: 139
[1728661491.212] [2902752:2902754] [TOO]     groupID: 1
[1728661491.212] [2902752:2902754] [TOO]     groupName: 
[1728661491.212] [2902752:2902754] [TOO]    }
[1728661491.212] [2902752:2902754] [DMG] ICR moving to [AwaitingDe]
[1728661491.212] [2902752:2902754] [EM] <<< [E:36150i S:22217 M:112546877 (Ack:19530953)] (S) Msg TX to 1:0000000000007283 [889F] [UDP:[fe80::7204:1dff:fecd:9ce4%enp3s0f1]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
[1728661491.212] [2902752:2902754] [EM] Flushed pending ack for MessageCounter:19530953 on exchange 36150i
>>> 
~~~

jonsmirl, 2024-10-11:
I suspect the code that dynamically constructs the group table on the groupkeymanagement cluster is not finding the bridged endpoints. Since it doesn't find them, it doesn't add them to the table.

jonsmirl, 2024-10-11:
Not sure what is going on here, I fully erased the flash and now it seems to be working. I will test more.

mortezaMoezzifar, 2024-10-14:
- Test the commands below, I got an answer

`2// add access to controller and self
./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null },{"fabricIndex": 1, "privilege": 4, "authMode": 3, "subjects": [1], "targets": null }]' 1 0
				
// create a key for group
./chip-tool groupkeymanagement key-set-write '{"groupKeySetID": 42, "groupKeySecurityPolicy": 0, "epochKey0": "d0d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime0": 2220000,"epochKey1": "d1d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime1": 2220001,"epochKey2": "d2d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime2": 2220002 }' 1 0

// add group to node ( note both command required )
./chip-tool groups add-group 0x0001 grp1 1 1 
./chip-tool groups add-group-if-identifying 0x0001 grp1 1 1 

// set write group key and group id 
./chip-tool groupkeymanagement write group-key-map '[{"groupId": 1, "groupKeySetID": 42, "fabricIndex": 1}]' 1 0

// add group to controller 
./chip-tool groupsettings add-group grp1 0x0001

// add key set to controller 
./chip-tool groupsettings add-keysets 0x0042 0 0x000000000021dfe0 hex:d0d1d2d3d4d5d6d7d8d9dadbdcdddedf

// bind key set in controller 
./chip-tool groupsettings bind-keyset 0x0001 0x0042

// toggle command on group
./chip-tool onoff toggle 0xffffffffffff0001 1`

jonsmirl, 2024-10-14:
That is the normal test sequence I use.

Normal devices are working fine. Bridged devices are inconsistently failing but I haven't figured out a pattern to the failure yet. My current guess is that it is something to do with writing the group data for bridged endpoints into flash. Sometimes when I access the group data later it isn't in the flash although I know my code sent the commands to write it. I haven't figured out yet why that happens.

jonsmirl, 2024-10-15:
I attached an example of the corruption which is making groups fail for me. In this case the failing device is a light bulb using esp-matter/examples/light.
[corrupt.txt](https://github.com/user-attachments/files/17382770/corrupt.txt)

Check in the log and the last group entry is an error.  So how do I decode the groups in that flash dump to see why the last one is messed up?

~~~
2024-10-15 13:00:50.932 11718-11813 EM                      com.lowpan.m2                        D  Rxd Ack; Removing MessageCounter:17564490 from Retrans Table on exchange 50172i
2024-10-15 13:00:50.932 11718-11813 DMG                     com.lowpan.m2                        D  WriteClient moving to [ResponseRe]
2024-10-15 13:00:50.932 11718-11813 DMG                     com.lowpan.m2                        D  WriteResponseMessage =
2024-10-15 13:00:50.932 11718-11813 DMG                     com.lowpan.m2                        D  {
2024-10-15 13:00:50.932 11718-11813 DMG                     com.lowpan.m2                        D  	AttributeStatusIBs =
2024-10-15 13:00:50.932 11718-11813 DMG                     com.lowpan.m2                        D  	[
2024-10-15 13:00:50.932 11718-11813 DMG                     com.lowpan.m2                        D  		AttributeStatusIB =
2024-10-15 13:00:50.932 11718-11813 DMG                     com.lowpan.m2                        D  		{
2024-10-15 13:00:50.932 11718-11813 DMG                     com.lowpan.m2                        D  			AttributePathIB =
2024-10-15 13:00:50.932 11718-11813 DMG                     com.lowpan.m2                        D  			{
2024-10-15 13:00:50.932 11718-11813 DMG                     com.lowpan.m2                        D  				Endpoint = 0x0,
2024-10-15 13:00:50.932 11718-11813 DMG                     com.lowpan.m2                        D  				Cluster = 0x3f,
2024-10-15 13:00:50.932 11718-11813 DMG                     com.lowpan.m2                        D  				Attribute = 0x0000_0000,
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			}
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  				
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			StatusIB =
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			{
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  				status = 0x00,
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			},
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  		},
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  		
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  		AttributeStatusIB =
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  		{
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			AttributePathIB =
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			{
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  				Endpoint = 0x0,
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  				Cluster = 0x3f,
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  				Attribute = 0x0000_0000,
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  				ListIndex = Null,
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			}
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  				
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			StatusIB =
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			{
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  				status = 0x00,
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			},
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  		},
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  		
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  		AttributeStatusIB =
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  		{
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			AttributePathIB =
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			{
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  				Endpoint = 0x0,
[... cut ...]
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  			
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  		},
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  		
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  	],
2024-10-15 13:00:50.933 11718-11813 DMG                     com.lowpan.m2                        D  	
~~~

jonsmirl, 2024-10-15:
I have a working theory now. While it is not in the spec, GroupKeyManagementClusterGroupKeyMapStruct has a column for FabricIndex. When I write the attribute I have to include that column (even though I can't change it).  So the write failed because I set in the wrong fabricId.

To fix this I have figured out now that I can get the fabricId off from chipDeviceController and set it into the array. Doing it that way which should prevent me from using the wrong FabricId.

Now I need to do some testing to see if this is the source of my group failures.

~~~
        update = false
        if (!keyMap.any { it.groupId == groupId }) {
            keyMap.add(ChipStructs.GroupKeyManagementClusterGroupKeyMapStruct(groupId, 42, chipClient.chipDeviceController!!.fabricIndex))
            update = true
        }
        val keyMapCluster = ChipClusters.GroupKeyManagementCluster(
            getConnectedDevicePointer(), 0
        )
        if (update) {
            Timber.d("JDS addGroup writeGroupKeyMapAttribute %x %s", deviceId, keyMap.toString())
            suspendCoroutine { c -> keyMapCluster.writeGroupKeyMapAttribute(cb(c), keyMap) }
        }
~~~

jonsmirl, 2024-10-16:
I believe that was my problem, on Android GroupKeyManagementClusterGroupKeyMapStruct changed behavior which broke my existing code. This wasn't an API change, it was a change in how CHIP detects errors. Now that I know what happened this wasn't hard to fix. Also, on Android I have to build my own CHIP libraries to include in my app so I don't need to worry about the OS API being out of sync with what I expect.  I will say the error code in the case was awful, error 1 is 'general failure' type error.  My code was catching and logging this error but it was lost in all of the other log spew from CHIP.

closed 2024-10-16 by jonsmirl