espressif/esp-matter #1117: Missing include.

HamzaHajeir, 2024-10-12:
**Describe the bug**

When `CONFIG_CUSTOM_DAC_PROVIDER` is enabled, this line of error pops in compilation process pointing to [this line](https://github.com/espressif/esp-matter/blob/186212e9c705913468c755e6d6668671c28a89a0/components/esp_matter/esp_matter_providers.cpp#L85):

`error: 'ESP_LOGE' was not declared in this scope`

which is resolved by inserting `#include<esp_log.h>`

**Environment**
- ESP-Matter version Id: 1.3 
- ESP-IDF version Id: 5.3.1
- SoC (eg: ESP32 or ESP32-C3): ESP32-S3
- Host Machine OS: Windows 11

shripad621git, 2024-10-14:
@HamzaHajeir , thanks for reporting the issue. The suggested change is added to release/v1.3. The change will be available soon.

dhrishi, 2024-10-15:
The fix is merged to `release/v1.3` branch

closed 2024-10-15 by dhrishi