espressif/esp-matter #1118: Connecting Device to Default and New WiFi During Commissioning"

mortezaMoezzifar, 2024-10-13:
I want the device to connect to a default WiFi when it is not commissioned, and during commissioning, with `code-wifi`, it should connect to a new WiFi provided during commissioning. Can you guide me?

sayondeep, 2024-10-15:
@mortezaMoezzifar 

Please run `idf.py menuconfig`

Please go to `Component config → CHIP Device Layer → WiFi Station Options` and configure default wifi and password.

It will connect to the configured AP on boot up.

Then you can commission the device using `chip-tool pairing code-wifi 0x<node-id> <SSID> <password> MT:Y.K9042C00KA0648G00`.

During commissioning, the device will connect to the SSID.

mortezaMoezzifar, 2024-10-15:
@sayondeep 
### Thanks so much! I gave it a try, and it worked perfectly.

closed 2024-10-15 by mortezaMoezzifar