espressif/esp-matter #1120: Accessories list in the Matter Bridge Apple/Google Home app is visible the empty.

halilakpinarr, 2024-10-15:
**Bug Description**

We are developing a Matter Bridge.
We are using ESP32 as Telemetry Ethernet. (BLE: disabled, WiFi: disabled)
Our problem is; when the bridge application is erased and flashed for the first time, we can create endpoints using the "app_bridge_create_bridged_device" API as in the Zigbee Bridge Example. After the first flashing, when we pair our bridge device with Apple Home, the accessory list is visible. After the first flashing, when we pair the mobile application and bridge device with Apple Home without pairing it with "idf.py monitor" or by cutting the power and returning it, the accessory list is visible the empty in the mobile application. After pairing while the accessory list is empty, we can add new dynamic endpoints with the same API "app_bridge_create_bridged_device". In this case, for example, while 10 endpoints are actually added in the mobile application, only the last endpoint we added is visible in the mobile application.

When we first flashed frimware to the device, for example, we added 10 dynamic points and then commissioned by apple home app, saw 10 accessories in the mobile application. Then, we deleted the fabric by saying "remove device" from the Apple Home application. Then, when we put the same device back into basic commissioning mode and paired it from the mobile application, although 10 endpoints were still added, only the bridge is visible in the mobile application. Again, if I add new accessories, I can only see the ones I added while they were already paired in the mobile application.

By the way, CONFIG_ESP_MATTER_ENABLE_DATA_MODEL and CONFIG_ESP_MATTER_ENABLE_MATTER_SERVER options are enabled.

Matter endpoints should normally be in the aggregator "PartsList" section and in the "PartsList" of the Root Node.
Espressif "attribute::create_parts_list(cluster, NULL, 0, 0);" has a comment line in "esp_matter_cluster.cpp" indicating that this is managed internally.
I think this may be the reason why we can't see the endpoints in the mobile application despite them being created.

By the way, after installing firmware on the device, we do not do anything via mfg_tool or chip-tool. Do we need to set anything? 

We are waiting for your support or suggestions for a solution.

**Environment**
- ESP-Matter Commit Id: 186212e9c705913468c755e6d6668671c28a89a0 (latest commit now on main branch)
- ESP-IDF Commit Id: v5.2.1
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file):

- Host Machine OS: Ubuntu 22.04

- Home hub app and versions if present: Apple TV, Google Nest Mini

jonsmirl, 2024-10-16:
Commission your bridge using chip-tool. Add the 10 accessories. Use chip-tool to verify they are there. Use chip-tool to remove the fabric. Put your accessories back. Finally use chip-tool to verify if the parts-lists are right.  Using chip-tool to go through all of the same steps will let you see what is going on.  Esp code should automatically populate the parts-lists.

Another test would be to reboot the bridge right after you remove the fabric and see if that has any impact on Apple Home.

You need to narrow down whether the problem is in the bridge or Apple Home. chip-tool will let you do that.

It is also a known fact that both Google and Apple will mess up if you combine fixed device endpoints (root/aggregator are not device endpoints) with bridged ones on the same device. The spec allows this but Google/Apple don't have it implemented correctly. The symptom you will see is lots of disappearing endpoints. 

Arbitrarily composed composite devices don't work on Google/Apple either.  For example if you put a light and a temp sensor together in a composed device it won't work right on Google/Apple even though they support both the light and temp sensor when not composed.

halilakpinarr, 2024-10-16:
> Commission your bridge using chip-tool. Add the 10 accessories. Use chip-tool to verify they are there. Use chip-tool to remove the fabric. Put your accessories back. Finally use chip-tool to verify if the parts-lists are right. Using chip-tool to go through all of the same steps will let you see what is going on.
> 
> Esp code should automatically populate the parts-lists. But... the remove fabric step could be breaking that behavior (which would be bug). Another test would be to reboot the bridge right after you remove the fabric and see if that has any impact on Apple Home.
> 
> You need to narrow down whether the problem is in the bridge or Apple Home. chip-tool will let you do that.
> 
> It is also a known fact that both Google and Apple will mess up if you combine fixed device endpoints (root/aggregator are not device endpoints) with bridged ones on the same device. The spec allows this but Google/Apple don't have it implemented correctly. The symptom you will see is lots of disappearing endpoints.
> 
> Arbitrarily composed composite devices don't work on Google/Apple either. For example if you put a light and a temp sensor together in a composed device it won't work right on Google/Apple even though they support both the light and temp sensor when not composed.

@jonsmirl I commissioned the device with the chip-tool, when I get the parts lists for endpoint 0(Root node) and endpoint 1(aggregrator) device types separately(chip-tool logs below), everything seems to be fine. However, it is interesting that only the bridge device is visible in mobile applications. By the way, we do not combine the device types, I add the ESP_MATTER_ON_OFF_LIGHT_DEVICE_TYPE_ID and ESP_MATTER_DIMMABLE_LIGHT_DEVICE_TYPE_ID devices in the standard SDK, I do not add any clusters on it. I do not want to accept that the problem is in the mobile application, the reason is that I have a professionally sold Matter Bridge device and it works fine with the same Apple and Google Home applications versions, they used a raspberry pi. We are missing something, but I have not found it yet. Do you have another idea?

### Root node endpoint partslist

chip-tool descriptor read parts-list 0x7283 0x0
[1729056319.669] [10407:10407] [DL] ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_tool_kvs
[1729056319.670] [10407:10407] [DL] ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1729056319.670] [10407:10407] [DL] ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_kvs
[1729056319.672] [10407:10407] [DL] ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1729056319.672] [10407:10407] [DL] ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1729056319.672] [10407:10407] [DL] ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1729056319.672] [10407:10407] [DL] wrote settings to /tmp/chip_counters.ini
[1729056319.672] [10407:10407] [DL] NVS set: chip-counters/reboot-count = 8 (0x8)
[1729056319.672] [10407:10407] [DL] Got Ethernet interface: enp2s0
[1729056319.672] [10407:10407] [DL] Found the primary Ethernet interface:enp2s0
[1729056319.673] [10407:10407] [DL] Got WiFi interface: wlo1
[1729056319.674] [10407:10407] [DL] Found the primary WiFi interface:wlo1
[1729056319.674] [10407:10407] [IN] UDP::Init bind&listen port=0
[1729056319.674] [10407:10407] [IN] UDP::Init bound to port=52579
[1729056319.674] [10407:10407] [IN] BLEBase::Init - setting/overriding transport
[1729056319.674] [10407:10407] [IN] TransportMgr initialized
[1729056319.674] [10407:10407] [FP] Initializing FabricTable from persistent storage
[1729056319.674] [10407:10407] [TS] Last Known Good Time: 2023-10-14T01:16:48
[1729056319.675] [10407:10407] [FP] Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xFB874A35140CD896, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1729056319.675] [10407:10407] [ZCL] Using ZAP configuration...
[1729056319.676] [10407:10407] [IM] WARNING ┌────────────────────────────────────────────────────
[1729056319.676] [10407:10407] [IM] WARNING │ Interaction Model Engine running in 'Checked' mode.
[1729056319.676] [10407:10407] [IM] WARNING │ This executes BOTH ember and data-model code paths.
[1729056319.676] [10407:10407] [IM] WARNING │ which is inefficient and consumes more flash space.
[1729056319.676] [10407:10407] [IM] WARNING │ This should be done for testing only.
[1729056319.676] [10407:10407] [IM] WARNING └────────────────────────────────────────────────────
[1729056319.676] [10407:10407] [CTL] System State Initialized...
[1729056319.676] [10407:10407] [CTL] Setting attestation nonce to random value
[1729056319.676] [10407:10407] [CTL] Setting CSR nonce to random value
[1729056319.676] [10407:10409] [DL] CHIP task running
[1729056319.676] [10407:10409] [DL] HandlePlatformSpecificBLEEvent 32786
[1729056319.676] [10407:10409] [CTL] Setting attestation nonce to random value
[1729056319.676] [10407:10409] [CTL] Setting CSR nonce to random value
[1729056319.676] [10407:10409] [CTL] Generating NOC
[1729056319.676] [10407:10409] [FP] Validating NOC chain
[1729056319.677] [10407:10409] [FP] NOC chain validation successful
[1729056319.677] [10407:10409] [FP] Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1729056319.677] [10407:10409] [TS] Last Known Good Time: 2023-10-14T01:16:48
[1729056319.677] [10407:10409] [TS] New proposed Last Known Good Time: 2021-01-01T00:00:00
[... cut ...]
I (951199) chip[EM]: <<< [E:30183r S:5419 M:253023451 (Ack:196640849)] (S) Msg TX to 4:00000000F040A290 [2409] [UDP:[FE80::83B:F58:88F:2CF4%en1]:62836] --- Type 0001:05 (IM:ReportData) (B:258)
I (951219) chip[EM]: >>> [E:30182r S:5419 M:196640850 (Ack:253023450)] (S) Msg RX from 4:00000000F040A290 [2409] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
I (951219) chip[EM]: >>> [E:30183r S:5419 M:196640851 (Ack:253023451)] (S) Msg RX from 4:00000000F040A290 [2409] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)

[image]

halilakpinarr, 2024-10-17:
When the device is reset, the descriptor cluster is not created, when you create the descriptor cluster manually, the problem is solved. I think this should be in the SDK.

closed 2024-10-17 by halilakpinarr