espressif/esp-matter #1121: How to create a composed device in a bridge device？

LinKvFM, 2024-10-17:
As shown in Figure 47 in the Matter-1.3-Core-Specification file, the endpoint 24, 25, and 26 are combined into a composed device. Now I want to create a multi-button switch as a bridge device according to this topology. How can I create it and what type of switch should it be?
[image]

jadhavrohit924, 2024-10-18:
You can create multiple endpoints representing multiple device types on the bridged node. You have to use [set_parent_endpoint](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.h#L291) to set the bridged node as a parent of your other endpoints.

> what type of switch should it be?

It's up to you, matter supports Latching, Momentary and Action (Development is in progress) switch.

LinKvFM, 2024-10-21:
I think I'm stupid, I could not create it because Apple home doesn't show what I want correctly. Is there an example that can help me?

jadhavrohit924, 2024-10-22:
We don’t have a bridge example that has composite bridged devices. We have showcased composite devices in a [refrigerator](https://github.com/espressif/esp-matter/blob/main/examples/refrigerator/main/app_main.cpp#L108-L130) example.

dhrishi, 2024-10-26:
> I think I'm stupid, I could not create it because Apple home doesn't show what I want correctly. Is there an example that can help me?

You are not. No one is :-) Please check the refrigerator example and see if it helps. Let us know in case of any questions/issues

LinKvFM, 2024-10-28:
@dhrishi Thanks. Back to the point, the composite switch still cannot be displayed in Apple home, but I can use chip-tool to access this endpoint.
Similarly, Matter bridges from other manufacturer can display composite devices in Apple home. I want to know how they do it.

My code for creating switch:
```
esp_err_t create_bridge_devices(esp_matter::endpoint_t *ep, uint32_t device_type_id, void *priv_data)
{
...
    case ESP_MATTER_ON_OFF_SWITCH_DEVICE_TYPE_ID: {
        on_off_switch::config_t switch_config;
        // err = on_off_switch::add(ep, &switch_config);
        node_t *node = node::get();
        for (uint8_t i = 0; i < count; i++)
        {
             endpoint_t *endpoint = on_off_switch::create(node, &switch_config, ENDPOINT_FLAG_NONE, (void*)switch_priv_data);
             uint16_t device_endpoint_id = esp_matter::endpoint::get_id(endpoint);
             uint16_t parent_endpoint_id = esp_matter::endpoint::get_id(ep);
             if (set_parent_endpoint(endpoint, ep) != ESP_OK) {
                    ESP_LOGE(TAG, "Failed to set parent endpoint for the bridged device");
                    endpoint::destroy(node, endpoint);
                    continue;
             }
             esp_matter::endpoint::enable(endpoint);
        }
        break;
    }
...
}
```

jonsmirl, 2024-10-31:
Arbitrary composed devices do not work in Google Home, Apple Home or Amazon.  You can certainly create the devices, but none of the big controller apps will support them. (I built my own composed devices and tested on all of the controllers). None of the big controllers support the bridge Actions cluster either.

Anyway, multi-button switch should not be a composed device. Each of the switches should be exposed as separate endpoints then add a FixedLabel to identify them.

LinKvFM, 2024-11-01:
How to set label value for endpoint? Is there a fixed format?

```
...
case ESP_MATTER_ON_OFF_SWITCH_DEVICE_TYPE_ID: {
        on_off_switch::config_t switch_config;
        err = on_off_switch::add(ep, &switch_config);
        cluster::fixed_label::config_t fixed_label_config;
        esp_matter::cluster::fixed_label::create(ep, &fixed_label_config, esp_matter::CLUSTER_FLAG_SERVER);

        esp_matter_attr_val_t attr_val = esp_matter_array("orientation/01", strlen("orientation/01"), 1);
        attribute_t *attribute = attribute::get(fixed_label_cluster, FixedLabel::Attributes::LabelList::Id);
        attribute::set_val(attribute, &attr_val);
        ...

```
The next key is marked as "orientation/02", the format is: orientation/<key zone>

jonsmirl, 2024-11-01:
Each bridged device has a Bridged Device Basic Information 0x39 cluster on the endpoint.
Attribute 5, Node label.

[image]

jonsmirl, 2024-11-04:
Matter 1.3 adds a TAGLIST on the descriptor cluster to disambiguate multiple identical endpoints.
https://github.com/espressif/esp-matter/issues/1141

You can still used Fixed Label which is free-form. The new scheme via TAGLIST is well defined.

jonsmirl, 2024-11-05:
It is not required to make it a bridge composed device, start with independent bridged devices which is much easier.

LinKvFM, 2024-11-06:
I submitted another issue #1144

jadhavrohit924, 2024-12-02:
Please close the issue, if your original issue is solved.

closed 2024-12-04 by LinKvFM