espressif/esp-matter #1122: install.sh required protoc on macOS

NinjaLikesCheez, 2024-10-18:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment**
- ESP-Matter Commit Id: 7d3ec1ec981a77280d4b5c246a6ace4e57df4a47
- ESP-IDF Commit Id: a97a7b0962da148669bb333ff1f30bf272946ade
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6
- Device Logs (Please attach the log file):

- Host Machine OS: macOS Sequoia
- Host Machine Python version: 3.12.4

- Commissioner app and versions if present: N/A
- Home hub app and versions if present: N/A
- Commissioner's logs if present: N/A

**Any additional details**
Following the [ESP-Matter dev guide](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#getting-the-repositories) there's no mention in [Matter Prerequisites](https://github.com/espressif/connectedhomeip/blob/v1.1-branch/docs/guides/BUILDING.md#prerequisites) for `protoc` to be installed. If you don't have `protoc` installed you get a cryptic error:

```Building wheels for collected packages: cffi, mobly, pyperclip, watchdog, websockets
  Building wheel for cffi (setup.py): started
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> See above for output.

  note: This error originates from a subprocess, and is likely not a problem with pip.
  Building wheel for cffi (setup.py): finished with status 'error'
  ERROR: Failed building wheel for cffi
  Running setup.py clean for cffi
  Building wheel for mobly (setup.py): started
  Building wheel for mobly (setup.py): finished with status 'done'
  Created wheel for mobly: filename=mobly-1.12.1-py3-none-any.whl size=160690 sha256=7452906c9315c9030ad7127e4f2b79d4afa18334cd26aa5cb4ec3e311d0f3e36
  Stored in directory: /Users/ninji/Library/Caches/pip/wheels/5a/7d/90/e3160a8fbcef960b5ca12aaca12f2fe5c9771d302221178383
  Building wheel for pyperclip (setup.py): started
  Building wheel for pyperclip (setup.py): finished with status 'done'
  Created wheel for pyperclip: filename=pyperclip-1.9.0-py3-none-any.whl size=11002 sha256=1d5a2b28d6efa396508228391a45b6520a84419f3aa47da1c4d7ddc687bc431e
  Stored in directory: /Users/ninji/Library/Caches/pip/wheels/e0/e8/fc/8ab8aa326e33bc066ccd5f3ca9646eab4299881af933f94f09
  Building wheel for watchdog (setup.py): started
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> See above for output.

  note: This error originates from a subprocess, and is likely not a problem with pip.
  Building wheel for watchdog (setup.py): finished with status 'error'
  ERROR: Failed building wheel for watchdog
  Running setup.py clean for watchdog
  Building wheel for websockets (setup.py): started
  Building wheel for websockets (setup.py): finished with status 'done'
  Created wheel for websockets: filename=websockets-10.4-cp312-cp312-macosx_14_0_arm64.whl size=95017 sha256=ece9312919ea10be9a3b63b75fdde6ae6eef5f811f8a9a695cb2dc7b2a235d85
  Stored in directory: /Users/ninji/Library/Caches/pip/wheels/80/cf/6d/5d7e4c920cb41925a178b2d2621889c520d648bab487b1d7fd
Successfully built mobly pyperclip websockets
Failed to build cffi watchdog
ERROR: Could not build wheels for cffi, watchdog, which is required to install pyproject.toml-based projects
ninja: build stopped: subcommand failed.
['ninja', '-C', '/Users/ninji/.esp/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']

Traceback (most recent call last):
  File "/Users/ninji/.esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 421, in install_packages
    subprocess.check_call(ninja_cmd, stdout=outs, stderr=outs)
  File "/Users/ninji/.pyenv/versions/3.12.4/lib/python3.12/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/ninji/.esp/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/ninji/.esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 995, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Users/ninji/.esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 987, in main
    return EnvSetup(**vars(parse())).setup()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/ninji/.esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 577, in setup
    result = step(spin)
             ^^^^^^^^^^
  File "/Users/ninji/.esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 766, in virtualenv
    if not virtualenv_setup.install(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/ninji/.esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 443, in install
    install_packages(gn_target)
  File "/Users/ninji/.esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 424, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/ninji/.esp/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.```

Once `protoc` is installed (`brew install protobuf`) the build proceeds.

Apologies if this is not the correct avenue to raise this issue, I couldn't raise an issue on [connectedhomeip's repo](https://github.com/espressif/connectedhomeip/) as they're disabled.

Thanks!

shubhamdp, 2024-10-18:
@NinjaLikesCheez Thanks for reporting, will check.
In a mean time if you'd like to raise it to connectedhomeip please try at - https://github.com/project-chip/connectedhomeip/issues

NinjaLikesCheez, 2024-10-18:
@shubhamdp Oh I'm so sorry - I thought I checked and couldn't see the issues but I must have been mistaken! I will raise it there too. Thank you!

dusikasss, 2024-10-18:
I have a similar error on mac os 15

`Building wheels for collected packages: cffi
  Building wheel for cffi (setup.py): started
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  Building wheel for cffi (setup.py): finished with status 'error'
  ERROR: Failed building wheel for cffi
  Running setup.py clean for cffi
Failed to build cffi
ERROR: Could not build wheels for cffi, which is required to install pyproject.toml-based projects
ninja: build stopped: subcommand failed.
['ninja', '-C', '/Users/dusikasss/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']

Traceback (most recent call last):
  File "/Users/dusikasss/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 421, in install_packages
    subprocess.check_call(ninja_cmd, stdout=outs, stderr=outs)
    ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.13/3.13.0_1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/subprocess.py", line 419, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/dusikasss/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/dusikasss/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 995, in <module>
    sys.exit(main())
             ~~~~^^
  File "/Users/dusikasss/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 987, in main
    return EnvSetup(**vars(parse())).setup()
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/Users/dusikasss/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 577, in setup
    result = step(spin)
  File "/Users/dusikasss/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 766, in virtualenv
    if not virtualenv_setup.install(
           ~~~~~~~~~~~~~~~~~~~~~~~~^
        project_root=self._project_root,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<17 lines>...
        use_pinned_pip_packages=self._use_pinned_pip_packages,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ):
    ^
  File "/Users/dusikasss/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 443, in install
    install_packages(gn_target)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^
  File "/Users/dusikasss/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 424, in install_packages
    raise subprocess.CalledProcessError(
        err.returncode, err.cmd, ins.read()
    )
subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/dusikasss/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.`

NinjaLikesCheez, 2024-10-19:
If you run the failing command you'll likely get a better picture of what went wrong as the errors won't be swallowed. It's likely an issue with locating `stdlib.h`

shubhamdp, 2024-12-10:
@dusikasss this was because of `cffi` not compatible with Python3.13 and it has been fixed in e09e20b693d1921f997f373055f92fb04e395fb1. Please pull the latest main.

closed 2024-12-10 by dhrishi