espressif/esp-matter #1123: Add an option to move BSS segment into PSRAM

jzwyssig, 2024-10-18:
I am currently working on freeing up internal memory. When esp-matter is configured to support 100 dynamic endpoints (`CONFIG_ESP_MATTER_MAX_DYNAMIC_ENDPOINT_COUNT=100`), the BSS section of `libesp_matter.a` consumes nearly 60 KB of internal memory. I measured that each endpoint allocates approximately 330 bytes of static, uninitialized, internal memory (BSS). This means that almost 30 KB are occupied by default, with an additional 30 KB for the 100 endpoints.

There is a menuconfig option, `CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY`, which moves some BSS segments (lwip, net80211, pp, bt) into PSRAM. According to the documentation [here](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/api-guides/external-ram.html#allow-bss-segment-to-be-placed-in-external-memory) and [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/linker-script-generation.html?highlight=linker%20lf#linker-script-template), the most elegant solution to reduce the internal memory footprint would be to move the esp-matter component's BSS segment into PSRAM as well.

I believe that for boards with PSRAM, this option could significantly free up internal memory and help reduce the large footprint of esp-matter. Could you provide some guidelines on how to move esp-matter's BSS segment into PSRAM?

Thank you!

shubhamdp, 2024-10-21:
@jzwyssig can you please try this `linker.lf` file (https://github.com/espressif/esp-matter/blob/main/examples/all_device_types_app/main/linker.lf), this places the BSS of `libCHIP.a` and `libesp_matter.a` into external RAM when `CONFIG_ ESP_ALLOW_BSS_SEG_EXTERNAL_MEMORY` option is enabled.

Please drop this file in your main component. Then you will need to make these minor changes to main/CMakeLists.txt...

https://github.com/espressif/esp-matter/blob/7d3ec1ec981a77280d4b5c246a6ace4e57df4a47/examples/all_device_types_app/main/CMakeLists.txt#L3 and https://github.com/espressif/esp-matter/blob/7d3ec1ec981a77280d4b5c246a6ace4e57df4a47/examples/all_device_types_app/main/CMakeLists.txt#L15

In the mean time I'll try to add these under `CONFIG_ESP_ALLOW_BSS_SEG_EXTERNAL_MEMORY` in esp-matter and chip component, so that the changes are streamlined.

jzwyssig, 2024-10-21:
Thank you @shubhamdp for the fast reply, it works perfecty! This really is a game changer. :-)

closed 2024-10-21 by jzwyssig