espressif/esp-matter #1124: ESP32-C6 Matter device certification with Matter over WiFi and Matter Thread in single EPS32-C6

bilalmalik76, 2024-10-20:
We are working on a project that involves implementing Matter over both WiFi and Thread in a single device using ESP32-C6 and a hardware button to select between matter over WiFi or Matter over Thread. I have a few questions regarding the certification process:

Is it possible to certify a device that supports Matter over both WiFi and Thread simultaneously?
Are there any specific requirements or guidelines for devices that support both protocols when seeking Matter certification?

Jerry-ESP, 2024-10-21:
> Is it possible to certify a device that supports Matter over both WiFi and Thread simultaneously?

For the device support both matter over WiFi and Thread, it can be certified in matter1.4, it's called secondary network interface, you can  get the device definition in the latest [Matter Device Library](https://groups.csa-iot.org/wg/matter-tsg/document/37009) document.

In the previous matter version, it should not support certify WiFi and Thread at the same time.

dhrishi, 2025-01-10:
Closing as the question is answered

closed 2025-01-10 by dhrishi