espressif/esp-matter #1126: PAA certificate and QR code reading in factory mode

MaplestoryAlen, 2024-10-22:
Preparing to start certification, I encountered a few issues:
1. The laboratory requires a PAA certificate, and the testing certificate in the SDK cannot be used for identity verification. Can the development certificate  (connected homeip \ credentials \ development) in the SDK be used for laboratory certification?
2.If the factory mode is used to write chip information, can the QR code be read? 
   For example:
     PrintOnboardingCodes(chip::RendezvousInformationFlag::kNone);
     char payloadBuffer[chip::QRCodeBasicSetupPayloadGenerator::kMaxQRCodeBase38RepresentationLength + 1];
     chip::MutableCharSpan qrCode(payloadBuffer);
     GetQRCode(qrCode,chip::RendezvousInformationFlag::kOnNetwork);	
     strcpy(device_info.qrcode,payloadBuffer);
     USER_LOGI(TAG,"QRCode: %s",device_info.qrcode);

shubhamdp, 2024-10-23:
1. Well, not sure if one from `credentials/development/attestation`  can be used but you can generate one using [chip-cert utility](https://github.com/project-chip/connectedhomeip/tree/master/src/tools/chip-cert#gen-att-cert).
    If you have run `esp-matter/install.sh`, then utility is present at `connectedhomeip/connectedhomeip/out/host/chip-cert` and this PATH is appended to $PATH so, after sourcing `esp-matter/export.sh` its available for use.
    Details about generating certs can be found in the [esp-matter's certification guide](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/certification.html#matter-certification).

2. As per section `5.1.7. Generation of the Passcode` in Core Matter Specification, it is recommended that you do not store the passcode on the device.
    If stored, its location must be physically isolated from where the verifier is stored.
    You will either have to keep a map of QRCode<-->device, OR you can take a look at [light_switch](https://github.com/espressif/esp-matter/tree/main/examples/light_switch/main/custom_provider) which implements custom provider which generates the random passcode and then you can use the `PrintOnboardingCodes()` to print the QRCode.

MaplestoryAlen, 2024-10-23:
> 2\. PrintOnboardingCodes()

I get discriminator, passcode, vid, pid, and then call the interface to print the QR code. An error is reported

CSV:
[image]

code:

void PrintOnboardingCodesWithFactoryDataProvider(ESP32FactoryDataProvider & provider)
{
    SetupPayload payload;

    uint16_t discriminatorValue = 2508;
    payload.discriminator.SetLongValue(discriminatorValue); 
    uint32_t setupPasscode = 10230953; 
    payload.setUpPINCode = setupPasscode; 
    payload.vendorID = 0x1575; 
    payload.productID = 0x2024; 

    PrintOnboardingCodes(payload);

    QRCodeSetupPayloadGenerator generator(payload);
    std::string qrCode;
    CHIP_ERROR err = generator.payloadBase38Representation(qrCode);

    if (err == CHIP_NO_ERROR) {
        USER_LOGW(TAG,"SetupQRCode: \n%s\n", qrCode.c_str());
    } else {
        USER_LOGW(TAG,"Failed to generate QR code: %s\n", ErrorStr(err));
    }

    std::string manualPairingCode;
    err = ManualSetupPayloadGenerator(payload).payloadDecimalStringRepresentation(manualPairingCode);

    if (err == CHIP_NO_ERROR) {
        USER_LOGW(TAG,"ManualPairingCode: %s\n", manualPairingCode.c_str());
    } else {
        USER_LOGW(TAG,"Failed to generate manual pairing code: %s\n", ErrorStr(err));
    }

}

log:
[image]

The manualPairingCode printed correctly, but the QR code reported an error. I don't know where the problem lies

MaplestoryAlen, 2024-10-24:
Thank you for your reply. The issue has been resolved

closed 2024-10-24 by MaplestoryAlen