espressif/esp-matter #1127: Can't build - Performing build step for 'chip_gn' FAILED

lacsap88, 2024-10-23:
[idf_py_stdout_output_54103.txt](https://github.com/user-attachments/files/17497025/idf_py_stdout_output_54103.txt)

Any Ideas, how to fix this?

shubhamdp, 2024-10-24:
@lacsap88 can you please elaborate more on what steps did you follow when building the `light_copy`? My guess says, there could be some stale files in `build/` directory and they may be messing this up. Can you clean the build directory and sdkconfig and try again?
```
rm -rf build sdkconfig
```

[Reference to Programming Guide: [Developing with the SDK](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#developing-with-the-sdk)]

lacsap88, 2024-10-25:
I ran through this tutorial until step 2.2.4 - this failed

I can build the blink example from the esp-idf without any problems.
but any matter example failed with

`FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a /Users/pascaljohann/esp-matter/examples/generic_switch/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build /Users/pascaljohann/esp-matter/examples/generic_switch/build/esp-idf/chip/lib/libCHIP.a `

so I'm kind of lost

shubhamdp, 2024-10-26:
Can you please help me with the below details

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- Target SoC (eg: ESP32 or ESP32-C3):
- Device Logs (Please attach the log file, starting from idf.py build or set-target):
- Host Machine OS with version:
- Host Machine Python version:

lacsap88, 2024-10-26:
ESP-Matter Commit ID: 5b4cc837172fef70bd7f7e8c15fa5cd5cbc32439 (grafted, HEAD -> main, origin/main, origin/HEAD)
ESP-IDF Commit ID: a322e6bdad4b6675d4597fb2722eea2851ba88cb (HEAD, tag: v5.2.1)
Target: ESP32C3
MacBook Pro M1 Pro - 15.0
Python 3.9.6
[Archive.zip](https://github.com/user-attachments/files/17529466/Archive.zip)

All Logs attached

lacsap88, 2024-10-30:
any ideas?

shubhamdp, 2024-10-30:
I looked at the logs and could think of few things that may be causing issue
- Can you please check if `/Users/pascaljohann/esp-matter/examples/light/build/esp-idf/chip` exists after running `idf.py build`. If yes, can you please check the permissions on that directory, or may be on `build` directory itself.
- I tried the steps from the doc but no luck reproducing the issue.
- Please make sure all the submodules are updated as per the revision that is being used for both esp-idf and esp-matter.
- I hope this is example is being built without any modifications, then can you please try in a new terminal...
```
cd /path/to/esp-matter
rm -rf `connectedhomeip/connectedhomeip/.environment/`
source /path/to/esp-idf/export.sh
./install.sh
source export.sh
cd examples/light
idf.py -v build
```

If this fails please paste/redirect the commands and their outputs (better option, cut-paste the complete terminal logs) to a file and attach it for further debugging.

If you want to just try out the firmware, you can try it through [esp-launchpad](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml).

If you want to build without setting up stuff, you can use the [esp-matter docker image](https://hub.docker.com/r/espressif/esp-matter/tags) along with [esp-idf docker guide](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/tools/idf-docker-image.html#using-the-image-interactively).

NOTE: esp-idf docker guide uses espressif/idf as the image, you will need to use the espressif/esp-matter instead.

herculesp17, 2024-11-09:
I encounter the same problem. when I run idf.py -v build I get the attached log.

[log.txt](https://github.com/user-attachments/files/17689598/log.txt)

herculesp17, 2024-11-16:
@xbrunosousa
I also tried different python versions as well as idf and matter versions on different host machines (e.g. macOS and endeavouros. In the end the docker image that @shubhamdp mentioned worked for me.

> If you want to build without setting up stuff, you can use the [esp-matter docker image](https://hub.docker.com/r/espressif/esp-matter/tags) along with [esp-idf docker [...]

These are the steps to make it work for me:

## docker
- install docker desktop and docker binary on your computer

- download the latest esp-matter image (i am using v5.2.3 right now) and this seems to work
[espressif/esp-matter Tags | Docker Hub](https://hub.docker.com/r/espressif/esp-matter/tags) 

## If you want to flash the firmware and an example on your esp32 you might have to make the ports available inside the docker container. If you just want to build the examples skip this step

```bash
pip3 install esptools
```
- Find out on which port the esp32 is connected. It's most often somewhere in `/dev/` and start the script

```bash
esp_rfc2217_server.py -v -p 4000 /path/to/your/port
```

## start the container 
copy the project directory from an esp-matter project where you store your usual projects and inside of that one run the following command to start the docker container:

```bash
docker run --rm -v $PWD:/project -w /project -u $UID -e HOME=/tmp -it --network host espressif/esp-matter
```

## erase the esp32 and flash the firmware

- you should now be able to run the following commands:

```bash
idf.py --port 'rfc2217://host.docker.internal:4000?ign_set_control' erase-flash
```

```bash
idf.py --port 'rfc2217://host.docker.internal:4000?ign_set_control' flash monitor
```

Hope this works for you as well!

North3rnL1ght, 2024-11-17:
Hi, @herculesp17 and @lacsap88. I found the main Reason for this problem. See #1157

CMake creates the Build step for chip_gn with a semicolon cutted cd command when executed with /bin/sh. It would work if the path to both folder wouldn't be escaped in a double quote. I edited the build.ninja and after that it Worked but that is not a good solution because it is auto generated. Does somebody know where the two folder Paths find place into the build.ninja? 

Maybe it is a bug in command `cd` because I only have a mac. Maybe the command `cd "folder1/test;folder2/test"` works on Debian/Ubuntu? On MacOs only `cd folder1/test;folder2/test` works fine. Could be a setting?

herculesp17, 2024-11-27:
@North3rnL1ght  thanks for the instructions. Removing the double quotes from the command didn't help. And it makes totally sense to me that neither `cd "folder1/test;folder2/test"` nor `cd folder1/test;folder2/test` are valid commands. I managed to set up a virtual machine running debian and installed everything there. then i had a look at the chip_gn build command inside ./build/build.ninja and it looks something like this:

```
#############################################
# Custom command for esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build

build esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a | ${cmake_ninja_workdir}esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build ${cmake_ninja_workdir}esp-idf/chip/lib/libCHIP.a: CUSTOM_COMMAND esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure || esp-idf/chip/args_gn
  COMMAND = cd /home/lennart/esp/esp-matter/examples/light_switch/build/esp-idf/chip && ninja esp32
  DESC = Performing build step for 'chip_gn'
  restat = 1
```
Whereas the command on my arch linux looked like you described with the double quotes and the weird cd-command
```
#############################################
# Custom command for esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build

build esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a | ${cmake_ninja_workdir}esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build ${cmake_ninja_workdir}esp-idf/chip/lib/libCHIP.a: CUSTOM_COMMAND esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure || esp-idf/chip/args_gn
  COMMAND = cd "/home/lennart/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip;/home/lennart/esp/sensor/build/esp-idf/chip" && ninja esp32
  DESC = Performing build step for 'chip_gn'
  restat = 1
```

Changing this gives me no more errors when building.

@shubhamdp  do you know how to resolve this issue since we now know that the automated generated cmkae file is generating wrong?

shubhamdp, 2024-11-28:
Duplicate of #1157.

closed 2024-12-04 by espressif-bot

shubhamdp, 2024-12-04:
We have rolled up the connectedhomeip submodule in https://github.com/espressif/esp-matter/commit/0e53bcab756254026df2406a39f391ef5a7ffd2c which has the patched fix.