espressif/esp-matter #1128: How to read the passcode(pincode)

MaplestoryAlen, 2024-10-24:
Using esp-matter-mfg-tool to generate files and download them onto the chip, only the discriminator can be read, but the passcode cannot be read. I tried using '-- enable dynamic passcode', but it also failed

  code:

       ESP32FactoryDataProvider dataProvider;
       if(dataProvider.GetVendorId(device_info.VendorId) == CHIP_NO_ERROR) 	
	{
	   USER_LOGI(TAG,"VendorId :%d",device_info.VendorId); 
	}
	if(dataProvider.GetProductId(device_info.ProductID) == CHIP_NO_ERROR) 	
	{
	   USER_LOGI(TAG,"ProductID :%d",device_info.ProductID); 
	}
	if(dataProvider.GetSetupDiscriminator(device_info.Discriminator) == CHIP_NO_ERROR) 	
	{
	   USER_LOGI(TAG,"device_info.Discriminator :%d",device_info.Discriminator); 
	}	
	if(dataProvider.GetSetupPasscode(device_info.PassCode) == CHIP_NO_ERROR) 	
	{
	   USER_LOGI(TAG,"device_info.PassCode :%ld",device_info.PassCode); 
	}
 log:
      
[image]

esp-matter-mfg-tool --discriminator 2508 --passcode 10230953 \
    --product-name 81MtAc01 --product-id 0x2024 \
    --vendor-name XzElec --vendor-id 0x1575 \
    --hw-ver 1003 --hw-ver-str 1.0.0.3  \
    --dac-in-secure-cert --target esp32s3 \
   --dac-cert  /home/alen/esp-matter/connectedhomeip/connectedhomeip/credentials/user/TrustAsia_cert/certificate.pem \
   --dac-key  /home/alen/esp-matter/connectedhomeip/connectedhomeip/credentials/user/TrustAsia_cert/privateKey.pem \
   --pai -c      /home/alen/esp-matter/connectedhomeip/connectedhomeip/credentials/user/TrustAsia_cert/Changsha_Xizhen_Electronic_Matter_Test_PAI.pem \
   -cd             /home/alen/esp-matter/connectedhomeip/connectedhomeip/credentials/user/TrustAsia_cert/esp_dac_1575_2024.der

partition.csv：
[image]

Partition.csv  does not have a pass code, but the pass code can be seen in another file
[image]

Is there any way for me to read the passcode from the factory partition? I need to generate QR codes and ManualPairingCode

shubhamdp, 2024-10-24:
As per (2) in https://github.com/espressif/esp-matter/issues/1126#issuecomment-2430989163, we are not recommending people to store the Passcode along with verifier. So, it won't be in the factory partition generated by `esp-matter-mfg-tool`.

For `-enable-dynamic-passcode`, please check the help content of [--enable-dynamic-passcode](https://github.com/espressif/esp-matter-tools/blob/main/mfg_tool/sources/mfg_tool.py#L540-L544)
```
g_commissioning.add_argument('--enable-dynamic-passcode', action="store_true", required=False,
    help='Enable dynamic passcode. If enabling this option, the generated binaries will \
    not include the spake2p verifier. so this option should work with a custom \
    CommissionableDataProvider which can generate random passcode and \
    corresponding verifier')
```
That said, if you use that option you will not have the passcode in the factory partition binary generated by `esp-matter-mfg-tool` and your app/custom-provider must be capable to generate one (as demonstrated in [light_switch](https://github.com/espressif/esp-matter/blob/main/examples/light_switch/main/custom_provider/dynamic_commissionable_data_provider.cpp)).

PS: Can you please describe the problem that you are solving rather than wondering why a solution (may not be intended for the problem you are solving) is not working for you. Can you please share (If you are willing/allowed to) what kind of device you are developing. Because, the dynamic passcode makes sense if your device really has a display or some sort of output capability.  For device such as light, won't be able to showcase the onboarding payload by itself and hence will be a part of packaging.

PPS: I would suggest you to please go through section 5.1.7. Generation of the Passcode in Core Matter Specification for better understanding about Passcode/verifier storage.

MaplestoryAlen, 2024-10-24:
> As per (2) in [#1126 (comment)](https://github.com/espressif/esp-matter/issues/1126#issuecomment-2430989163), we are not recommending people to store the Passcode along with verifier. So, it won't be in the factory partition generated by .`esp-matter-mfg-tool`
> 
> For , please check the help content of [--enable-dynamic-passcode](https://github.com/espressif/esp-matter-tools/blob/main/mfg_tool/sources/mfg_tool.py#L540-L544)`-enable-dynamic-passcode`
> 
> ```
> g_commissioning.add_argument('--enable-dynamic-passcode', action="store_true", required=False,
>     help='Enable dynamic passcode. If enabling this option, the generated binaries will \
>     not include the spake2p verifier. so this option should work with a custom \
>     CommissionableDataProvider which can generate random passcode and \
>     corresponding verifier')
> ```
> 
> That said, if you use that option you will not have the passcode in the factory partition binary generated by and your app/custom-provider must be capable to generate one (as demonstrated in [light_switch](https://github.com/espressif/esp-matter/blob/main/examples/light_switch/main/custom_provider/dynamic_commissionable_data_provider.cpp)).`esp-matter-mfg-tool`
> 
> PS: Can you please describe the problem that you are solving rather than wondering why a solution (may not be intended for the problem you are solving) is not working for you. Can you please share (If you are willing/allowed to) what kind of device you are developing. Because, the dynamic passcode makes sense if your device really has a display or some sort of output capability. For device such as light, won't be able to showcase the onboarding payload by itself and hence will be a part of packaging.
> 
> PPS: I would suggest you to please go through section 5.1.7. Generation of the Passcode in Core Matter Specification for better understanding about Passcode/verifier storage.

We are developing a temperature controller device with a screen. Due to production reasons, it is necessary to display the QR code and manual pairing code. Of course, it does not need to be dynamic. I saw that there is a "Get Setup Passcode()", so I thought it could read the information like "Get Setup Discretionar()", making it more convenient and efficient to generate QR code information in the code and display it on the screen. If I cannot read it, I can only manually enter the Passcode to write it into NVS and extract it from NVS. This is very inconvenient when producing products

shubhamdp, 2024-10-24:
Thanks for the information.

If it has screen to display onboarding code, I'd suggest a using dynamic passcode. Passcode is the proof of possession and initial PAKE happens using this as a password.

Yes, there are APIs like [GetSetupPasscode](https://github.com/project-chip/connectedhomeip/blob/master/src/include/platform/CommissionableDataProvider.h#L124C24-L124C40) or [SetSetupPasscode](https://github.com/project-chip/connectedhomeip/blob/master/src/include/platform/CommissionableDataProvider.h#L144) in [CommissionableDataProvider](https://github.com/project-chip/connectedhomeip/blob/master/src/include/platform/CommissionableDataProvider.h#L32) interface. But, in [ESP32FactoryDataProvider](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32FactoryDataProvider.h#L53) or [dynamic_commissionable_data_provider]( https://github.com/espressif/esp-matter/blob/main/examples/light_switch/main/custom_provider/dynamic_commissionable_data_provider.h) we have implemented the recommendation from the specification.

> If I cannot read it, I can only manually enter the Passcode to write it into NVS and extract it from NVS. 

You can definitely do that. But make sure you comply with those `SHALL` statements from `5.1.7. Generation of the Passcode` (please check attached screenshot)

> This is very inconvenient when producing products

I know right? But, when you think of security, it makes more sense!

[image]

MaplestoryAlen, 2024-10-25:
Thank you for your answer. I will store the passcode in another way

closed 2024-10-25 by MaplestoryAlen